zephyrproject-rtos/zephyr #39075: Unable to bind on IPv6 loopback

lodup29, 2021-10-03:
**Description**
Berkeley socket's bind() fails with errror 2 (ENOENT, No such file or directory) when attempting to bind on IPv6 loopback interface.

**To Reproduce**
Relevant configuration :
```
CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_LOOPBACK=y
CONFIG_NET_IF_MAX_IPV6_COUNT=2
```

C function (didn't find a straightforward to get loopback interface index, just tried every possibility):

```cpp
static int getSocket()
{ 
    int outSocket;
    struct sockaddr_in6 lLoopBack;

    outSocket = socket(AF_INET6, SOCK_DGRAM, IPPROTO_UDP);
    if (outSocket  < 0)
    {
       LOG_ERR("prv_createSysSocket: Failed to create TCP/UDP socket: %d", errno);
       return -1;
    }

    memset((char *)& lLoopBack, 0, sizeof(lLoopBack));

    lLoopBack.sin6_family = AF_INET6;
    lLoopBack.sin6_port = htons(60000);
    lLoopBack.sin6_addr = in6addr_loopback;

    //bind socket to port
    for(uint16_t i = 0; i < 256; i++){
        lLoopBack.sin6_scope_id = i;
        if (bind(outSocket, (struct sockaddr *) &lLoopBack, sizeof(lLoopBack)) != -1) {
            LOG_INF("Bind success with scope_id %d", lLoopBack.sin6_scope_id);
            return outSocket;
        }
    }
    switch(errno){
        case ENOENT : 
            LOG_ERR("Bind error : No such file or directory (error # %d)", errno);
            break;
         default : 
            LOG_ERR("Bind error with error %d", errno);
            break;           
    }
    
    close(outSocket);

    return -1;
}
```

**Expected behavior**

Able to find the proper scope_id and bind.

**Logs and console output**

Fails to bind with errorno 2 (No such file or directory):
```
[00:00:01.621,002] <err> client_platform: Bind error : No such file or directory (error 2)
```

**Environment (please complete the following information):**

Toolchain : gcc-arm-none-eabi-10-2020-q4-major-x86_64-linux
Zephyr version : v2.6.99-ncs1

jukkar, 2021-10-04:
Could you check if the loopback network interface has `::1` address set to it? If it is missing, then the bind will fail. You can use network shell and `net iface` command to check that.

lodup29, 2021-10-04:
@jukkar Good guess. Indeed there was simply no address associated with the loopback. Quick, not very clean, fix:
```c
struct in6_addr lLoopbackAddress = IN6ADDR_LOOPBACK_INIT;
net_if_ipv6_addr_add_by_index(1, &lLoopbackAddress, NET_ADDR_MANUAL, 0xFFFFFFFF);
```
Is it expected? IPv4 loopback interface comes with 127.0.0.1 by default, no need to set it.

jukkar, 2021-10-05:
I just wonder who sets the `127.0.0.1` address to `lo` interface, could not spot it from the code. One place where to set it could be in `drivers/net/loopback.c` and `loopback_init()`. Anyway, this looks like a bug, the address was just not set to the interface.

lodup29, 2021-10-05:
@jukkar I updated the report accordingly. Thanks.