zephyrproject-rtos/zephyr #39085: Ordering of device_map() breaks PCIe config space mapping on ARM64

superna9999, 2021-10-04:
**Describe the bug**
While developping PCIe support on ARM64, on Qemu and FVP platforms, ordering of device_map() breaks PCIe config space mapping.

With:
```
CONFIG_KERNEL_VM_SIZE=0x20000000
CONFIG_ARM64_VA_BITS_40=y
CONFIG_ARM64_PA_BITS_40=y
```

When:
`device_map(0x8080000, 131072) => 0x5ffe0000`
then
`device_map(0x4010000000, 268435456) => 0x4ffe0000`

Values at adress:
-  `4ffe0000+20000` is the same as `4ffe0000+40000`
-  `4ffe0000+28000` is the same as `4ffe0000+48000`

Whis doesn't happen if I add a static mapping OR inverse the mapping.

When:
`device_map(0x4010000000, 268435456) => 0x50000000`
then
`device_map(0x8080000, 131072) => 0x4ffe0000`

Everything works.

**To Reproduce**

I pushed a simple test to reproduce at: https://github.com/superna9999/zephyr/commit/5f11b62220683422305a8b64817736cf43e0e9b4

```
west build -b qemu_cortex_a53 tests/arch/arm64/arm64_huge_map -t run
```

**Expected behavior**
Values at adress:
-  `4ffe0000+20000` is NOT the same as `4ffe0000+40000`
-  `4ffe0000+28000` is NOT the same as `4ffe0000+48000`

**Impact**
It implies ordering of device_map()

**Logs and console output**
MMU dumps:
- when OK: https://pastebin.com/PeDB2gAB
- when NOT ok: https://pastebin.com/PPY6i2yU

**Environment (please complete the following information):**
 - OS: Any
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 6cfc6fa8e1ba6efe153c7099f113995017bd63c4

** Possible fix/hack **

I hacked `virt_region_alloc()` by increasing `offset` until the virtual address is aligned with the mapping size, and it fixes the issue.

See: https://github.com/superna9999/zephyr/commit/9926f867fdcec0a60bf97d143379142d595b07d5

BUT it's not a solution since a following device_map() will break.

AFAIK a mapping with virtual address not aligned to size should work, BUT would be suboptimal since the MMU code will need to use smaller MMU pages at beginning and end of mapping.

So maybe the bug is in the ARM64 MMU code ?

carlocaione, 2021-10-04:
tagging @npitre and @dcpleung (and myself)

dcpleung, 2021-10-05:
It looks like the same page is mapped at both addresses, `x+0x20000` and `x+0x40000`. I am not familiar with ARM64 MMU code. @carlocaione could you take a look?

superna9999, 2021-10-06:
I managed to reduce the error with RAM mapping at https://github.com/superna9999/zephyr/commit/1f211764be398a72bd23a66b7fe0d3093441ef4f and found a possible fix I pushed at #39182