zephyrproject-rtos/zephyr #39096: DNS responders assume interfaces are up at initialization

mrfuchs, 2021-10-04:
The mDNS and LLMNR responders assume network interfaces are up at system initialization/boot time. If a network interface is not up at that time (e.g. because it is configured as `NET_IF_NO_AUTO_START` and the application set `NET_CONFIG_AUTO_INIT=n`), the multicast address is added to the interface, but the multicast group will never be joined:

https://github.com/zephyrproject-rtos/zephyr/blob/643242423b9861c55be7d10310d1a56ee2d8dd7a/subsys/net/ip/igmp.c#L274-L288

With PR #39089 this becomes a problem, because while the multicast address is added to the interface, it will never receive these frames because the MAC hardware filter was not configured to do so (because it uses the multicast group monitor for its configuration).

Callstack: `mdns_responder_init()`/`llmnr_responder_init()` → `init_listener()` → `setup_ipv4_addr()` → `net_if_foreach()` → `iface_ipv4_cb()` → `net_ipv4_igmp_join()` → `igmp_send_generic()` → `igmp_send()` → `net_send_data()` → `net_if_send_data()`.

https://github.com/zephyrproject-rtos/zephyr/blob/643242423b9861c55be7d10310d1a56ee2d8dd7a/subsys/net/ip/net_if.c#L436-L441

Shouldn't the multicast groups be joined every time a network interface goes up?

jukkar, 2021-12-08:
> Shouldn't the multicast groups be joined every time a network interface goes up?

Yes, that should be done. For various IPv6 specified system level multicast addresses this is done automatically. For the IPv4, this currently needs to be done by the application as that is the only one that knows that multicast addresses it wants to use.

jukkar, 2021-12-08:
There is perhaps a way to join to multicast addresses if the network interface comes up (and it was joined already). The multicast address struct in net_if has `is_joined` field and we could go through them in network interface up case and join the addresses automatically.

rlubos, 2022-01-21:
> The multicast address struct in net_if has is_joined field and we could go through them in network interface up case and join the addresses automatically.

Hmm, that would still require the application to join the mcast group manually once, since if the interface is not up at boot it wouldn't be marked as joined? At least that's the impression I've got when I was inspecting the IGMP code.

>  For the IPv4, this currently needs to be done by the application as that is the only one that knows that multicast addresses it wants to use.

I'm wondering if this could still be automated somehow. Perhaps the networking modules that are interested in registering multicast IPv4 addresses (LLMNR, mDNS) could monitor `NET_EVENT_IF_UP` event (like DHCPv4) and rejoin the group each time net iface goes up? Perhaps it should also unset the `is_joined` flag when the iface goes down.

jukkar, 2022-01-21:
> Perhaps the networking modules that are interested in registering multicast IPv4 addresses (LLMNR, mDNS) could monitor `NET_EVENT_IF_UP` event (like DHCPv4) and rejoin the group each time net iface goes up? Perhaps it should also unset the `is_joined` flag when the iface goes down.

Sure, that sounds like a good idea. Indeed, the is_joined flag should be cleared when the interface goes down.

PR #42124 (2022-01-26): net: ipv4: Rejoin IPv4 multicast groups when iface is brought up
Make sure IPv4 multicast groups are rejoined when interface is brought up (and left when it goes down).

Fixes #39096
closed 2022-01-26 by carlescufi