zephyrproject-rtos/zephyr #39097: Bluetooth: Add a dynamic TX Power API

carlescufi, 2021-10-04:
Implement a new Bluetooth API that is able to set the TX power dynamically, sending the VS HCI commands  that are already in the tree and implemented manually in this file:
https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/hci_pwr_ctrl/src/main.c

saleh-unikie, 2021-11-16:
Is there any update to develop this feature?

Also I found that the above sample (https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/hci_pwr_ctrl/src/main.c) only works when "Extended Advertising" has not been used. While as you can find in this file:

zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv_aux.c
```
static int prepare_cb(struct lll_prepare_param *p)
{
...
#if defined(CONFIG_BT_CTLR_TX_PWR_DYNAMIC_CONTROL)
	radio_tx_power_set(lll->tx_pwr_lvl);
#else
	radio_tx_power_set(RADIO_TXP_DEFAULT);
#endif /* CONFIG_BT_CTLR_TX_PWR_DYNAMIC_CONTROL */
...
}
```

It is expected to set tx power to a desired value when "CONFIG_BT_CTLR_TX_PWR_DYNAMIC_CONTROL" has been set, but there is no way to set `lll->tx_pwr_lvl` outside of this function, even the above example can't affect on this variable.

cvinayak, 2022-07-02:
> It is expected to set tx power to a desired value when "CONFIG_BT_CTLR_TX_PWR_DYNAMIC_CONTROL" has been set, but there is no way to set `lll->tx_pwr_lvl` outside of this function, even the above example can't affect on this variable.

Support for Dynamic Tx Power in  Extended Advertising is fixed here: https://github.com/zephyrproject-rtos/zephyr/commit/a3127afb53bdbcdf0046db0a196160aadbd0d159

alwa-nordic, 2023-11-23:
Related: https://github.com/zephyrproject-rtos/zephyr/pull/65273