zephyrproject-rtos/zephyr #39131: GATT DB hash calculation is wrong on characteristic declarations using 128-bit UUIDs.

thoh-ot, 2021-10-05:
**Describe the bug**
In the function https://github.com/zephyrproject-rtos/zephyr/blob/ef0d955758e549f39d97b07089dedc4bdd05bc0e/subsys/bluetooth/host/gatt.c#L609-L682

when hashing attributes for the UUID cases:
BT_UUID_GATT_PRIMARY_VAL
BT_UUID_GATT_SECONDARY_VAL
BT_UUID_GATT_INCLUDE_VAL
BT_UUID_GATT_CHRC_VAL
BT_UUID_GATT_CEP_VAL

where (Attribute Handle || Attribute Type || Attribute Value) is to be hashed, the worst case size of attribute value is 19:

Characteristic declaration (BT_UUID_GATT_CHRC_VAL)
Attribute Value:
* Characteristic Properties: 1 octets
* Characteristic Value Handle: 2 octets
* Characteristic UUID: 2 or 16 octets

The data array used to store the attribute value is only 16 bytes long and therefor the last 3 bytes of the characteristic UUID is not included in the hash.

**Expected behavior**
Proper GATT DB hash generation.

**Impact**
Under certain conditions changes in UUID in the GATT DB will not be detected by a client, leading to errors accessing attributes

joerchan, 2021-10-06:
@thoh-ot Only 16-bits UUIDs appear to be included:
```c
 	if (attr->uuid->type != BT_UUID_TYPE_16) 
 		return BT_GATT_ITER_CONTINUE; 
 ```

Thalley, 2021-10-06:
> @thoh-ot Only 16-bits UUIDs appear to be included:
> 
> ```c
>  	if (attr->uuid->type != BT_UUID_TYPE_16) 
>  		return BT_GATT_ITER_CONTINUE; 
> ```

Was just about to write that too. I can't find anything in the spec stating that only SIG UUIDs should be used for the database hash (which also wouldn't make sense), so I think this is more a matter of missing the ability to include vendor specific UUIDs in the database hash, rather than a bug.

thoh-ot, 2021-10-06:
@joerchan @Thalley 
Was i was referring to is the characteristic declarations that is using the 16-bit Attribute UUID 0x2803 but the Attribute value is:
Characteristic Properties: 1 octets
Characteristic Value Handle: 2 octets
Characteristic UUID: 2 or 16 octets

Thalley, 2021-10-06:
> @joerchan @Thalley Was i was referring to is the characteristic declarations that is using the 16-bit Attribute UUID 0x2803 but the Attribute value is: Characteristic Properties: 1 octets Characteristic Value Handle: 2 octets Characteristic UUID: 2 or 16 octets
> 
> Where the Characteristic UUID is 128-bit.
> 
> See BLUETOOTH CORE SPECIFICATION Version 5.3 | Vol 3, Part G, 3.3.1 Characteristic declaration

Right. 
But as far as I can tell, the 16-octet array `data` is only used for the UUID, but what you are saying is that 
```
len = attr->read(NULL, attr, data, sizeof(data), 0);
``` 
returns 16 + 2 + 1 => 19 and the `data` contains the properties and the handle?

Thalley, 2021-10-06:
Looks like the `attr->read` function for reading a service, like what is done here, is using the 
```
ssize_t bt_gatt_attr_read_service(struct bt_conn *conn,
				  const struct bt_gatt_attr *attr,
				  void *buf, uint16_t len, uint16_t offset)
{
	struct bt_uuid *uuid = attr->user_data;

	if (uuid->type == BT_UUID_TYPE_16) {
		uint16_t uuid16 = sys_cpu_to_le16(BT_UUID_16(uuid)->val);

		return bt_gatt_attr_read(conn, attr, buf, len, offset,
					 &uuid16, 2);
	}

	return bt_gatt_attr_read(conn, attr, buf, len, offset,
				 BT_UUID_128(uuid)->val, 16);
}
```
function, which either return 2 or 16 octets. 

Might be a special case that causes the error. 

Do you have any additional information?

Thalley, 2021-10-06:
Ah! For `BT_UUID_GATT_CHRC_VAL` it calls `bt_gatt_attr_read_chrc` which returns a 
```
struct gatt_chrc {
	uint8_t properties;
	uint16_t value_handle;
	union {
		uint16_t uuid16;
		uint8_t  uuid[16];
	};
} __packed;
```

So you are correct, this is a bug. It also looks like the included read may fail in the future, as it only supports 16-bit UUIDs at the moment.

thoh-ot, 2021-10-06:
Data is used to read the attribute value and the attribute value can be 19 bytes long for a characteristic declaration.

The call
`len = attr->read(NULL, attr, data, sizeof(data), 0);`
reads the attribute value, but limits the length to `sizeof(data)`.

For a  characteristic declaration `attr->read` is `bt_gatt_attr_read_chrc` which calls `bt_gatt_attr_read`.

thoh-ot, 2021-10-06:
Yes i think we agree.

Thalley, 2021-10-06:
Looks like the `tc_cmac_update` function handles just about any data size we throw at it, so the solution may be as easy as updating the buffer size to 19 (`sizeof(gatt_chrc)`), but if `struct gatt_incl` is updating to handle 128-bit UUIDs, then it probably needs to be 20:
```
struct gatt_incl {
	uint16_t start_handle;
	uint16_t end_handle;
	uint16_t uuid16;
} __packed;
```

Thalley, 2021-10-06:
> According to spec (BLUETOOTH CORE SPECIFICATION Version 5.3 | Vol 3, Part G, 3.2 INCLUDE DEFINITION) The UUID is only valid for 16-bit Bluetooth UUID:
> 
> > The include declaration is an Attribute with the Attribute Type set to the UUID
> > for «Include». The Attribute Value shall be set to the included service Attribute
> > Handle, the End Group Handle, and the service UUID. The Service UUID shall
> > only be present when the UUID is a 16-bit Bluetooth UUID.The Attribute
> > Permissions shall be read only and not require authentication or authorization.

Oh, even better, than we can probably just update the buffer to 19 and be done with this issue :)

thoh-ot, 2021-10-06:
Good, i have done that in a draft PR https://github.com/zephyrproject-rtos/zephyr/pull/39133, but EDTT GATT tests are now acting up, and i'm still trying to figure out the best path forward to fix that (there seems to be some auto generated XML file, that specify what the expected values of all attributes are, but as the hashing has now been fixed, it doesn't match anymore. I'm doing some archaeological work to figure out where the source and scripts for this auto generated file is.

carlescufi, 2022-01-12:
@sjanc do you know why this doesn't make PTS fail, is there not a test case for this?

thoh-ot, 2022-01-12:
> @sjanc do you know why this doesn't make PTS fail, is there not a test case for this?

The GATT test specification has a test for it "GATT/SR/GAS/BV-02-C [Computing and Returning the Database Hash Characteristic]", and so does PTS.

the validity of that test all depends on the contents of the database, if the IUT doesn't have any 128-bit UUIDs then the test will pass just fine.

The GATT test specification has a rather long list of requirements for the contents of the test database in Section 3.3.2 Test Database Requirements, not sure if autopts follows those rules though.

PR #39133 (2022-01-26): Bluetooth: host: Fix DB hash calculation
When hashing attributes for the UUID cases:
    BT_UUID_GATT_PRIMARY_VAL
    BT_UUID_GATT_SECONDARY_VAL
    BT_UUID_GATT_INCLUDE_VAL
    BT_UUID_GATT_CHRC_VAL
    BT_UUID_GATT_CEP_VAL

where (handle || UUID || value) is to be hashed, the worst case size of
attribute value is 19:

Characteristic declaration (BT_UUID_GATT_CHRC_VAL)
Attribute Value:
* Characteristic Properties: 1 octets
* Characteristic Value Handle: 2 octets
* Characteristic UUID: 2 or 16 octets

Update the data array to accomodate for the worst case size.

Fixes #39131
commit f10daf7c (2022-01-26): Bluetooth: host: Fix DB hash calculation

When hashing attributes for the UUID cases:
    BT_UUID_GATT_PRIMARY_VAL
    BT_UUID_GATT_SECONDARY_VAL
    BT_UUID_GATT_INCLUDE_VAL
    BT_UUID_GATT_CHRC_VAL
    BT_UUID_GATT_CEP_VAL

where (handle || UUID || value) is to be hashed, the worst case size of
attribute value can be deduced from the associated union that describes
all the attribute values in the given cases.

Update the data array to accommodate for the worst case size.

Fixes #39131
closed 2022-01-26 by carlescufi