zephyrproject-rtos/zephyr #39135: samples/compression/lz4 build failed (lz4.h: No such file or directory)

frantony, 2021-10-05:
**Describe the bug**
samples/compression/lz4 failed with zephyr toolchain.

**To Reproduce**

Use Docker image from https://github.com/zephyrproject-rtos/docker-image (038d2a28d92920dc66199137652d646a2876d703) as build environment.

```
export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
export ZEPHYR_SDK_INSTALL_DIR=/opt/toolchains/zephyr-sdk-0.13.1/

cd /workdir/zephyr
source zephyr-env.sh

rm -rf build

west build -b nucleo_f103rb samples/compression/lz4

...

-- Configuring done
-- Generating done
-- Build files have been written to: /workdir/zephyr/build
-- west build: building application
[1/126] Preparing syscall dependency handling

[43/126] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /opt/toolchains/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=v2.7.0-rc3-276-ga81f8af70173 -DHSE_VALUE=8000000 -DKERNEL -DSTM32F103xB -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/workdir/zephyr/include -Izephyr/include/generated -I/workdir/zephyr/soc/arm/st_stm32/stm32f1 -I/workdir/zephyr/lib/libc/newlib/include -I/workdir/zephyr/drivers -I/workdir/zephyr/soc/arm/st_stm32/common -I/workdir/modules/hal/cmsis/CMSIS/Core/Include -I/workdir/modules/hal/stm32/stm32cube/stm32f1xx/soc -I/workdir/modules/hal/stm32/stm32cube/stm32f1xx/drivers/include -I/workdir/modules/hal/stm32/stm32cube/stm32f1xx/drivers/include/Legacy -I/workdir/modules/hal/stm32/stm32cube/common_ll/include -Os -imacros /workdir/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m3 -mthumb -mabi=aapcs -mfp16-format=ieee -imacros /workdir/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/workdir/zephyr/samples/compression/lz4=CMAKE_SOURCE_DIR -fmacro-prefix-map=/workdir/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/workdir=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=nano.specs -std=c99 -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj -c /workdir/zephyr/samples/compression/lz4/src/main.c
/workdir/zephyr/samples/compression/lz4/src/main.c:11:10: fatal error: lz4.h: No such file or directory
   11 | #include "lz4.h"
      |          ^~~~~~~
compilation terminated.
[76/126] Building C object zephyr/arch/arch/arm/core/aarch32/cortex_m/CMakeFiles/arch__arm__core__aarch32__cortex_m.dir/fault.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /workdir/zephyr/build
```

**Environment:**
    OS: (Linux)
    Toolchain (zephyr-sdk-0.13.1)
    Commit SHA: a81f8af7017313c7cf6901528245258e9b4b0f45

carlescufi, 2021-10-05:
Can you try running `west update` inside Docker?

frantony, 2021-10-05:
@carlescufi 

> Can you try running west update inside Docker?

After ```west update``` build finished successfully, thanks!

```
+ west build -b nucleo_f103rb samples/compression/lz4
-- west build: generating a build system
Including boilerplate (Zephyr base): /workdir/zephyr/cmake/app/boilerplate.cmake
-- Application: /workdir/zephyr/samples/compression/lz4
-- Zephyr version: 2.7.99 (/workdir/zephyr), build: v2.7.0-rc3-276-ga81f8af70173
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Found west (found suitable version "0.11.1", minimum required is "0.7.1")
-- Board: nucleo_f103rb
-- Cache files will be written to: /workdir/zephyr/.cache
-- Using toolchain: zephyr 0.13.1 (/opt/toolchains/zephyr-sdk-0.13.1)
-- Found dtc: /opt/toolchains/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /workdir/zephyr/boards/arm/nucleo_f103rb/nucleo_f103rb.dts
-- Generated zephyr.dts: /workdir/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /workdir/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /workdir/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /workdir/zephyr/build/zephyr/dts.cmake
Parsing /workdir/zephyr/Kconfig
Loaded configuration '/workdir/zephyr/boards/arm/nucleo_f103rb/nucleo_f103rb_defconfig'
Merged configuration '/workdir/zephyr/samples/compression/lz4/prj.conf'
Configuration saved to '/workdir/zephyr/build/zephyr/.config'
Kconfig header saved to '/workdir/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/toolchains/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /workdir/zephyr/build
-- west build: building application
[1/128] Preparing syscall dependency handling

[121/128] Linking C executable zephyr/zephyr_prebuilt.elf

[128/128] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       19992 B       128 KB     15.25%
            SRAM:        4080 B        20 KB     19.92%
        IDT_LIST:          0 GB         2 KB      0.00%
```

closed 2021-10-05 by frantony