zephyrproject-rtos/zephyr #39136: SD disk access runs into TXUNDERRUN and RXOVERRUN of SDMMC driver

benediktibk, 2021-10-05:
When using the SD disk access on a STM32H735 under load, the driver faults due to TXUNDERRUN (in case of writing to the SD card) and RXOVERRUN (in case of reading from the SD card) errors.

**To Reproduce**
Steps to reproduce the behavior:
1. Activate ethernet
2. Start downloading stuff and write it immediately to the SD card

**Expected behavior**
The data should be written without any issues to the SD card.

**Impact**
It basically makes the disk access unusable in case there is any other load on the chip. The basic cause is in the HAL itself, which uses a polling mode to get the data out:
https://github.com/zephyrproject-rtos/hal_stm32/blob/5c8275071ec1cf160bfe8c18bbd9330a7d714dc8/stm32cube/stm32h7xx/drivers/src/stm32h7xx_hal_sd.c#L923

**Environment (please complete the following information):**
 - STM32H735G Discovery Kit
 - Zephyr Version fcb73bd4f68b0cf545591bf80328f79a4ea399bf

**Additional context**
I found a workaround for this issue. In the chip there is a hardware flow control unit available, which is disabled by default. Setting Bit 17 of SDMMC_CLKCR (HWFC_EN) activates this and then the issue is gone. Of course this is still not really a nice solution, as an active polling is pretty bad for the overall system performance. I suppose the proper solution would be to use DMA for read and write operations to the SD card.

erwango, 2021-10-11:
> On top of that, I think enabling the hardware flow control by default in zephyr would be a good idea? Or am I a missing a possible issue with other usages?

It would be definitely a good idea. In a first step it could be added as a Kconfig option to limit risks (if any).

jhansen3141, 2021-10-12:
Regarding adding hardware flow control and enabling it by default, STM31F1 and STM32F4 (there maybe others) cannot use SDIO flow control due to an errata so this would not be a good idea for these SoCs.

PR #39364 (2021-11-03): stm32: sdmmc: Add IT mode and HW flow control
Fixes #39136

Convert existing SDMMC polling mode to IT driver r/w procedures.
Additionally, add a Kconfig option to enable HWFC feature.
closed 2021-11-03 by cfriedt