zephyrproject-rtos/zephyr #39176: overflow in sensor_value_from_double

5frank, 2021-10-06:
**Describe the bug**
The following function will overflow the fractional part (val2) when input value is large (when `inp >=  INT32_MAX/1000000.0 + 0.01`)
https://github.com/zephyrproject-rtos/zephyr/blob/d198da7bb4bafdd185df5122150a843c1dcea5b1/include/drivers/sensor.h#L670-L675

**To Reproduce**
```C
int main(int argc, char *argv[])
{
    double fp_in = 65536.65536; // UINT16_MAX
    struct sensor_value sv;
    sensor_value_from_double(&sv, fp_in);

    double fp_out = sensor_value_to_double(&sv);
    printf("fp_in=%f, fp_out=%f, val1=%d, val2=%d\n", fp_in, fp_out, sv.val1, sv.val2);
}
```
output:
`fp_in=65536.655360, fp_out=65536.483647, val1=65536, val2=483647`

**Expected behavior**
output from above example should be:
`fp_in=65536.655360, fp_out=65536.655360, val1=65536, val2=655360`

**Impact**
Possible hidden error as converting back to double with `sensor_value_to_double` might not reveal the overflow

**Environment (please complete the following information):**
 - Commit main/master branch

**Suggested fix**
```C
static inline int sensor_value_from_double(struct sensor_value *val, double inp)
{
    if (inp < INT32_MIN || inp > INT32_MAX) {
        return -ERANGE;
    }

    double val2 = (inp - (int32_t) inp) * 1000000;
    if (val2 < INT32_MIN || val2 > INT32_MAX) {
        return -ERANGE;
    }

    val->val1 = (int32_t) inp;
    val->val2 = (int32_t) val2;
    
    return 0;
}
```
I assume adding a return value will not break anything

carlescufi, 2021-10-06:
@5frank why don't you send a pull request with the suggested fix?

5frank, 2021-12-16:
bump! fix here https://github.com/zephyrproject-rtos/zephyr/pull/39229 but documentation build "Cancelled after 30m"

PR #39229 (2022-02-24): sensor: overflow in sensor_value from double
Fixes #39176
closed 2022-02-24 by mbolivar-nordic