zephyrproject-rtos/zephyr #39179: twister: --generate-hardware-map ends up in RuntimeError

erwango, 2021-10-06:
**Describe the bug**
When running `twister --generate-hardware-map ../map.yaml`, I'm getting the following error:
```
$ twister --generate-hardware-map ../map.yaml 
Renaming output directory to /local/mcu/zephyrproject/zephyr/twister-out.5
INFO    - Scanning connected hardware...
WARNING - Unsupported device (None): /dev/ttyS4 - n/a
Traceback (most recent call last):
  File "/local/mcu/zephyrproject/zephyr/scripts/twister", line 1315, in <module>
    main()
  File "/local/mcu/zephyrproject/zephyr/scripts/twister", line 814, in main
    hwm.save(options.generate_hardware_map)
  File "/local/mcu/zephyrproject/zephyr/scripts/pylib/twister/twisterlib.py", line 4158, in save
    yaml.dump(hwm, yaml_file, Dumper=Dumper, default_flow_style=False)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/__init__.py", line 290, in dump
    return dump_all([data], stream, Dumper=Dumper, **kwds)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/__init__.py", line 278, in dump_all
    dumper.represent(data)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 27, in represent
    node = self.represent_data(data)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 48, in represent_data
    node = self.yaml_representers[data_types[0]](self, data)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 199, in represent_list
    return self.represent_sequence('tag:yaml.org,2002:seq', data)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 92, in represent_sequence
    node_item = self.represent_data(item)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 48, in represent_data
    node = self.yaml_representers[data_types[0]](self, data)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 207, in represent_dict
    return self.represent_mapping('tag:yaml.org,2002:map', data)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 118, in represent_mapping
    node_value = self.represent_data(item_value)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 52, in represent_data
    node = self.yaml_multi_representers[data_type](self, data)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 317, in represent_object
    reduce = data.__reduce_ex__(2)
  File "/usr/lib/python3.6/multiprocessing/synchronize.py", line 101, in __getstate__
    context.assert_spawning(self)
  File "/usr/lib/python3.6/multiprocessing/context.py", line 356, in assert_spawning
    ' through inheritance' % type(obj).__name__
RuntimeError: Lock objects should only be shared between processes through inheritance
```

**Impact**
Blocking for test bench.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK, v0.13.1
 - Commit SHA: `v2.7.0-rc3-309-g02f095eb9b`

**Additional context**
After facing initialy the issue, I've run the following with no success:
-`pip3 install --user -r scripts/requirements-run-test.txt`
-`pip3 install --user -r scripts/requirements-base.txt`

PerMac, 2021-10-06:
I am getting the same error and I have the same env configuration. I found that the issue occurs when an already existing map is tried to be modified, e.g. when a new board is added to the setup

ABOSTM, 2021-10-06:
Yes, it seems to be linked to addition of new board

erwango, 2021-12-01:
I'm hitting this issue once again, and adding a new board to the bench becomes a nightmare.
Besides, for some reason adding a board manually doesn't work either as the same error happens at the next --generate-hardware-map run.

@nashif Would it be possible for you to have a look ?

erwango, 2021-12-01:
So, I've managed to work around the issue with the following steps:
-Generate a new file map2.yml (this file is empty, device information are missing)
-Fill device information 
-Rename map2.yml to the existing map.yml file

PR #41236 (2021-12-15): Various twister bug fixes
- twister: remove lock from generated hardware map
- twister: fix baud setting for detected devices
- twister: fix documentation of baud setting
- twister: do not use deprecated arguments to nrfprog

Fixes #40449 
Fixes #40450
Fixes #39179
commit d1991e99 (2021-12-15): twister: remove lock from generated hardware map

remove lock before storing yaml file.

Fixes #39179
closed 2021-12-15 by nashif