zephyrproject-rtos/zephyr #39184: HawkBit hash mismatch

oxymoron87, 2021-10-06:
**Describe the bug**
Update process always fails with 
`<err> hawkbit: Hash mismatch`

Reverting this commit https://github.com/zephyrproject-rtos/zephyr/commit/e45c07cd60d05256c4d33d52a0f164b9908dd6b4 solved issue.

Because `rsp->body_start` points to different address than `rsp->rcv_buf` by doing this 

```
else {
	/*
	 * more general case where body data is set, but no need
	 * to take the HTTP header into account
	 */
	body_data = rsp->body_start;
	body_len = rsp->data_len;
}
```

`body_data` also points to the wrong position and `mbedtls_md_update(&hb_context.dl.hash_ctx, body_data, body_len)` is feeded with wrong data.
Next IF statement
```
if ((rsp->body_found == 1) && (body_data == NULL)) {
	body_data = rsp->recv_buf;
	body_len = rsp->data_len;
}
```

would set `body_data` correctly but `body_data` is already set and thus != NULL

**Expected behavior**
Successful update retrieval

**Logs and console output**
```
[00:10:13.106,000] <dbg> hawkbit.response_cb: Download percentage: 33% 
[00:10:13.113,000] <dbg> hawkbit.response_cb: Download percentage: 66% 
[00:10:13.124,000] <dbg> hawkbit.response_cb: Download percentage: 100% 
[00:10:13.124,000] <err> hawkbit: Hash mismatch
[00:10:13.124,000] <dbg> hawkbit: resp
                                  d3 dc 9a de ed 48 d9 19  ad f9 36 1d bc e6 ac 3b |.....H.. ..6....;
                                  56 9e 44 d8 a1 42 45 5d  9b 60 7f 5a 32 63 fe 63 |V.D..BE] .`.Z2c.c
[00:10:13.124,000] <dbg> hawkbit: file
                                  97 c9 56 7b 8a 68 d0 58  f8 75 74 59 f9 da 6f 18 |..V{.h.X .utY..o.
                                  e4 91 7a 23 71 8e 4a 57  2b 41 f9 a1 7e 09 38 54 |..z#q.JW +A..~.8T

```

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr OS build v2.7.0-rc2

ycsin, 2021-10-13:
@oxymoron87 There's an inflight PR trying to fix this, could you try #39210 ?

ycsin, 2022-03-20:
Should be fixed by #43237

closed 2022-03-20 by ycsin