zephyrproject-rtos/zephyr #39216: Twister: Broken on NRF52840 with pyocd option timeout error

pratapnaidu, 2021-10-06:
**Describe the bug**
The twister framework bails with timeout error when executed while choosing pyocd option as west runner

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout the branch above 2.5.0 version
2. set the GNUARMEMB toolchain, both path and toolchain
3. ./scripts/twister --device-testing  --device-serial /dev/ttyACM0 --west-flash --west-runner=pyocd -p nrf52840dk_nrf52840 -T tests/drivers/adc/adc_api -vv

**Expected behavior**
```
<<<-----------------------------------(on zephyr-v2.5.0)----------------------------------------------->>>

DEBUG   - DEVICE: PROJECT EXECUTION SUCCESSFUL
DEBUG   - run status: nrf52840dk_nrf52840/tests/drivers/adc/adc_api/drivers.adc passed
INFO    - 1/1 nrf52840dk_nrf52840       tests/drivers/adc/adc_api/drivers.adc    PASSED (device 5.319s)
```

**Impact**
Projects using pyocd  with cmsis daplink firmware on nrf52840 dk board

**Logs and console output**
```
DEBUG   - Timed out while monitoring serial output on nrf52840dk_nrf52840
DEBUG   - run status: nrf52840dk_nrf52840/tests/drivers/adc/adc_api/drivers.adc timeout
INFO    - 1/1 nrf52840dk_nrf52840       tests/drivers/adc/adc_api/drivers.adc   FAILED Timeout (device 64.248s)
```

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain: gnuarmemb
 - Commit SHA or Version used  (1af1052d285fb0a826e7cc7037cf0ba7e73bb330) / (2.7.0-rc2)

**Additional context**
There is one interesting observation though,the same twister command  works on 2.5.0 version, and when we execute on 2.6 and higher we get this warning of possible DAPLINK:Overflow error, as highlighted below

```
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_adc_sample_with_interval
DEBUG   - DEVICE: sample_with_interval_callback: sampling 0
DEBUG   - DEVICE: sample_with_interval_callback: sampling 1
DEBUG   - DEVICE: sample_with_interval_callback:<**DAPLink:Overflow**>
DEBUG   - DEVICE: *** Booting Zephyr OS build v2.7.0-rc2  ***
DEBUG   - DEVICE: Running test suite adc_basic_test
```

carlescufi, 2021-10-07:
Could you try setting CONFIG_ NULL_POINTER_EXCEPTION_DETECTION= NULL_POINTER_EXCEPTION_DETECTION_NONE and see if that helps?

niklasparnesten, 2021-11-12:
> Could you try setting CONFIG_ NULL_POINTER_EXCEPTION_DETECTION= NULL_POINTER_EXCEPTION_DETECTION_NONE and see if that helps?

I am having the same problem when running twister with openocd and nrf.

Setting CONFIG_NULL_POINTER_EXCEPTION_DETECTION_NONE=y in the testcase.yml works for me, but I do not understand why...

nashif, 2021-12-15:
Now a twister issue, this is the good old CONFIG_NULL_POINTER_EXCEPTION_DETECTION bug.

carlescufi, 2021-12-20:
This is essentially this issue: #33590

PR #41335 (2021-12-20): soc: nrf: Disable DWT NULL pointer detection by default
Having DWT-based NULL-pointer detection enabled by default trips a lot
of users since it will make any debugging fail when enabled. Disable it
by default and let users enable it if they desire to.

Fixes #39216.
commit f0765409 (2021-12-20): soc: nrf: Disable DWT NULL pointer detection by default

Having DWT-based NULL-pointer detection enabled by default trips a lot
of users since it will make any debugging fail when enabled. Disable it
by default and let users enable it if they desire to.

Fixes #39216.
closed 2021-12-20 by carlescufi