zephyrproject-rtos/zephyr #39220: stm32: g0b1re: LOG (& probably shell) won't work if `CONFIG_PM=y`

ycsin, 2021-10-07:
**Describe the bug**
At least the logging subsys won't work if both `CONFIG_SHELL` and `CONFIG_PM` is enabled. Im not very sure about shell, since the UART doesn't work when MCU entered STOP mode.

**To Reproduce**
Compile the following on `nucleo_g0b1re`

main.c:
```c
#include <zephyr.h>

#include <logging/log.h>
LOG_MODULE_REGISTER(main, 4);

void main(void)
{
	while(1) {
		LOG_INF("Hello World! %s", CONFIG_BOARD);
		printk("printk! %s\r\n", CONFIG_BOARD);
		k_sleep(K_SECONDS(2));
	}
}
```

prj.conf

```conf
CONFIG_SHELL=y
CONFIG_LOG=y
CONFIG_PM=y
```

Log output, notice that there's also some seemingly garbage data between the "p" and the "rintk!...", they are actually the color code + the `uart:~$` from the shell:
```
*** Booting Zephyr OS build v2.7.0-rc3-156-gb55cd9bc77d7  ***
printk! nucleo_g0b1re
pintk! nucleo_g0b1re
p[rintk! nucleo_g0b1re
pmrintk! nucleo_g0b1re
rintk! nucleo_g0b1re
p
 rintk! nucleo_g0b1re
rintk! nucleo_g0b1re
p
 rintk! nucleo_g0b1re
pintk! nucleo_g0b1re
p[rintk! nucleo_g0b1re
p1rintk! nucleo_g0b1re
p;rintk! nucleo_g0b1re
p3rintk! nucleo_g0b1re
p2rintk! nucleo_g0b1re
pmrintk! nucleo_g0b1re
```

If I compile without `CONFIG_SHELL`, `prj.conf`:

```conf
CONFIG_LOG=y
CONFIG_PM=y
```

Log output:

```
*** Booting Zephyr OS build v2.7.0-rc3-156-gb55cd9bc77d7  ***
printk! nucleo_g0b1re
[00:00:00.005,000] <inf> main: Hello World! nucleo_g0b1re
printk! nucleo_g0b1re
[00:00:02.007,000] <inf> main: Hello World! nucleo_g0b1re
printk! nucleo_g0b1re
[00:00:04.009,000] <inf> main: Hello World! nucleo_g0b1re
printk! nucleo_g0b1re
[00:00:06.011,000] <inf> main: Hello World! nucleo_g0b1re
printk! nucleo_g0b1re
[00:00:08.013,000] <inf> main: Hello World! nucleo_g0b1re
printk! nucleo_g0b1re
[00:00:10.016,000] <inf> main: Hello World! nucleo_g0b1re
printk! nucleo_g0b1re
```

If I compile without `CONFIG_PM`, `prj.conf`:

```conf
CONFIG_SHELL=y
CONFIG_LOG=y
```

Log output:

```
*** Booting Zephyr OS build v2.7.0-rc3-156-gb55cd9bc77d7  ***
printk! nucleo_g0b1re

[00:00:00.006,000] <inf> main: Hello World! nucleo_g0b1re
uart:~$ printk! nucleo_g0b1re
[00:00:02.008,000] <inf> main: Hello World! nucleo_g0b1re
uart:~$ printk! nucleo_g0b1re
[00:00:04.010,000] <inf> main: Hello World! nucleo_g0b1re
uart:~$ printk! nucleo_g0b1re
[00:00:06.012,000] <inf> main: Hello World! nucleo_g0b1re
uart:~$ printk! nucleo_g0b1re
[00:00:08.014,000] <inf> main: Hello World! nucleo_g0b1re
uart:~$ printk! nucleo_g0b1re
[00:00:10.016,000] <inf> main: Hello World! nucleo_g0b1re
uart:~$ printk! nucleo_g0b1re
```

**Impact**
Harder to debug without LOG.

**Environment (please complete the following information):**
 - win10
 - gnuarmemb
 - 0e1ecb373a

ycsin, 2021-10-07:
@erwango ^^

ycsin, 2021-10-12:
> Then can you rename as :
> stm32: g0b1re: SHELL not functional if `CONFIG_LOG=y & CONFIG_PM=y` ?

But LOG subsys doesn't work when `prj.conf`:

```
CONFIG_SHELL=y
CONFIG_LOG=y
CONFIG_PM=y
```

See my first "Log" (console) output, the "[00:00:00.005,000] <inf> main: Hello World! nucleo_g0b1re" isn't being printed.

erwango, 2021-10-12:
> See my first "Log" (console) output, the "[00:00:00.005,000] main: Hello World! nucleo_g0b1re" isn't being printed.

Well, I can see "[00:00:00.006,000] <inf> main: Hello World! nucleo_g0b1re", and log messages printed every 2 sec. It seems pretty close to me.

erwango, 2021-11-02:
> I'm not sure if stm32's uart driver support poll_out and interrupt APIs at the same time, but most likely it is not supported

It is.

emillindq, 2021-11-03:
I am 99% sure this problem has to do with mixed polling/irq calls. I just pulled latest Zephyr and no logs came out if LOG, SHELL & PM was active. Shell uart backend calls the irq based uart functions, and receives its callbacks fine in the beginning but then as soon as the banner comes ("Booting Zephyr OS...") it stops to receive callbacks. The banner uses polling based uart calls. Disable the banner and it works flawless.

I also tracked it down to commit e3f4907efe6438a4b28f17167c58a2e25865c295 before this commit it worked

erwango, 2021-11-03:
@emillindq, @ycsin Can you check if https://github.com/zephyrproject-rtos/zephyr/pull/39752 is fixing this issue ?

emillindq, 2021-11-03:
> @emillindq, @ycsin Can you check if #39752 is fixing this issue ?

Unfortunately not, although for one of my builds it does. I suspect it has to do with order of execution and this PR resolves one case of multiple. Once again, disabling CONFIG_BOOT_BANNER with mentioned PR and LOG_XXX messages comes out

erwango, 2021-11-08:
@emillindq, @ycsin New attempt from @jdascenzio: https://github.com/zephyrproject-rtos/zephyr/pull/40173

closed 2022-02-22 by github-actions[bot]