zephyrproject-rtos/zephyr #39242: net: sockets: Zephyr Fatal in dns_resolve_cb if dns request was attepmted in offline state

AndreyDodonov-EH, 2021-10-07:
**Describe the bug**

In case if dns resolving was attempted in offline state,
going back online triggers Zephyr Fatal.

Source of the crash:
subsys\net\lib\sockets\getaddrinfo.c

The reason is that `dns_resolve_cb` tries to give semaphore which doesn't exist anymore.
Error lifecycle:
- `z_impl_z_zsock_getaddrinfo_internal ` creates semaphore as part of `struct getaddrinfo_state ai_state;`
- it waits for the semaphore: `k_sem_take(&ai_state.sem, K_MSEC(CONFIG_NET_SOCKETS_DNS_TIMEOUT + 100));`
- if semaphore was not given, it returns: `if (ret == -EAGAIN) {return DNS_EAI_AGAIN;}`, thus deallocating the semaphore
- we are back online (e.g. wifi network is back up, cable plugged, etc.), and thread rx_q eventually calls `dns_resolve_cb`
- `dns_resolve_cb` tries to give the semaphore:  `k_sem_give(&state->sem);`. But it doesn't exist anymore. ZEPHYR FATAL.

**To Reproduce**
Steps to reproduce the behavior:
1. Any platform with network connectivity 
2. Call zsock_getaddrinfo when the internet is not present.
3. After  K_MSEC(CONFIG_NET_SOCKETS_DNS_TIMEOUT + 100) get back online (platform-depedent)
4. See error

**Expected behavior**
No crash occurs.

**Impact**
It is showstopper, because there can be many reasons for network disconnects and business logic doesn't have to check low-level connection state (loose coupling).

Imagine if your PC goes offline (e.g. wi-fi network down), meanwhie you attempt to make a `git push`. Or send an e-mail.
Then you go back online and enjoy BSOD.

**Logs and console output**
```
[00:00:14.225,128] <err> os: ***** HARD FAULT *****
[00:00:14.231,018] <err> os:   Fault escalation (see below)
[00:00:14.237,609] <err> os: ***** BUS FAULT *****
[00:00:14.243,408] <err> os:   Precise data bus error
[00:00:14.249,481] <err> os:   BFAR Address: 0x6174626f
[00:00:14.255,798] <err> os: r0/a1:  0x64656e69  r1/a2:  0x0006a504  r2/a3:  0x6174626f
[00:00:14.265,045] <err> os: r3/a4:  0x20746f4e r12/ip:  0x000000fe r14/lr:  0x000659d3 
[00:00:14.274,261] <err> os:  xpsr:  0x21000000
[00:00:14.279,846] <err> os: Faulting instruction address (r15/pc): 0x000657a6
[00:00:14.288,208] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:14.296,478] <err> os: Current thread: 0x20016018 (rx_q[0])
[00:00:14.308,715] <err> os: Halting system
```
**Environment (please complete the following information):**
Interestingly enough, faulting instruction resolves to zephyr/include/sys/dlist.h:513 (with addr2line tool), and 
r14/lr:  0x000659d3  to kernel/include/ksched.h:167
Probably because scheduler crashes on an attempt to access some random address as semaphore.

**Additional context**
Minimalistic and most robust fix would be to make structure with semaphore static, e.g.
in `z_impl_z_zsock_getaddrinfo_internal`:
`static struct getaddrinfo_state ai_state;`

Sir-Branch, 2021-12-13:
Is there any updates on this? I believe this might be causing some issues for me, I'm not experiencing a crash but my call to getaddrinfo is hanging (running on an LTE-M modem, so it might be platform dependent)

AndreyDodonov-EH, 2021-12-13:
@Sir-Branch Not yet, there is a workaround in this PR, which worked for me:
https://github.com/zephyrproject-rtos/zephyr/pull/39243

But this workaround will prevent you from dns resolving from multiple threads, so it is not an acceptable solution.

Sir-Branch, 2021-12-13:
I appreciate the update :smile:

rlubos, 2022-01-31:
I've submitted a fix proposal in https://github.com/zephyrproject-rtos/zephyr/pull/42321

PR #42321 (2022-02-01): sockets: getaddrinfo: Fix possible crash when callback is delayed
In case system workqueue processing is delayed for any reason, and
resolver callback is exectued after `getaddrinfo()` call already timed
out, the system would crash as the callback makes use of the user data
allocated on a stack within `getaddrinfo()` funtcion.

Prevent that, by cancelling the DNS request explicitly from the
`getaddrinfo()` context, therefore preventing the resolver callback
from being executed after the `getaddrinfo()` call ends.

Fixes #39242
closed 2022-02-01 by MaureenHelm