zephyrproject-rtos/zephyr #39347: Static object constructors do not execute on the NATIVE_POSIX_64 target

aleksei-timofeyev, 2021-10-12:
**Describe the bug**
All C++ global object constructors are not executed upon startup of a NATIVE_POSIX_64 executable. This breaks code that relies upon constructor side effects (such as registering unit tests with a test framework).

**To Reproduce**
Steps to reproduce the behavior:
1. Create a trivial C++ application with the following `main.cc`:
```c++
#include <string>
#include <iostream>

struct SideEffectConstructor {
    SideEffectConstructor(std::string s) {
        std::cerr << "Constructing " << s << std::endl;
    }
};

static auto object_a = SideEffectConstructor{"Object A"};
static auto object_b = SideEffectConstructor{"Object B"};
static auto object_c = SideEffectConstructor{"Object C"};
static auto object_d = SideEffectConstructor{"Object D"};

void main() {
    exit(0);
}

```

2. Build the application for the `native_posix_64` target with following `Kconfig`:
```
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_STD_CPP20=y
CONFIG_ASSERT=y
CONFIG_SPEED_OPTIMIZATIONS=y
```
3. Execute the application.
4. Observe the following console output:
```
(.venv) bash-5.1$ ./build/zephyr/zephyr.elf 
*** Booting Zephyr OS build v2.7.0-rc3-259-ga6dcf333a1b8  ***
```

**Expected behavior**
The following lines should be printed to the terminal exactly once:
```
Constructing Object A
Constructing Object B
Constructing Object C
Constructing Object D
```

**Impact**
This bug results in a major discrepancy between the behavior of code running on the native_posix_64 board and the behavior of that same code running on physical board. This bug also interferes with the use of common unit test frameworks that rely on auto-registration of tests via global object constructors.

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain: Host GNU 11.2.1
 - Zephyr OS build v2.7.0-rc3-259-ga6dcf333a1b8

**Additional context**
The bug first appears at commit d10b46dc3be4492f5e7794c0a73c0e925d51c2ee.

#36858

stephanosio, 2021-10-13:
@palchak-google FYI

palchak-google, 2021-10-14:
Can you confirm that those four objects are actually in the resulting executable? What happens if you disable optimizations?

When I built my original example, I had all optimizations disabled. I think what may be happening is that the compiler can see that those four static objects are never used and thus they're being elided.

aleksei-timofeyev, 2021-10-14:
> Can you confirm that those four objects are actually in the resulting executable? What happens if you disable optimizations?
```
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_STD_CPP20=y
CONFIG_ASSERT=y
CONFIG_NO_OPTIMIZATIONS=y
```
With configuration above all objects are present:
```
 .bss._ZL8object_a
                0x0000000000411d3d        0x1 app/libapp.a(main.cpp.obj)
 .bss._ZL8object_b
                0x0000000000411d3e        0x1 app/libapp.a(main.cpp.obj)
 .bss._ZL8object_c
                0x0000000000411d3f        0x1 app/libapp.a(main.cpp.obj)
 .bss._ZL8object_d
                0x0000000000411d40        0x1 app/libapp.a(main.cpp.obj)
```
Below you can find 2 archive with following files:
`zephyr.elf  zephyr.lst  zephyr.map  zephyr.stat`

With patch:
[1.tar.gz](https://github.com/zephyrproject-rtos/zephyr/files/7349301/1.tar.gz)
Without patch:
[2.tar.gz](https://github.com/zephyrproject-rtos/zephyr/files/7349303/2.tar.gz)

palchak-google, 2021-10-20:
I am unable to reproduce this issue. I'm building with:

OS: Linux
Toolchain: gcc (Debian 10.3.0-11) 10.3.0
Zephyr: commit 86ad71827c8e56ec5d20da488fad3e514b155486 (HEAD, tag: v2.7.99, tag: v2.7.0-rc3)

I get the same output with `native_posix` and `native_posix_64` targets:
```
$ ./zephyr/zephyr.elf
Constructing Object A
Constructing Object B
Constructing Object C
Constructing Object D
*** Booting Zephyr OS build v2.7.99  ***

```

I do not have the ability at the moment to upgrade to GCC 11.2.1 to confirm whether the issue is related to the toolchain version (10.3.0 vs 11.2.1).

Project files used:
[static_objects.zip](https://github.com/zephyrproject-rtos/zephyr/files/7384781/static_objects.zip)

palchak-google, 2021-11-03:
Well we have ourselves a real head scratcher here. I was able to install GCC v11.2.0 last week, and when building with that toolchain I also failed to reproduce the issue (for either `native_posix` or `native_posix_64` target):
```
~/projects/zephyr-upstream/app/static_objs/BUILD.64 ❯❯❯ cmake .. -DBOARD=native_posix_64
Including boilerplate (Zephyr workspace): /usr/local/google/home/palchak/projects/zephyr-upstream/zephyr/cmake/app/boilerplate.cmake
-- Application: /usr/local/google/home/palchak/projects/zephyr-upstream/app/static_objs
-- Zephyr version: 2.7.0-rc3 (/usr/local/google/home/palchak/projects/zephyr-upstream/zephyr), build: v2.7.99
-- Found Python3: /usr/bin/python3.9 (found suitable exact version "3.9.7") found components: Interpreter 
-- Found west (found suitable version "0.11.0", minimum required is "0.7.1")
-- Board: native_posix_64
-- Cache files will be written to: /usr/local/google/home/palchak/.cache/zephyr
-- Found dtc: /usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /usr/local/google/home/palchak/projects/zephyr-upstream/zephyr/boards/posix/native_posix/native_posix_64.dts
-- Generated zephyr.dts: /usr/local/google/home/palchak/projects/zephyr-upstream/app/static_objs/BUILD.64/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /usr/local/google/home/palchak/projects/zephyr-upstream/app/static_objs/BUILD.64/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /usr/local/google/home/palchak/projects/zephyr-upstream/app/static_objs/BUILD.64/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /usr/local/google/home/palchak/projects/zephyr-upstream/app/static_objs/BUILD.64/zephyr/dts.cmake
Parsing /usr/local/google/home/palchak/projects/zephyr-upstream/zephyr/Kconfig
Loaded configuration '/usr/local/google/home/palchak/projects/zephyr-upstream/zephyr/boards/posix/native_posix/native_posix_64_defconfig'
Merged configuration '/usr/local/google/home/palchak/projects/zephyr-upstream/app/static_objs/prj.conf'
Configuration saved to '/usr/local/google/home/palchak/projects/zephyr-upstream/app/static_objs/BUILD.64/zephyr/.config'
Kconfig header saved to '/usr/local/google/home/palchak/projects/zephyr-upstream/app/static_objs/BUILD.64/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
CMake Deprecation Warning at /usr/local/google/home/palchak/projects/zephyr-upstream/modules/lib/civetweb/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Warning at /usr/local/google/home/palchak/projects/zephyr-upstream/zephyr/CMakeLists.txt:1643 (message):
  __ASSERT() statements are globally ENABLED

-- Configuring done
-- Generating done
-- Build files have been written to: /usr/local/google/home/palchak/projects/zephyr-upstream/app/static_objs/BUILD.64
~/projects/zephyr-upstream/app/static_objs/BUILD.64 ❯❯❯ ninja
[1/99] Preparing syscall dependency handling

[94/99] Linking CXX executable zephyr/zephyr_prebuilt.elf

[99/99] Linking CXX executable zephyr/zephyr.elf
~/projects/zephyr-upstream/app/static_objs/BUILD.64 ❯❯❯ ./zephyr/zephyr.elf
Constructing Object A
Constructing Object B
Constructing Object C
Constructing Object D
*** Booting Zephyr OS build v2.7.99  ***
```

As a sanity check, what happens if you build the executable without Zephyr at all? That is, change the signature of `main` to `int main()` and then build a simple executable:
```
gcc -o test.elf ../main.cc -lstdc++ -lgcc
```
I get the following output (as expected):
```
~/projects/zephyr-upstream/app/static_objs/BUILD.noz ❯❯❯ ./test.elf
Constructing Object A
Constructing Object B
Constructing Object C
Constructing Object D
```

palchak-google, 2021-11-03:
Also, since you happen to be using GCC-11, take note of issue #40023.

wojciechslenska, 2022-04-27:
Hello,
I observed this problem after upgrade of Ubuntu from 20.04 to 22.04. In tests I used static_obs example from this thread. Only change is that C++ is set to 17 instead of 20.

Ubuntu 20.04 GCC 9.4 -> works perfect
```
user@ubuntu20 /b/g/z/static_objs (develop)> west build -b native_posix -t run
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /build/griffin_root/zephyr_base/static_objs
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 3.0.99 (/build/griffin_root/zephyr_base/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: native_posix
-- Found dtc: /usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /build/griffin_root/zephyr_base/zephyr/boards/posix/native_posix/native_posix.dts
-- Generated zephyr.dts: /build/griffin_root/zephyr_base/static_objs/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /build/griffin_root/zephyr_base/static_objs/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /build/griffin_root/zephyr_base/static_objs/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /build/griffin_root/zephyr_base/static_objs/build/zephyr/dts.cmake
Parsing /build/griffin_root/zephyr_base/zephyr/Kconfig
Loaded configuration '/build/griffin_root/zephyr_base/zephyr/boards/posix/native_posix/native_posix_defconfig'
Merged configuration '/build/griffin_root/zephyr_base/static_objs/prj.conf'
Configuration saved to '/build/griffin_root/zephyr_base/static_objs/build/zephyr/.config'
Kconfig header saved to '/build/griffin_root/zephyr_base/static_objs/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
CMake Warning at /build/griffin_root/zephyr_base/zephyr/CMakeLists.txt:1803 (message):
  __ASSERT() statements are globally ENABLED

-- Configuring done
-- Generating done
-- Build files have been written to: /build/griffin_root/zephyr_base/static_objs/build
-- west build: running target run
[1/102] Preparing syscall dependency handling

[2/102] Generating include/generated/version.h
-- Zephyr version: 3.0.99 (/build/griffin_root/zephyr_base/zephyr), build: zephyr-v3.0.0-2756-gc5dc184b7432
[97/102] Linking CXX executable zephyr/zephyr_pre0.elf

[101/102] Linking CXX executable zephyr/zephyr.elf

[101/102] cd /build/griffin_root/zephyr_base/static_objs/build && /build/griffin_root/zephyr_base/static_objs/build/zephyr/zephyr.exe
Constructing Object A
Constructing Object B
Constructing Object C
Constructing Object D
*** Booting Zephyr OS build zephyr-v3.0.0-2756-gc5dc184b7432  ***

```

Ubuntu 22.04 GCC 11.2 -> ctors are not executed
```
user@ubuntu22 /m/w/c/g/z/static_objs (develop)> west build -b native_posix -t run
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /media/wsl/code/griffin_root/zephyr_base/static_objs
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.4") found components: Interpreter
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 3.0.99 (/media/wsl/code/griffin_root/zephyr_base/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: native_posix
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /media/wsl/code/griffin_root/zephyr_base/zephyr/boards/posix/native_posix/native_posix.dts
-- Generated zephyr.dts: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/include/generated/device_extern.h
[... 12 lines cut ...]

-- Configuring done
-- Generating done
-- Build files have been written to: /media/wsl/code/griffin_root/zephyr_base/static_objs/build
-- west build: running target run
[1/102] Preparing syscall dependency handling

[2/102] Generating include/generated/version.h
-- Zephyr version: 3.0.99 (/media/wsl/code/griffin_root/zephyr_base/zephyr), build: zephyr-v3.0.0-2756-gc5dc184b7432
[97/102] Linking CXX executable zephyr/zephyr_pre0.elf

[101/102] Linking CXX executable zephyr/zephyr.elf

[101/102] cd /media/wsl/code/griffin_root/zephyr_base/static_objs/build && /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/zephyr.exe
*** Booting Zephyr OS build zephyr-v3.0.0-2756-gc5dc184b7432  ***
```
When I compiled and run this as normal application constructors works properly.
```
user@ubuntu22 /m/w/c/g/z/static_objs (develop)> gcc -o test.elf main.cc -lstdc++
user@ubuntu22 /m/w/c/g/z/static_objs (develop)> ./test.elf
Constructing Object A
Constructing Object B
Constructing Object C
Constructing Object D
```

palchak-google, 2022-04-27:
The differing results with different toolchains points indicates that this likely has something to do with a change in libstdc++. For reference, I used the following toolchain and library to build for `native_posix`, `native_posix_64`, and native Linux (non-Zephyr) targets:

* gcc (Debian 11.2.0-16) 11.2.0
* libstdc++-11-dev:amd64 11.2.0-16
* libstdc++-11-dev:i386  11.2.0-16

These version are all installed via official Debian packages.

Re-building with the latest ToT for Zephyr again fails to reproduce for me:
```
Constructing Object A
Constructing Object B
Constructing Object C
Constructing Object D
*** Booting Zephyr OS build zephyr-v3.0.0-3121-gd490358e283f  ***
```

wojciechslenska, 2022-04-28:
I made some more checks in Ubuntu 22.04, looks like constructors works properly only in GCC 9. 
```
wsl@T15 /m/w/c/g/z/static_objs (develop)> sudo apt install gcc-9 g++-9 g++-9-multilib
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
g++-9 is already the newest version (9.4.0-5ubuntu1).
g++-9-multilib is already the newest version (9.4.0-5ubuntu1).
gcc-9 is already the newest version (9.4.0-5ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
wsl@T15 /m/w/c/g/z/static_objs (develop)> sudo update-alternatives --set gcc /usr/bin/gcc-9
update-alternatives: using /usr/bin/gcc-9 to provide /usr/bin/gcc (gcc) in manual mode
wsl@T15 /m/w/c/g/z/static_objs (develop)> sudo update-alternatives --set g++ /usr/bin/g++-9
update-alternatives: using /usr/bin/g++-9 to provide /usr/bin/g++ (g++) in manual mode
wsl@T15 /m/w/c/g/z/static_objs (develop)> gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.4.0-5ubuntu1' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-9-bVKGhJ/gcc-9-9.4.0/debian/tmp-nvptx/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 9.4.0 (Ubuntu 9.4.0-5ubuntu1) 
wsl@T15 /m/w/c/g/z/static_objs (develop)> rm build -fr
wsl@T15 /m/w/c/g/z/static_objs (develop)> west build -b native_posix -t run
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /media/wsl/code/griffin_root/zephyr_base/static_objs
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.4") found components: Interpreter 
-- Cache files will be written to: /home/wsl/.cache/zephyr
-- Zephyr version: 3.0.99 (/media/wsl/code/griffin_root/zephyr_base/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: native_posix
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /media/wsl/code/griffin_root/zephyr_base/zephyr/boards/posix/native_posix/native_posix.dts
-- Generated zephyr.dts: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/dts.cmake
Parsing /media/wsl/code/griffin_root/zephyr_base/zephyr/Kconfig
Loaded configuration '/media/wsl/code/griffin_root/zephyr_base/zephyr/boards/posix/native_posix/native_posix_defconfig'
Merged configuration '/media/wsl/code/griffin_root/zephyr_base/static_objs/prj.conf'
Configuration saved to '/media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/.config'
Kconfig header saved to '/media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
CMake Warning at /media/wsl/code/griffin_root/zephyr_base/zephyr/CMakeLists.txt:1803 (message):
  __ASSERT() statements are globally ENABLED

-- Configuring done
-- Generating done
-- Build files have been written to: /media/wsl/code/griffin_root/zephyr_base/static_objs/build
-- west build: running target run
[1/102] Preparing syscall dependency handling

[3/102] Generating include/generated/version.h
-- Zephyr version: 3.0.99 (/media/wsl/code/griffin_root/zephyr_base/zephyr), build: zephyr-v3.0.0-3163-g7ad2e604bb31
[97/102] Linking CXX executable zephyr/zephyr_pre0.elf

[101/102] Linking CXX executable zephyr/zephyr.elf

[101/102] cd /media/wsl/code/griffin_root/zephyr_base/static_objs/build && /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/zephyr.exe
Constructing Object A
Constructing Object B
Constructing Object C
Constructing Object D
*** Booting Zephyr OS build zephyr-v3.0.0-3163-g7ad2e604bb31  ***
[... 47 lines cut ...]
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
CMake Warning at /media/wsl/code/griffin_root/zephyr_base/zephyr/CMakeLists.txt:1803 (message):
  __ASSERT() statements are globally ENABLED

-- Configuring done
-- Generating done
[... cut ...]
[101/102] cd /media/wsl/code/griffin_root/zephyr_base/static_objs/build && /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/zephyr.exe
*** Booting Zephyr OS build zephyr-v3.0.0-3163-g7ad2e604bb31  ***

```
Exactly the same behavior I observed for `native_posix` and `native_posix_64`

Nevertheless current behavior, when constructors are not executed directly after start of the application, and after reverting of this commit https://github.com/zephyrproject-rtos/zephyr/commit/d10b46dc3be4492f5e7794c0a73c0e925d51c2ee is matching to this Enhancement https://github.com/zephyrproject-rtos/zephyr/issues/40444

palchak-google, 2022-04-28:
Commit https://github.com/zephyrproject-rtos/zephyr/commit/d10b46dc3be4492f5e7794c0a73c0e925d51c2ee addresses a defect in Zephyr reported in https://github.com/zephyrproject-rtos/zephyr/issues/39347. However, further investigation has revealed that https://github.com/zephyrproject-rtos/zephyr/commit/d10b46dc3be4492f5e7794c0a73c0e925d51c2ee does not correctly resolve the root cause of the issue. Additionally,  https://github.com/zephyrproject-rtos/zephyr/commit/d10b46dc3be4492f5e7794c0a73c0e925d51c2ee is currently masking a second, latent defect in Zephyr.

### Background
To understand the root cause of the issue, it's necessary to have familiarity with the start-up sequence for processes on Linux. This [article](http://dbp-consulting.com/tutorials/debugging/linuxProgramStartup.html) provides a great explanation of each of the steps in detail.

### Root cause of https://github.com/zephyrproject-rtos/zephyr/issues/36858
The root cause of https://github.com/zephyrproject-rtos/zephyr/issues/36858 lies in the interaction between the link script used by Zephyr for `native_posix` targets and the start-up sequence for glibc. To understand what's happening it helps to look at the map file for a `native_posix` binary. Here is the map for a binary built on my system using the example code from https://github.com/zephyrproject-rtos/zephyr/issues/36858. 

```
.init_array     0x000000000040bd90        0x0
                [!provide]                        PROVIDE (__init_array_start = .)
 *(SORT_BY_INIT_PRIORITY(.init_array.*) SORT_BY_INIT_PRIORITY(.ctors.*))
 *(SORT_BY_ALIGNMENT(.init_array) SORT_BY_ALIGNMENT(EXCLUDE_FILE(*crtend?.o *crtend.o *crtbegin?.o *crtbegin.o) .ctors))
                [!provide]                        PROVIDE (__init_array_end = .)

.ctors
 *crtbegin.o(SORT_BY_ALIGNMENT(.ctors))
 *crtbegin?.o(SORT_BY_ALIGNMENT(.ctors))
 *(SORT_BY_ALIGNMENT(EXCLUDE_FILE(*crtend?.o *crtend.o) .ctors))
 *(SORT_BY_NAME(SORT_BY_ALIGNMENT(.ctors.*)))
 *(SORT_BY_ALIGNMENT(.ctors))

ctors           0x00000000080532c8       0x14
                0x00000000080532c8                . = ALIGN (0x4)
                0x00000000080532c8                __CTOR_LIST__ = .
                0x00000000080532c8        0x4 LONG 0x3 (((__CTOR_END__ - __CTOR_LIST__) / 0x4) - 0x2)
 *(SORT_BY_NAME(SORT_BY_ALIGNMENT(.ctors*)))
 .ctors         0x00000000080532cc        0x4 /usr/lib/gcc/x86_64-linux-gnu/11/32/crtbegin.o
 .ctors         0x00000000080532d0        0x4 app/libapp.a(main.cc.obj)
 .ctors         0x00000000080532d4        0x4 /usr/lib/gcc/x86_64-linux-gnu/11/32/crtend.o
                0x00000000080532d8        0x4 LONG 0x0
                0x00000000080532dc                __CTOR_END__ = .

init_array      0x00000000080532dc        0x0
                0x00000000080532dc                . = ALIGN (0x4)
                0x00000000080532dc                __init_array_start = .
 *(SORT_BY_NAME(SORT_BY_ALIGNMENT(.init_array*)))
                0x00000000080532dc                __init_array_end = .

```
Note here that the `ctors` section ends up containing five quad words:
```
$ > readelf -x ctors zephyr.elf

Hex dump of section 'ctors':
  0x080532c8 03000000 ffffffff 90960408 00000000 ................
  0x080532d8 00000000                            ....
```
The first word (0x03) and the last word (0x0) are inserted by Zephyr's link script (the `QUAD` directives in the map file). The middle three words come from `crtbegin.o`, `main.cc.obj`, and `crtend.o`. The word from `crtbegin.o` is a value of -1, the word from `crtend.o` is 0x0, and the word from `main.cc.obj` is 0x08049690 which is the address of the initialization function that invokes the constructors for our static objects:
```
$> objdump -t zephyr.elf | grep "08049690" -  
08049690 l     F .text	000000c4              _GLOBAL__sub_I_zephyr_app_main
```
It's also important to note that there are *two* symbols in the binary, both named `__CTOR_LIST__`, that point into the `ctors` section:
```
$> objdump -t zephyr.elf | grep "CTOR_LIST" -
080532cc l     O ctors	00000000              __CTOR_LIST__
080532c8 g       ctors	00000000              __CTOR_LIST__
```
The first symbol refers to a local object (the 'l' and 'O' flags) at address 0x080532cc, which is the address of the value -1 that was contributed by `crtbegin.o`. The second symbol is a global symbol referring to the 0x03 inserted by Zephyr's link script.

The final piece of the puzzle lies in the `_init` function for this `native_posix` executable:
```
$> objdump -dj .init zephyr.elf 

zephyr.elf:     file format elf64-x86-64

Disassembly of section .init:

0000000000402000 <_init>:
[... 125 lines cut ...]
    [4] = 0x0 
}
```
The first entry indicates that the list contains three valid pointers, and the last entry is the trailing zero. However, entry[1], which is supposedly a function pointer, has the value -1. Similarly, entry[3] is a nullptr. Only the entry[2] is a valid function pointer. What Zephyr's link script has done is take a valid list (entries 1-3) and wrapped with with a leading size (0x03) and a trailing zero, resulting in a final list that contains invalid pointers. And indeed, if commit https://github.com/zephyrproject-rtos/zephyr/commit/d10b46dc3be4492f5e7794c0a73c0e925d51c2ee is reverted, Zephyr's version of `__do_global_ctors_aux` produces a segfault when it tries to deference the entry[1]:
```
$> ./zephyr.elf
Constructing Object A
Constructing Object B
Constructing Object C
Constructing Object D
[... cut ...]
In contrast, Zephyr's version of `__do_global_ctors_aux` references the second symbol, the global version. This symbol points to entry[0]. Thus when Zephyr's `__do_global_ctors_aux` attempts to dereference entry[1] as a function pointer a fault occurs.

Because the proposed fix for https://github.com/zephyrproject-rtos/zephyr/issues/39347 and https://github.com/zephyrproject-rtos/zephyr/issues/36858 will also eliminate this latent defect, it doesn't seem necessary to file a separate bug report for it. However, if others disagree, then please feel free to copy this coda into a stand-alone report.

palchak-google, 2022-06-08:
@aleksei-timofeyev 
@wojciechslenska 

I have submitted a fix for this issue. It works as best as I have been able to test. Will you please confirm the fix works properly on your system(s)?
https://github.com/zephyrproject-rtos/zephyr/pull/46357

closed 2022-06-09 by carlescufi

wojciechslenska, 2022-06-13:
@palchak-google 
Hello,
I can confirm that now constructors are working properly on Ubuntu 22.04 on GCC 10.3.0

```
wsl@T15 /m/w/c/g/z/static_objs (wsl/database_test)> west build -b native_posix -t run
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /media/wsl/code/griffin_root/zephyr_base/static_objs
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.4") found components: Interpreter 
-- Cache files will be written to: /home/wsl/.cache/zephyr
-- Zephyr version: 3.1.99 (/media/wsl/code/griffin_root/zephyr_base/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: native_posix
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /media/wsl/code/griffin_root/zephyr_base/zephyr/boards/posix/native_posix/native_posix.dts
-- Generated zephyr.dts: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/dts.cmake
Parsing /media/wsl/code/griffin_root/zephyr_base/zephyr/Kconfig
Loaded configuration '/media/wsl/code/griffin_root/zephyr_base/zephyr/boards/posix/native_posix/native_posix_defconfig'
Merged configuration '/media/wsl/code/griffin_root/zephyr_base/static_objs/prj.conf'
Configuration saved to '/media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/.config'
Kconfig header saved to '/media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
CMake Warning at /media/wsl/code/griffin_root/zephyr_base/zephyr/CMakeLists.txt:1809 (message):
  __ASSERT() statements are globally ENABLED

-- Configuring done
-- Generating done
-- Build files have been written to: /media/wsl/code/griffin_root/zephyr_base/static_objs/build
-- west build: running target run
[1/102] Preparing syscall dependency handling

[2/102] Generating include/generated/version.h
-- Zephyr version: 3.1.99 (/media/wsl/code/griffin_root/zephyr_base/zephyr), build: zephyr-v3.1.0-346-gb9e833366f13
[97/102] Linking CXX executable zephyr/zephyr_pre0.elf

[101/102] Linking CXX executable zephyr/zephyr.elf

[101/102] cd /media/wsl/code/griffin_root/zephyr_base/static_objs/build && /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/zephyr.exe
*** Booting Zephyr OS build zephyr-v3.1.0-346-gb9e833366f13  ***
Constructing Object A
Constructing Object B
Constructing Object C
Constructing Object D
```
and GCC 12.0.1
```
wsl@T15 /m/w/c/g/z/static_objs (wsl/database_test)> west build -b native_posix -t run
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /media/wsl/code/griffin_root/zephyr_base/static_objs
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.4") found components: Interpreter 
-- Cache files will be written to: /home/wsl/.cache/zephyr
-- Zephyr version: 3.1.99 (/media/wsl/code/griffin_root/zephyr_base/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: native_posix
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /media/wsl/code/griffin_root/zephyr_base/zephyr/boards/posix/native_posix/native_posix.dts
-- Generated zephyr.dts: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/dts.cmake
Parsing /media/wsl/code/griffin_root/zephyr_base/zephyr/Kconfig
Loaded configuration '/media/wsl/code/griffin_root/zephyr_base/zephyr/boards/posix/native_posix/native_posix_defconfig'
Merged configuration '/media/wsl/code/griffin_root/zephyr_base/static_objs/prj.conf'
Configuration saved to '/media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/.config'
Kconfig header saved to '/media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.0.1
-- The CXX compiler identification is GNU 12.0.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
CMake Warning at /media/wsl/code/griffin_root/zephyr_base/zephyr/CMakeLists.txt:1809 (message):
  __ASSERT() statements are globally ENABLED

-- Configuring done
-- Generating done
-- Build files have been written to: /media/wsl/code/griffin_root/zephyr_base/static_objs/build
-- west build: running target run
[1/102] Preparing syscall dependency handling

[2/102] Generating include/generated/version.h
-- Zephyr version: 3.1.99 (/media/wsl/code/griffin_root/zephyr_base/zephyr), build: zephyr-v3.1.0-346-gb9e833366f13
[97/102] Linking CXX executable zephyr/zephyr_pre0.elf

[101/102] Linking CXX executable zephyr/zephyr.elf

[101/102] cd /media/wsl/code/griffin_root/zephyr_base/static_objs/build && /media/wsl/code/griffin_root/zephyr_base/static_objs/build/zephyr/zephyr.exe
*** Booting Zephyr OS build zephyr-v3.1.0-346-gb9e833366f13  ***
Constructing Object A
Constructing Object B
Constructing Object C
Constructing Object D
```

I see the same behavior on `native_posix` and `native_posix_64`
Many thanks for your contribution

PR #46357 (2022-06-09): linker: ensure global constructors only run once
Rename the symbols used to denote the locations of the global
constructor lists and modify the Zephyr start-up code accordingly.
On POSIX systems this ensures that the native libc init code won't
find any constructors to run before Zephyr loads.

Fixes #39347
Fixes #36858
commit b4a7f0f2 (2022-06-09): linker: ensure global constructors only run once

Rename the symbols used to denote the locations of the global
constructor lists and modify the Zephyr start-up code accordingly.
On POSIX systems this ensures that the native libc init code won't
find any constructors to run before Zephyr loads.

Fixes #39347, #36858