zephyrproject-rtos/zephyr #39358: twister: add ability to select toolchain to be used at test/sample yaml level

PerMac, 2021-10-12:
**Is your enhancement proposal related to a problem? Please describe.**
In the CI for sdk-nrf we use gnuarmemb toolchain to build tests and samples. However, there is a need to test configurations using qemu_x86, which is not supported by the gnuarmemb toolchain. To run them in CI a separate workflow is needed, basically repeating the same process but with zephyr-sdk toolchain. It also causes problems with maintainance, since features like integration_platforms cannot be used for configurations requiring different toolchains. 

**Describe the solution you'd like**
It should be possible to modify twister to be able to switch toolchains between scenarios. An additional field, e.g. `toolchain_use: <toolchain>` can be used in sample/testcase.yaml which will tell which toolchain to use for a given scenario. Of course, to maintain current workflow, if no toolchian is defined, nothing will be changed and env var ZEPHYR_TOOLCHAIN_VARIANT will be used

tejlmand, 2021-10-13:
> An additional field, e.g. toolchain_use: <toolchain> can be used in sample/testcase.yaml which will tell which toolchain to use for a given scenario. Of course, to maintain current workflow, if no toolchian is defined, nothing will be changed and env var ZEPHYR_TOOLCHAIN_VARIANT will be used

I'm not sure i'm fully convinced here.
The `ZEPHYR_TOOLCHAIN_VARIANT=A` is set in environment which means a user is expecting `A` to be used.
And then suddenly another toolchain `B` is used behind the back of the user.

That sounds like something that's gonna be hard to debug and easy to create confusion.

> I believe a solution, which will allow twister to switch toolchains on the fly would make the above use case way simpler. 

simpler in your use-case, but harder for everyone else to understand and debug on why my `ZEPHYR_TOOLCHAIN_VARIANT` is no longer honored.

> twister checks if there is another toolchain on the system, which supports the given platform. 

and what if that toolchain is an outdated one because the user is not using it for daily development and thus updates it seldomly, or it was merely installed for evaluation/test purposes and then forgotten.

Users are expected to be able to run twister locally, twister is not only for CIs.

> However, there are tests in sdk-nrf repo, where developers want to use emulators not supported by gnuarmemb. In particular qemu_x86. It seems to be one of the more advanced emulators allowing to tests some higher functionality (e.g. networking).

So downstream test projects.
Why not tag those projects like:
```
tests:
  special.networking.test:
    tags: qemu_x86_zephyr
```
and then run twister in the CI like:
```
export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
./scripts/twister ---tag qemu_x86_zephyr
```
that ensures that twister with the zephyr toolchain is only executed for those particular tests.

And it avoids the risk of swapping / messing up toolchains inside `twister`.

PerMac, 2021-10-13:
I already explained rules behind integration mode. Maybe in other words: "with `--integration` all platforms in `integration_platforms` must result in either pass, failure or error". 
Your examples of split calls would give different results, you can try it by yourself. There are a couple of issues, probably due to the integration mode not being clear enough. You will get an error when calling `--integration` combined with any `--platform` as using `--platform` applies a filter causing any other platform to be skipped -> error hence "no skip on integration platforms in integration mode". The whole point of "integration_platforms" is to control CI scope on the test (in yaml) level and these two args exclude themselves. 
The same story will happen with other filters like tag/--tag-exclude or split --use-toolchains. Some tests have to be skipped in either of calls resulting in errors. 
You can try it by yourself:
Replace zephyr/samples/hello_world/sample.yaml content with:
```
sample:
  description: Hello World sample, the simplest Zephyr
    application
  name: hello world
common:
    harness: console
    harness_config:
      type: one_line
      regex:
        - "Hello World! (.*)"
tests:
  sample.basic.helloworld.zephyr:
    tags: zephyr_tool
    integration_platforms:
      - qemu_x86
  sample.basic.helloworld.gnuarm:
    tags: gnuarm_tool
    integration_platforms:
      - nrf52840dk_nrf52840
```
Do `export ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb` and run: `./scripts/twister -T samples/hello_world/ -G --tag gnuarm_tool`
Do  `export ZEPHYR_TOOLCHAIN_VARIANT=zephyr` and run: `./scripts/twister -T samples/hello_world/ -G --tag zephyr_tool`
Observe the errors. 

The `--use-toolchains` would give an order in which toolchains will be tried if the preceding on doesn't match test requirements. It won't mean "first run with TOOLCHAIN_A, then run with TOOLCHAIN_B"

Will sth like `allow-alternate-toolchain TOOLCHAIN` be more descriptive? Then ZEPHYR_TOOLCHAIN_VARIANT will take precedence but when a test is getting skipped due to the primary toolchain mismatch, the alternative one will be used.

tejlmand, 2021-10-13:
> @tejlmand all idea behind integration_platofrm is to avoid complicated process of choosing which test will be evaluated.

well, if the alternative is the complication proposed in this issue, then I vote for allowing `tags` and `toolchains`.
Else we are just re-introducing `complicated process of choosing which test will be evaluated` under a new name.

and if we allow the `alternative toolchain` scheme, then sooner or later some comes along and wants `alternative <place desired feature here>` scheme.

nashif, 2021-10-21:
isnt this a duplicare of #12997?