zephyrproject-rtos/zephyr #39376: Each CONFIG_BT_PER_ADV_SYNC_MAX requires ~11.5KB more RAM when CONFIG_BT_DF_CONNECTIONLESS_CTE_RX is enabled

jakkra, 2021-10-12:
**Describe the bug**
A lot RAM is required for Direction Finding IQ sampling making it unusable for more than just a few tags.

**To Reproduce**
1. Use samples/bluetooth/direction_finding_connectionless_rx
2. Add in prj.conf `CONFIG_BT_PER_ADV_SYNC_MAX=1`
3. `west build -b nrf52833dk_nrf52833`
4. Search for `mem_iq_report` in the generated map file and note the size
5. Change in prj.conf `CONFIG_BT_PER_ADV_SYNC_MAX=2` or anything larger
6. Study map file again (or check SRAM usage printed after compilation) and note ~11.5KB RAM needed for each added periodic sync.

**Expected behavior**
Expect less RAM to be required when gathering IQs. Expect being able to track 64 devices (CONFIG_BT_PER_ADV_SYNC_MAX max value allowed and still have some RAM left for application).

**Impact**
Showstopper for anything but very simple Bluetooth Direction Finding applications.

**Environment:**
 - OS: Windows
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: `5a661012407e3588447ad343644589aac673d60c (HEAD -> main, origin/main, origin/HEAD)`

Looking at the code you are today allocating buffers for pretty much the worst case, I guess that makes it simple and there is no need to decide when to throw away IQs etc. but doing it this way is way to limiting for the application.

ppryga, 2021-11-02:
> Is there a plan to optimize this or do you think this is how it will be for a while/forever? It's still about 1KB more RAM required to support each more transmitter. For us having a pool of RAM instead and allowing to support more devices would work much better.

Current implementation is configurable, at least to some extent, in my humble opinion acceptable extent. The solution is tailored to extended and periodic advertising design. It is based on memory pool. Amount of data allocated in the pool is connected with maximum number of periodic scanners and maximum number of CTEs allowed for a single periodic advertising chain.

> The solution we are looking for/need is something a bit more dynamic, like assigning the controller IQ buffers the size of MAX(calculation_you_have_now, customer_config) and allowing controller to throw away IQ data if the buffers are filled etc. and based on amount of free buffers support CONFIG_BT_CTLR_DF_PER_SCAN_CTE_NUM_MAX>1 without consuming a lot of extra RAM. However this introduces problems that we may "starve" some synced advertiser which is sending less CTEs than others. So logic to handle this would be needed (like dynamically changing amount of buffered IQs allowed for each synced device etc).

I hope I've understood you correctly. You want to limit number of reported IQ samples by configuration. That is possible but it will not comply with Bluetooth Core spec. The controller tests specification validates number of IQ samples reported by controller (see: Link Layer (LL) Test Suite 4.2.3.6 AoD Connectionless CTE Scanning). Also pay attention that in current implementation IQ reports are stored in a common memory pool. They are not fixed for periodic advertisement scanners.
E.g. If you have two per. advertisements with different intervals, then scanned for the one that has shorter interval will request node_rx_iq_report more often and use more memory from the pool. At the end of the day some other scanner may be starved also. This is something I'm going to handle in a near future.

> Another thing that would be very useful is if application could "share" this mem_iq_report buffer and let application free the reports after procssing them, right now application would also need its own buffer of IQs, and when this application buffer is full application would need to throw away the IQ reports generated by controller anyway. I guess now they are freed after the IQ report events has been sent to application? I'm fully aware that this is not possible since you implement the HCI interface and nothing else, also may not work for NRF5340 BT architecture, so something like this would require additional APIs which I guess you are not interested in Zephyr, meaning this is something we would have to implement ourselves.

The controller is separated from host (and application) by HCI interface. There is no such implementation in the controller, that allows to give an ownership of the memory allocated by controller to application. This will not work for any use case where the controller is not part of the same memory space as host (so nRF52 only for builds where host and controller are part of the same binary). Besides that, there is a separation of concerns. In case this ownership hand down had been implemented the controller would be vulnerable for resources starvation that it couldn't control at all.
Also it would include major change into net_buf handling (if possible at all) or separate implementation for this particular case, that would not comply with current interface. So changes in controller and host would be required. 

In current implementation the IQ samples are copied once into net_buf in HCI layer. Then there is no additional storage, mem copying etc. up to the application callback execution. It depends on the application what is done later.

jakkra, 2021-11-05:
Hi, thank you for the input.

> At the end of the day some other scanner may be starved also. This is something I'm going to handle in a near future.

_Great, looking forward to this._

 I understand your points and the limitations you have due to Zephyr architecture and the BT SIG specificaton. Still the current solution will not work for us. 

**But** let me give one example before we close this ticket.

**Let's take this example:**

- direction_finding_connectionless_rx
- CONFIG_BT_PER_ADV_SYNC_MAX=64
- CONFIG_BT_CTLR_DF_PER_SCAN_CTE_NUM_MAX=1

This will consume 101kB RAM in total
| Name          | RAM usage |
|---------------|-----------|
| mem_iq_report | 46kB      |
| mem_pdu_rx    | 20kB      |
| ll_sync_pool  | 11kB      |

From what I see the IQ samples will be placed in the `mem_iq_report` from the moment the radio puts it there (actually reserved before radio is started, but this time is short), until application receives it in `cte_report_cb`. The time it takes from radio fills the buffer to application receives it I measured is most of the time 1ms, sometimes 2ms, so let's say worst case 2ms.
I tried doing some simulation in order to show what I mean (using above config). 

Let's assume we have control over tag periodic advertising interval and 64 tags. I do the assumtion to simply that each tag starts advertising after a random time, if all tags are advertising at the same time of course buffers would be filled more, but that is very unlikely to happen in real life.
[image]
[image]
[image]

It's a pretty naive simulation, but I still think it gives the "ballpark" what to expect for a real life use-case. The graph shows the worst case mem_pdu_rx buffer usage each "second" during one "hour".

This is why I think the buffer size allocated today based on `CONFIG_BT_PER_ADV_SYNC_MAX `and `CONFIG_BT_CTLR_DF_PER_SCAN_CTE_NUM_MAX` is wasting a lot of RAM. 

Without looking much into how the `mem_pdu_rx` pool works, I guess the above reasoning also is applicable for that.

I would test this in real to confirm but cannot until https://github.com/zephyrproject-rtos/zephyr/issues/39438 is fixed.

ppryga, 2021-11-08:
@jakkra thank for broadening the perspective.
With this input I think there is a space for improvement. To do that I need to get some numbers. It would be very appreciated to get similar results for `mem_iq_report`. 

Would it be also possible to get some measurements for latency periodic advertising are synchronized (time from receiving extended advertising to time when periodic advertising sync_cb is executed? It would give some input how many devices may be "tracked" in a given period of time. 

I've provided a response to #39438. It shouldn't be a blocker anymore.

jakkra, 2021-11-08:
Thanks @ppryga, I'll try to setup a real test later this week with 10-20 tags/advertisers and collect some data for RAM usage, assuming I manage to get it working at all with that many tags.

jakkra, 2021-11-12:
So far I have some data for around 8-10 tags with 25ms and 100ms (no chain) and I will summarize it more next week, but so far this is what I have: 
25ms => worst case 4 IQ buffers, ~90% of the time 1 IQ buffer used.
100ms => worst case 2 IQ buffers, ~95% if the tune 1 IQ buffer used.

I will collect more data next week and also do this for `mem_pdu_rx` pool. I have attached the code I use. 
[iq_buf_usage.zip](https://github.com/zephyrproject-rtos/zephyr/files/7529296/iq_buf_usage.zip)
Instructions to use this SW can be found in this issue: https://github.com/zephyrproject-rtos/zephyr/issues/40317

Also apply this patch in zephyr to get the stats. Please provide input if this is the wrong/there is a better way to find the usages.
[iq_buf_usage.txt](https://github.com/zephyrproject-rtos/zephyr/files/7529320/iq_buf_usage.txt)

jakkra, 2021-11-16:
Ok so I ran over the night with 10 tags. All tags randomly restarts advertising every 15 minutes+ small random delay minutes to avoid accidental synchronization. Data is taken each second, and consists of the most simultaniously used IQ buffers during the last second. Tags have set min and max periodic adv interval to 25ms (40 IQ samples/s). Since the receiver crashes often I have discarded samples where <7 advertisers are tracked. 
[image]

So looking at the data for this case at least 4 IQ buffers/synced tag is enough to report 99.8% of the IQ samples to the application.

Unfortunately I did not quickly find a good place to keep count of the `mem_pdu_rx` pool usage. But if you can point me where to get the data I'll do that also.

jakkra, 2022-02-24:
Hi, thanks for all the fixes and CTE gathering using periodic adv. is now working. We still have the issues described here: https://github.com/zephyrproject-rtos/zephyr/issues/40317#issuecomment-1014660743 which according to you guys is memory related, hence related to this ticket. Is there any progress or plan for the RAM optimizations?

Thank you

ppryga, 2022-02-24:
@jakkra regarding the memory footprint optimization. There is a plan to do it, but I can't give you timeline because it depends on other things. Anyway it is on my tasks list to be implemented.

jakkra, 2022-06-30:
We are now at the point where we are really hitting this constraint of "unoptimized" RAM usage for IQs. Do you have any update? Thank you.

carlescufi, 2022-06-30:
> We are now at the point where we are really hitting this constraint of "unoptimized" RAM usage for IQs. Do you have any update? Thank you.

@ppryga do you have plans to work on this?

ppryga, 2022-07-01:
This is on my ToDo list as of now. If anything changes I'll inform you and @cvinayak.

ppryga-nordic, 2022-08-18:
This PR: https://github.com/zephyrproject-rtos/zephyr/pull/48936 provides an implementation of reporting that Controller was unable to store all IQ samples and missed or will miss sampling of a CTE.
This will prevent assertions in lll_df.c if there are no free `node_rx_iq_report`.

The implementation is done in lower link layer of periodic advertising synchronization, hence it is aimed for reception of a CTE in a chained PDUs that are very close to each other (used LLL scheduling to reception of chained PDUs). This helps to reduce number of required `node_rx_iq_report` for reception of CTE with periodic advertising.

There is missing implementation for ULL scheduling, for chains of PDUs where time between PDUs is enough to use ticker for start of reception of next PDU.

jakkra, 2022-08-23:
Hi, I'm not able to test this since there seems to be either a new behaviour or bug on main branch that after a sync termination then no new advertisments with periodic adv. are reported in the `bt_le_scan_cb` `recv` callback. 
Is this some new baheviour I need to update the application to handle or should I file an issue?

cvinayak, 2022-08-23:
@jakkra do you have `BT_LE_SCAN_OPT_FILTER_DUPLICATE` set in the scan parameters? Could you check if removing this option helps in finding advertisements?

jakkra, 2022-08-23:
@cvinayak No I don't have `BT_LE_SCAN_OPT_FILTER_DUPLICATE ` set. Also powering on new tags with not seen MAC addresses does not show either, so it's not that tha tag is filtered out.

cvinayak, 2022-08-23:
@jakkra Do you see the issue with latest upstream/main `periodic_adv` and `periodic_sync` sample (i.e. with the changes in #48936) ? if so, then I request you to kindly create a github issue with detailed steps to reproduce the issue and assign it to me.

jakkra, 2022-08-23:
@cvinayak I'm using our own application which is working on nRF Connect 2.0 and now I run it with https://github.com/zephyrproject-rtos/zephyr/pull/48936 and it's not working, I also tried reverting the commit in https://github.com/zephyrproject-rtos/zephyr/pull/48936 and still have the same issue. 
I'll try to reproduce with modified samples and open a new issue.

cvinayak, 2022-08-23:
> I'll try to reproduce with modified samples and open a new issue.

@jakkra Thank you, appreciate your help. We have been successfully executing the conformance tests and have not encounter any such bugs, may be the observed issue is not exercised in these tests or the BabbleSim simulation tests that run on every PR submitted.

jakkra, 2022-08-29:
Hi @ppryga-nordic @cvinayak  I have now tested https://github.com/zephyrproject-rtos/zephyr/pull/48936 together with https://github.com/zephyrproject-rtos/zephyr/pull/49579 and it works. Each new `CONFIG_BT_PER_ADV_SYNC_MAX` results in about 1.3kB RAM, which is still a bit much as we'd like to have it at least 50. 

I have 45 tags to test with.
Some observations:
1. With per. adv. interval of 50ms and NO chained PDUs I'm max able to sync to around 30/45 tags, I would assume having chained PDUs would decreas the 30 to something less, as the radio will be more busy.
2. With per. adv. interval of 250ms and NO chained PDUs I'm able to sync to somewhere around ~40/45 tags (see image here https://github.com/zephyrproject-rtos/zephyr/issues/49465#issuecomment-1228631737) 
3.  With per. adv. interval of 50ms and NO chained PDUs I'm getting around <0-9 CTE's (often<4) per second and not 20, I guess radio misses/skips some.
4. None of the aboce configurations report `BT_DF_CTE_INSUFFICIENT_RESOURCES`. To make sure sync slots are filled I also tested with CONFIG_BT_PER_ADV_SYNC_MAX to 30, interval of 50ms and NO chained PDUs  and run with 30 (all synced) tags I'm able to sync with all of them, however at no point I receive  `BT_DF_CTE_INSUFFICIENT_RESOURCES`.

Conclusions:
- From 1. I draw the conclusion that due to the radio no matter how many tags are active only a certaion amount of CTEs are really captureable since you are not able to sync to more than let's say 30 tags anyway and together with point 3. not all per. adv. packets are found by the radio anyway, according to point 3. less than 50% (mostly less than 30%) of per. adv. packets are seen and CTE is sampled by the radio in that scenario.

- The number of RAM required per CONFIG_BT_PER_ADV_SYNC_MAX can be decreased or reworked. It does not have to be linear with CONFIG_BT_PER_ADV_SYNC_MAX as for example CONFIG_BT_PER_ADV_SYNC_MAX=100 and tags with 50ms adv interval, we will never be able to sync to all 100, max is around 30 for that setup. 

Questions:
- Since the controller now can handle when running out of IQ buffers, I should be able to make this POOL_SIZE smaller. Where is that size defined, that is here right: https://github.com/zephyrproject-rtos/zephyr/blob/3e9dfdd85099a63a769bb1a8381354a5e1f39e06/subsys/bluetooth/controller/ll_sw/ull_df_types.h#L31 
- https://github.com/zephyrproject-rtos/zephyr/blob/3e9dfdd85099a63a769bb1a8381354a5e1f39e06/subsys/bluetooth/controller/ll_sw/ull_df_types.h#L31  that pool is not shared across tags right, at least it seems as each tag "reserves one". Set 30 tags to 1s per. adv. interval, randomly started. I set SYNC_IQ_REPORT_CNT=10 but I'm getting alot of `BT_DF_CTE_INSUFFICIENT_RESOURCES`, I guess they are reserved so that some tags doesn't starve others? If I want to experiment and LET tags be starved, is there an easy way I could test that? I want to have a pool of X size and let all sync share that one.

New issues found:
- When syncing with more than about 5 tags with per.adv interval of about 10ms then after a while (not long, like 10s) no events are generated from the controller (no scan rsp, no ext. scan rsp, no sync term, no cte etc.).

ppryga-nordic, 2022-08-29:
@jakkra I'll start with information that @cvinayak is now responsible for Direction Finding in Zephyr Controller.

Regarding #48936 and memory usage.  The PR should allow to decrease amount of memory allocated for DF by Controller. I hadn't got time to implement this changes. You can try to do that and provide a PR,  that would be very welcome. Current implementation still allocates a number of node_rx_iq_report that is equal to 2 times max per. adv. chain length.

ppryga-nordic, 2025-09-17:
@cvinayak  I've assigned the issue to you. I have not capacity to work on it in foreseeable future.