zephyrproject-rtos/zephyr #39395: STM32L496 uart rx interrupt mode not working

glin00, 2021-10-13:
**Describe the bug**
I had problem that on my STM32L496 uart rx interrupt mode is not working. 
I tested both the uart tx and rx polling mode and both are working. 

**Expected behavior**
uart rx interrupt should work.

**Impact**
Although uart rx polling mode is working for me, but it is not efficient comparing to interrupt mode.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

glin00, 2021-10-19:
after I change my code from usart3 to usart2 in dts file, the above interrupt mode program can start to work. So, I will try to use usart2 in the meantime.  
But if I want to use usart3 for interrupt mode later on, do you have any advises how to debug this problem?

erwango, 2021-10-19:
From what I can see hardware blocks USART3 and USART2 are strictly equivalent.
From software point of view, they will also share the same instantiated code.
The only difference could be the device tree description (but it seems correct), or the pin assignation.
But if you say that USART3 is working in polling mode then I guess that pin and dts description are also correct.
So I'm not sure what could be wrong here.

erwango, 2021-10-19:
> how to debug this problem?

Advice would be to monitor USART3 registers
See if properly configured and then monitor USART_ISR

erwango, 2021-11-03:
@glin00 is this issue still valid ?

glin00, 2021-11-04:
I will come back to this issue later since I need to focus on other higher priority issues first.

closed 2022-01-18 by github-actions[bot]