zephyrproject-rtos/zephyr #39426: USB console stalls after host reboot

lmaciejonczyk, 2021-10-14:
The usb console stalls after host reboot. The input is not working. 

**To Reproduce**
Steps to reproduce the behavior:
1. build and flash the shell module sample with USB support:
west build -b nrf52840dk_nrf52840 zephyr/samples/subsys/shell/shell_module/  -p always -- -DOVERLAY_CONFIG='overlay-usb.conf' -DDTC_OVERLAY_FILE=usb.overlay && west flash --erase
2. reboot host (in my case Ubuntu 20.04 LTS)
3. sudo minicom -b 115200 -D /dev/ttyACM1
4. See error:

[00:00:58.057,800] <inf> usb_cdc_acm: Device suspended
[00:00:58.306,213] <inf> usb_cdc_acm: Device resumed
[00:00:58.326,263] <inf> usb_cdc_acm: Device suspended
[00:00:58.509,155] <inf> usb_cdc_acm: Device resumed
[00:00:58.526,763] <inf> usb_cdc_acm: Device suspended
[00:00:58.938,842] <inf> usb_cdc_acm: Device resumed
[00:00:59.182,800] <wrn> usb_device: Endpoint 0x81 already enabled
[00:00:59.182,800] <wrn> usb_device: Endpoint 0x82 already enabled
[00:00:59.182,830] <wrn> usb_device: Endpoint 0x01 already enabled
[00:00:59.182,830] <inf> usb_cdc_acm: Device configured
uart:~$

**Expected behavior**
I expect to see input data printed on the terminal after I provide it with my keyboard on the host side.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04 LTS, Windows 10
 - Toolchain: zephyr-sdk-0.13.1
 - zephyr: zephyr/main (9e97b3b0f0)

jfischer-no, 2021-11-19:
Can not reproduce it,
`[00:00:59.182,830] usb_device: Endpoint 0x01 already enabled` is warning not an error.

carlescufi, 2022-01-26:
No feedback from reporter, closing this. Please reopen if this is an issue.

closed 2022-01-26 by carlescufi