zephyrproject-rtos/zephyr #39432: Periodic adv. syncing takes longer and bt_le_per_adv_sync_delete returns error after commit ecf761b4e9

jakkra, 2021-10-14:
**Describe the bug**
There is a change in behaviour in my DF test application that I'm not sure is expected. Using `git bisect` I tracked it down to commit ecf761b4e9.

What my application does is to start a timeout after calling `bt_le_per_adv_sync_create` that is 2 seconds. Before this commit I almost never hit that timeout (it's cancelled when sync is done), but after it every other sync hits it. What I also noticed is that when I try to cancel the sync using `bt_le_per_adv_sync_delete` after my timeout hits I get error back (BT_HCI_ERR_CMD_DISALLOWED (-5)) checking the code it seems as this is the reason for the ERROR:
`scan->per_scan.sync` is `NULL` [here in ull_sync.c hence it fails](https://github.com/zephyrproject-rtos/zephyr/blob/9e97b3b0f0913a46b90555e797e979dd7344ce3b/subsys/bluetooth/controller/ll_sw/ull_sync.c#L228)
Looking at the log when this happens:
```
*** Booting Zephyr OS build v2.7.99-403-gecf761b4e986  ***
Starting Connectionless Locator Demo
Bluetooth initialization...success
Scan callbacks register...success.
Periodic Advertising callbacks register...success.
Start scanning...success
Waiting for periodic advertising...
Scan is running...
Found device sending per adv: 0A:76:7A:EB:55:33 (random)
(2900) Creating Periodic Advertising Sync...
success.
Sync index: 0
(4906) sync_create_timeout for 0A:76:7A:EB:55:33 (random)
Failed cancel sync: -5
(5186) PER_ADV_SYNC[0]: [DEVICE]: 0A:76:7A:EB:55:33 (random) synced, Interval 0x0780 (2400 ms), PHY LE 2M
Enable receiving of CTE...
success. CTE receive enabled.
(5201) PER_ADV_SYNC[0]:
(5203) PER_ADV_SYNC[0]:
(5206) PER_ADV_SYNC[0]:
(5208) PER_ADV_SYNC[0]:
(5210) PER_ADV_SYNC[0]:
(5213) PER_ADV_SYNC[0]:
(5215) PER_ADV_SYNC[0]:
(5218) PER_ADV_SYNC[0]:
(5220) PER_ADV_SYNC[0]:
(5222) PER_ADV_SYNC[0]:
(7587) PER_ADV_SYNC[0]:
(7589) CTE[0]: samples count 45
(7592) PER_ADV_SYNC[0]:
(7595) CTE[0]: samples count 45
...
```
- We can see that sync is created at 2900 (ms)
- At 4906 (ms) the timeout hits and I call `bt_le_per_adv_sync_delete`, **here I expect the cancel to not return failure**.
- At 5186 (ms), 280ms later I get the `synced` callback. **I don't expect this.**

**To Reproduce**
Steps to reproduce the behavior:
1. Use latest Zephyr direction_finding_connectionless_tx on one device
2. Run the modfied direction_finding_connectionless_rx attached to this ticket.
3. You just need two devices, have the tag always transmitting and only reset the direction_finding_connectionless_rx device.
4. Reset the direction_finding_connectionless_rx device until you get the error in the log

**Expected behavior**
I expect no error response from `bt_le_per_adv_sync_delete`

**Impact**
Application ends up in a state I'm not sure how to handle.

**Logs and console output**
Logs for different scenarios and commits.
[log_bad_9e97b3b0f0.txt](https://github.com/zephyrproject-rtos/zephyr/files/7345070/log_bad_9e97b3b0f0.txt)
[log_bad_9e97b3b0f0_no_sync_timeout.txt](https://github.com/zephyrproject-rtos/zephyr/files/7345071/log_bad_9e97b3b0f0_no_sync_timeout.txt)
[log_bad_ecf761b4e9_breaking_commit.txt](https://github.com/zephyrproject-rtos/zephyr/files/7345072/log_bad_ecf761b4e9_breaking_commit.txt)
[log_good_9b66a8b47d.txt](https://github.com/zephyrproject-rtos/zephyr/files/7345073/log_good_9b66a8b47d.txt)

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK
 - Commit SHA or Version used: 9e97b3b0f0

**Additional context**
Application to reproduce: 
[direction_finding_connectionless_rx.zip](https://github.com/zephyrproject-rtos/zephyr/files/7345387/direction_finding_connectionless_rx.zip)

cvinayak, 2021-10-15:
The periodic advertising interval is 2.4 seconds, periodic sync is established on the first successful reception of AUX_SYNC_IND, i.e. there can be upto 2.4 seconds until sync is established, 2 second timeout is short. Please close this issue, if a timeout > 2.4 seconds solves your issue.

jakkra, 2021-10-18:
Hi @cvinayak, I bumped the timeout as promised and as expected It's not getting hit anymore, since it got more time, however the issue is still there. For example in the log below I reset the transmitter as soon as I see the `Creating Periodic Advertising Sync...success.` in the log. This simulates for example a tag going out of range just after we initialize the sync or similar. As you can see expected behaviour happend once first, and then next time I got -5 error code. I also added in to retry to call `bt_le_per_adv_sync_delete` if it failed but it just keeps reporting error. The controller don't allow any new sync to be created as it's syncing with some device (which I cannot stop), to recover I need to reset the module.
```

*** Booting Zephyr OS build v2.7.99-571-gecf6e22b3ab4  ***
Starting Connectionless Locator Demo
Bluetooth initialization...success
Scan callbacks register...success.
Periodic Advertising callbacks register...success.
Start scanning...success
Waiting for perioScan is running...
dic advertising...
(1216) Found device sending per adv: 1F:63:2C:36:35:FF (random)
(1221) Creating Periodic Advertising Sync...
success.
Sync index: 0 <---- Here I reset the tag
(5227) sync_create_timeout for 1F:63:2C:36:35:FF (random)
Sync deleted
Scan is running...
(11684) Found device sending per adv: 05:CD:F4:2C:5E:E8 (random)
(11690) Creating Periodic Advertising Sync...
success.
Sync index: 0 <---- Here I reset the tag
(15697) sync_create_timeout for 05:CD:F4:2C:5E:E8 (random)
Failed cancel sync: -5
Scan is running...
(21582) Found device sending per adv: 18:4A:2C:71:E0:42 (random)
(21588) Creating Periodic Advertising Sync...
Sync for some device already in progress.
failed (err -16)
(22348) Found device sending per adv: 18:4A:2C:71:E0:42 (random)
(22354) Creating Periodic Advertising Sync...
Sync for some device already in progress.
failed (err -16)
(23004) Found device sending per adv: 18:4A:2C:71:E0:42 (random)
(23010) Creating Periodic Advertising Sync...
Sync for some device already in progress.
failed (err -16)
...
continues forever to fail
...
```

cvinayak, 2021-10-19:
> Maybe there was a bug before ecf761b that sent the synced callback too early maybe?

Yes, it was a bug, to send Sync Established before the reception of first AUX_SYNC_IND.

> \> 2.4s

This needs investigation, probably there wasn't a periodic adverting event every 2.4 seconds.

cvinayak, 2022-01-06:
> Yes, it was a bug, to send Sync Established before the reception of first AUX_SYNC_IND.

This fixed in upstream/main  and latest periodic adv/sync changes are available in #41091

closed 2022-03-02 by carlescufi