zephyrproject-rtos/zephyr #39483: LSM6DS0 Gyroscope rad/s Calculation Error

stuianna, 2021-10-16:
**Describe the bug**
It appears the LSM6DS0 driver uses an incorrect constant when calculating the rad/s angular velocity for its gyroscope.

The following calculation is used:
```c
// drivers/sensor/lsm6ds0/lsm6ds0.c
static inline void lsm6ds0_gyro_convert(struct sensor_value *val, int raw_val,
          float numerator)
{
  double dval;

  dval = (double)(raw_val) * numerator / 1000.0 * SENSOR_DEG2RAD_DOUBLE;
  val->val1 = (int32_t)dval;
  val->val2 = ((int32_t)(dval * 1000000)) % 1000000;
}
``` 

With the `SENSOR_DEG2RAD_DOUBLE` constant defined as:
```c
// drivers/sensor/lsm6ds0/lsm6ds0.h
#define SENSOR_DEG2RAD_DOUBLE (180.0 / SENSOR_PI_DOUBLE)
```
Resulting in an incorrect value (in deg^2 / rad).

The correct value can be obtained by modifying the macro definition to:
```c
// drivers/sensor/lsm6ds0/lsm6ds0.h
#define SENSOR_DEG2RAD_DOUBLE (SENSOR_PI_DOUBLE / 180.0 )
```

**To Reproduce**
The `samples/shields/x_nucleo_iks01a1` project can be used. However an additional `sensor_sample_fetch` needs to be added for the lsm6ds0 gyroscope, as it is currently not included in the example.

**Expected behavior**
When the lsm6ds0 is near stationary, the measured angular velocity should be close to 0. Currently the readings for each axis are floating up to +- 100 rad/s.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain :  gcc-arm-none-eabi 1.80201.181220 (8.2.1).
 - Commit SHA or Version used: Zephyr 2.20600.210721 (2.6.0) ,

avisconti, 2021-11-02:
@stuianna Please take a look to the fix. I missed this bug until now, sorry.

closed 2021-11-03 by MaureenHelm

stuianna, 2021-11-09:
@avisconti Works as expected. Sorry I couldn't get back to you earlier.

Thanks.

PR #39986 (2021-11-03): drivers/sensor: lsm6ds0: Fix SENSOR_DEG2RAD_DOUBLE macro
In order to convert from degrees to radiants the SENSOR_DEG2RAD_DOUBLE
macro must divide the argument by 180 degrees and multiply by PI
radiants, and not the other way around. Please note that same macro
is already defined in the correct way in other sensor drivers as well.
(Fix #39483)
commit aa3f3d94 (2021-11-03): drivers/sensor: lsm6ds0: Fix SENSOR_DEG2RAD_DOUBLE macro

In order to convert from degrees to radiants the SENSOR_DEG2RAD_DOUBLE
macro must divide the argument by 180 degrees and multiply by PI
radiants, and not the other way around. Please note that same macro
is already defined in the correct way in other sensor drivers as well.
(Fix #39483)