zephyrproject-rtos/zephyr #39487: esp32 IRQ01 stack utilisation is 100%

pavlohamov, 2021-10-17:
**Describe the bug**
IRQ stack usage is 100% seems to indicate some memory corruption.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/subsys/shell/fs; mkdir build; cd build
2. cmake -GNinja -DBOARD=esp32 .. && ninja
3. ninja flash
4. execute "kernel stacks " in shell

**Expected behavior**
IRQ01 stack usage is smaller then 100%

**Impact**
Seems to be memory corruption that causes other applications to fail (willing to create sample app. Please see Additional context)

**Logs and console output**
```
kernel stacks 
0x3ffb51a0 sysworkq   (real size 1024):	unused 716	usage 308 / 1024 (30 %)
0x3ffb4f38 shell_uart (real size 2048):	unused 636	usage 1412 / 2048 (68 %)
0x3ffb4e68 logging    (real size 1024):	unused 748	usage 276 / 1024 (26 %)
0x3ffb5068 idle 01    (real size 1024):	unused 940	usage 84 / 1024 (8 %)
0x3ffb4fd0 idle 00    (real size 1024):	unused 828	usage 196 / 1024 (19 %)
0x3ffe5f50 IRQ 00     (real size 2048):	unused 1824	usage 224 / 2048 (10 %)
0x3ffe6750 IRQ 01     (real size 2048):	unused 0	usage 2048 / 2048 (100 %)
```

**Environment (please complete the following information):**
 - Linux 5.11.0-37-generic 20.04.2-Ubuntu SMP x86_64/macOS 11.6 (20G165)
 - espressif
 - b7063164916e9e9359a58b871f1cce12cf1debfb

**Additional context**
My own application (esp32/esp32s2) fails for various reasons (load/store prohibited/alignment). Same app build for cc3220/native_posix - works fine. Suppose this is Memory corruption is root case 
```
[00:00:09.391,000] <err> os:  ** FATAL EXCEPTION
[00:00:09.394,000] <err> os:  ** CPU 0 EXCCAUSE 9 (load/store alignment)
[00:00:09.404,000] <err> os:  **  PC 0x4008bca4 VADDR 0x4012cea2
[00:00:09.412,000] <err> os:  **  PS 0x50520
[00:00:09.420,000] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:5 CALLINC:1)
[00:00:09.431,000] <err> os:  **  A0 0x4008bcd0  SP 0x3ffea290  A2 0x4012cea2  A3 0x4012ceb2
[00:00:09.442,000] <err> os:  **  A4 0x4008bd64  A5 0x3ffea26c  A6 0x3ffbfef0  A7 0xf
[00:00:09.453,000] <err> os:  **  A8 (nil)  A9 0x7105b7e7 A10 0x710748bb A11 (nil)
[00:00:09.463,000] <err> os:  ** A12 0x8008cce0 A13 0x3ffea24c A14 0x865a21f0 A15 0x8
[00:00:09.474,000] <err> os:  ** LBEG 0x40080db5 LEND 0x40080dbd LCOUNT 0x26
[00:00:09.484,000] <err> os:  ** SAR 0x9

Backtrace:0x4008BCA1:0x3FFEA290 0x4008BCCD:0x0000000C  |<-CORRUPTED

[00:00:09.508,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:09.518,000] <err> os: Current thread: 0x3ffbfef0 (idle 00)
[00:00:09.527,000] <err> os: Halting system
```

backtrace point to 
```
(gdb) b *0x4008BCCD
Breakpoint 1 at 0x4008bccd: file zephyr/arch/xtensa/core/xtensa-asm2-util.S, line 182.
(gdb) b *0x4008BCA1
Breakpoint 2 at 0x4008bca1: file zephyr/arch/xtensa/core/xtensa-asm2-util.S, line 104.
```

pavlohamov, 2021-10-17:
@glaubermaroto @sylvioalves @uLipe Could you be kind to advise?

sylvioalves, 2021-10-17:
@pavlohamov CPU1 is not supported in ESP32. It means that IRQ01 should not be there. The reason for that is the line below:  
https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/xtensa/esp32/Kconfig.defconfig#L15  

It needs to be changed to 1 core. Then, no more 100% stack usage for ESP32:  
```c
0x3ffb0790 sysworkq   (real size 1024):	unused 716	usage 308 / 1024 (30 %)
0x3ffb05c0 shell_uart (real size 2048):	unused 604	usage 1444 / 2048 (70 %)
0x3ffb0658 idle 00    (real size 1024):	unused 828	usage 196 / 1024 (19 %)
0x3ffe5750 IRQ 00     (real size 4096):	unused 3936	usage 160 / 4096 (3 %)
```