zephyrproject-rtos/zephyr #39491: Add a hal module for Nuclei RISC-V core (NMSIS)

soburi, 2021-10-17:
Applying the results of the discussions in #34971 and #38657, I propose to add an NMSIS module.
This module is different from proposed in #34971. Reissue this PR.

## Origin

NMSIS
https://github.com/Nuclei-Software/NMSIS/

## Purpose

Add support for Nuclei RISC-V IP core required by GigaDevice's GD32VF103 SoC.

## Mode of integration

As External Module.

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/34970

(Cooperating with #38661 by @nandojve and #36833 by @feilongfl to add support GigaDevice SoCs.)

## Description

It is a BSP that is needed to support Nuclei's RISC-V core-based SoCs.
I sent PR #34970 to support GigaDevice GD32V SoCs.
This package referenced from that PR. 

Previously I sent #34971 to propose to add hal_nuclei.
#34971 contains GigaDevice's and Nuclei's sources.
#38727 already introduce GigaDevice's SoC drivers, So this PR is adding a Nuclei only module. 

# Dependencies

https://github.com/zephyrproject-rtos/hal_gigadevice

## Revision

https://github.com/soburi/NMSIS
bfb0709d919e9fb3d96a8c75b5d04500943b7240

[scancode.html.gz](https://github.com/zephyrproject-rtos/zephyr/files/7361084/scancode.html.gz)

## License

Apache-2.0 and BSD-3-clauses

- - - 

Nuclei System Technology provides RISC-V IP core for SoC vendors.
GigaDevice is a Silicon Vendor, one of the customers of Nuclei System Technology.
GD32VF103 is an SoC product of GigaDevice, the chip implementing Nuclie's RISC-V IP core.

NMSIS is basic definitions and utilities for Nuclei's RISC-V IP core, includes below.
- Register definitions (RISC-V standard and Nuclie's vendor extensions)
- Interrupt controllers utilities
- Timer utilities
- Memory protection utilities
- etc

NMSIS inspire by ARM's CMSIS module.
It has a similar role and structure. 

#38657 introduce a module to support GigaDevice's SoCs, including GD32VF103 implements Nuclei's RISC-V IP core.

GD32VF103 construct from GigaDevice's peripheral and Nuclie's RISC-V core.
GigaDevice supplied sources are mainly supported GD32VF103's peripherals IP such as GPIO, I2C, etc...
Sources to support RISC-V provide from Nuclei.
GD32VF103's driver contains sources to support Nuclei's RISC-V core, but it is too old.
(In the discussion, Nuclei developer @fanghuaqi recommends replacing the old source with NMSIS.)

So I propose to add the NMSIS module separately from GigaDevice's hal module.

For example, Nuclei's RISC-V IP is available as open-source (https://github.com/riscv-mcu/e203_hbirdv2), and NMSIS also supports this.
And also, Nuclei and their customer plans to release a product that implements Nuclie's RISC-V IP core to market in near 
future, these are also supported by NMSIS.

carlescufi, 2021-11-04:
Now that we have a [hal_gigadevice](https://github.com/zephyrproject-rtos/hal_gigadevice) repository, the right path forward would be to open a [Pull Request](https://github.com/zephyrproject-rtos/hal_gigadevice/pulls) in it adding the NMSIS layer. At that point I would like others to chime in in that PR, to evaluate if it makes sense or not for Zephyr.
So please open the PR and let's continue the discussion there.

nandojve, 2021-11-05:
No objections, I think it is time to us look into NMSIS.

soburi, 2021-11-08:
@carlescufi @nandojve
Thanks for comment, And I agree.
I'm now preparing new PR on hal_gigadevice repository.
Please wait a little bit ...

nandojve, 2021-11-08:
Hi @soburi , @carlescufi ,

I was thinking that NMSIS will be a Zephyr module. That way we can use for both gigadevice, HummingBird and others.

carlescufi, 2021-11-11:
After a lengthy discussion, we have come to the following conclusions:

- We will not integrate the NMSIS codebase with Zephyr as a module
- Instead, support for Nuclei cores will be added natively in the main tree, in soc/riscv/

This request for an external HAL is therefore no longer needed.

closed 2021-11-11 by carlescufi