zephyrproject-rtos/zephyr #39587: Zephyr 3.0 Release Checklist

cfriedt, 2021-10-20:
- [ ] Major enhancements:
   - See https://github.com/zephyrproject-rtos/zephyr/projects/9

- [ ] Pre RC1 Steps
   - [ ] Create v3.0.0 milestone so that features that don't make it to the release can be assigned a milestone
   - [ ] Verify that all external components and external dependencies are up to date, for example 
      - mbedTLS
      - tinycrypt
      - FatFS
   - [ ] Check known vulnerabilities, and fix exploitable vulnerabilities or verify them as un-exploitable (Security working group).

See https://docs.zephyrproject.org/latest/development_process/release_process.html#tagging for tagging details

--------
- [ ] Release Notes WIP
  - [ ] Make sure to tell everybody to use the **past tense** (i.e. "X has been added" or "X is now Y")
  - [ ] Create draft `doc/releases/release-notes-3.0.rst` with feature summary from https://www.zephyrproject.org/developers/#releases-overview #22923
  - [ ] Update `doc/releases/release-notes-3.0.rst` detailed sections (relative to 2.6.0 release) as important issues are addressed (subsystem owners). See #25869
--------
- [ ] Finalize Release Notes
  - [ ] Add list of GitHub issues (spell check issue list and fix in GitHub as needed) using `scripts/release/list_issues.py`
  - [ ] Review `doc/releases/release-notes-3.0.rst` overview summary and details sections (see Sign-off below)
    - [ ] Request summary details from code owners
- [ ] Update `doc/reference/api/overview.rst`
- [ ] Update `doc/conf.py` (add version to pick list)
- [ ] Update `doc/LICENSING.rst` for new components that do not have a license
--------
- [ ]  Final Release Notes Sign-off
  - Subsystems
    - [ ] Kernel - @andyross,
    - [ ] Build and infrastructure - @tejlmand @nashif @galak  
    - [ ] Devicetree - @galak, @mbolivar-nordic 
    - [ ] Drivers and sensors - @MaureenHelm, @tbursztyka, @gmarull, @mnkp  
    - [ ] Filesystems - @nvlsianpu, @de-nordic 
    - [ ] Networking - @rlubos, @jukkar
    - [ ] Bluetooth - @jhedberg, @carlescufi 
    - [ ] USB - @jfischer-no 
    - [ ] CAN - @alexanderwachter  
  - Architectures
    - [ ] ARC - @ruuddw 
    - [ ] ARM - @galak, @MaureenHelm 
    - [ ] X86 - @jhedberg 
    - [ ] RISCV - @pgielda
    - [ ] Xtensa, NIOS-II - @nashif, @dcpleung 
  - Board an Soc support - many
  - [ ] Security - @d3zd3z 
--------
- [ ] Update version in VERSION
- [ ] Tag both v3.0.0 and zephyr-v3.0.0
- [ ] Tag v3.0.0 in [example-application](https://github.com/zephyrproject-rtos/example-application)
- [ ] Create release and add notes on https://github.com/zephyrproject-rtos/zephyr/releases. Make sure you use the `zephyr-v3.0.0` tag so that the `.zip` that GitHub offers has a reasonable name. Add the release notes to the GitHub release.
--------
- [ ] After Release
  - [ ] Update patchlevel in Makefile (PATCHLEVEL = 99)
  - [ ] Update https://www.zephyrproject.org/developer-resources/#current-release (email Brett)
  - [ ] Create `v3.0-branch` branch for 3.0.x releases (from v3.0.0 tag, don't include the switch to patchlevel 99)
  - [ ] Create `backport v3.0-branch` label
  - [ ] Merge all post-release PRs in https://github.com/zephyrproject-rtos/zephyr/projects/27
  - [ ] Move all PRs/Issues in the v3.0.0 milestone to the next milestone
  - [ ] Close v3.0.0 milestone 
  - [ ] Email devel@lists.zephyrproject.org & announce@lists.zephyrproject.org lists about release and announce the opening of the merge window
  - [ ] Create checklist issue for next release
  - [ ] Add a comment to this issue linking to the next release checklist issue
  - [ ] Marketing Blog for announcing release (https://www.zephyrproject.org/blog)
  - [ ] Update https://www.zephyrproject.org with latest stats from GitHub (email Brett)
  - [ ] Store the number of currently open PRs to keep track of post-release PR overhead over time

de-nordic, 2022-06-03:
The release is done. Shouldn't this be closed?

dkalowsk, 2022-06-03:
@de-nordic yep.  Thought I did close it, thanks for calling it out.