zephyrproject-rtos/zephyr #39598: use of __noinit with ecc memory hangs system

cfriedt, 2021-10-20:
**Describe the bug**
The `__noinit` attribute is used primarily for the purposes listed below:
1. to preserve a section of SRAM across reboots
1. to speedup init time by reducing the size of `.bss`

Some configurations may also use `__noinit` to skip initialization of memory regions initialized by other application-specific hardware.

There have been two identified areas where memory regions marked as `__noinit` are read-from before they are written-to. This might be reasonable in certain scenarios, but for some systems with ECC-enabled memory controllers, a read operation before a write operation will cause the memory controller to issue a fault, hanging the system.

**To Reproduce**
1. obtain a system with ECC memory and an ECC-capable memory controller
2. enabling logging and / or the shell
3. log messages
4. observe hang

In the above cases, the system is completely hung before the boot banner was even printed.

**Expected behavior**
The system is not hung.

**Impact**
Showstopper

**Logs and console output**
There are no logs, nor is there any output on the console. It may be possible to find a stacktrace.

**Environment (please complete the following information):**
 - OS: Zephyr v2.6.0
 - Toolchain: crosstool-ng / riscv64
 - Commit SHA or Version used: 79a6c07536bc14583198f8e3555df6134d8822cf

**Additional context**
This problem cannot be reproduced on non-ECC systems, and specifically, the ECC memory controller must also be configured to check SRAM memory regions. It is not obvious if there are any readily-available Zephyr community boards that include an ECC memory controller.

evgeniy-paltsev, 2021-10-21:
Hi @cfriedt 

> There have been two identified areas where memory regions marked as __noinit are read-from before they are written-to - specifically, the shell and logging subsystems.

Do you know exact places where it happens? I can't imagine the reason for shell / logging for reading from __noinit before writing to it.

cfriedt, 2021-10-21:
> Hi @cfriedt
>
>
> Do you know exact places where it happens? I can't imagine the reason for shell / logging for reading from __noinit before writing to it.

I can try to find some details (e.g. stack traces) but it may be confidential. Will try my best.

Alternatively, it may be possible to try and trigger faults using MMU / MPU regions.

cfriedt, 2022-09-18:
The solution suggested in #48172 is that the solution should be part of a specific memory controller driver implementation.

closed 2022-12-18 by github-actions[bot]