zephyrproject-rtos/zephyr #39604: west/sign: fix mapping of Zephyr board names to rimage config file name

marc-hb, 2021-10-20:
> This is merely mapping a Zephyr codename to the name of an `rimage/config/$target.toml` config file in SOF so it seems a bit far from "describing hardware components" ... @mbolivar-nordic Could you sketch a bit how it could look like?

Sorry, I wasn't reading in detail and just assumed this was coming from devicetree because the function name begins with `edt_`. If that is not appropriate, there are any number of places you can put the metadata you need in the board directory itself:

- `$BOARD.yaml`
- `board.cmake` (currently just used for runners, but maybe you can set a value here that ultimately tells rimage what to do?)
- some Kconfig symbol

_Originally posted by @mbolivar-nordic in https://github.com/zephyrproject-rtos/zephyr/pull/39050#r729421324_

cfriedt, 2021-11-16:
@mbolivar-nordic - can you follow up with this?

marc-hb, 2021-11-16:
That's SOF specific and not for @mbolivar-nordic

MaureenHelm, 2021-11-16:
> Sorry, I wasn't reading in detail and just assumed this was coming from devicetree because the function name begins with `edt_`. If that is not appropriate, there are any number of places you can [...]

I stumbled upon this on my own last week. I think `board.cmake` is the way to go, I'll try to put a PR together.

closed 2021-11-18 by nashif

mbolivar-nordic, 2021-11-18:
Thanks @MaureenHelm !

PR #40431 (2021-11-18): boards: xtensa: Use a CMake variable to set the rimage target name
Removes hardcoded logic in the west signing script that translates
Zephyr board names to rimage target names. Instead, use a cached CMake
variable set at the board level to define its respective rimage target
name. This eliminates the need to modify the west signing script when
new SOF-supported boards are introduced to Zephyr.

Fixes #39604