zephyrproject-rtos/zephyr #39608: stm32: lpuart: 9600 baudrate doesn't work

ycsin, 2021-10-21:
`lpuart1` won't work when it is set to 9600 baudrate on `nucleo_g0b1re` board.

[image]

The `LPUARTDIV` based on the formula above is ~1,706,666, which doesn't fit into a 20bit `BRR` register and was truncated into 658,091 instead, see pictures below.

[image]

[image]
[image]
[image]

ycsin, 2021-10-21:
I'm able to get it to work by editing the [driver](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/serial/uart_stm32.c#L96) like so:

```c
#if HAS_LPUART_1
	if (IS_LPUART_INSTANCE(UartInstance)) {
		LL_LPUART_SetPrescaler(UartInstance, LL_USART_PRESCALER_DIV4);
		LL_LPUART_SetBaudRate(UartInstance,
				      clock_rate,
#ifdef USART_PRESC_PRESCALER
				      LL_USART_PRESCALER_DIV4,
#endif
				      baud_rate);
	} else {
```

But I think we might need something more so that the driver can select the correct prescaler value to cover whole baudrate range

erwango, 2021-10-21:
> I'm able to get it to work by editing the driver like so:

Ok, nice. Feel free to contribute

PR #39631 (2021-11-23): driver: serial: uart_stm32: Calculate suitable PRESCALER value for lpuart
Current driver set a fixed prescaler value for the lpuart that caused certain baudrate configurations to fail due to LPUARTDIV overflow the LPUART_BRR register.

This PR attempt to calculate a suitable PRESCALER for the selected baudrate, throws error and set the closest possible PRESCALER value if it couldn't get an optimal one.

fixes #39608
closed 2021-11-23 by cfriedt