zephyrproject-rtos/zephyr #39944: Issue: Build the Blinky Sample for RISC-V HiFive1 rev B board using Windows 10

TaLucGiaHoang, 2021-10-30:
Hi, I am following the [Getting Started Guide](https://docs.zephyrproject.org/latest/getting_started/index.html) to build a zephyr sample for my SiFive HiFive1 rev B board (RISC-V). I am using Windows 10 to build.

For toolchain setting, I have added two environment variables:
- ZEPHYR_TOOLCHAIN_VARIANT = cross-compile
- CROSS_COMPILE = D:\hifive1-revb\FreedomStudio-2019-08-2-win64\SiFive\riscv64-unknown-elf-gcc-8.3.0-2019.08.0\bin\riscv64-unknown-elf-

I meet this error message and don't know how to solve. Please help me
```
(.venv) PS D:\zephyrproject\zephyr> west build -p auto -b hifive1_revb .\samples\basic\blinky\
-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): D:/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: D:/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.7.99 (D:/zephyrproject/zephyr), build: v2.7.99-779-gf6e829a68b1a
-- Found west (found suitable version "0.11.1", minimum required is "0.7.1")
-- Board: hifive1_revb
-- Cache files will be written to: D:/zephyrproject/zephyr/.cache
-- Found dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
CMake Error at D:/zephyrproject/zephyr/cmake/generic_toolchain.cmake:36 (include):
  include could not find requested file:

    D:/zephyrproject/zephyr/cmake/toolchain/D:\hifive1-revb\FreedomStudio-2019-08-2-win64\SiFive\riscv64-unknown-elf-gcc-8.3.0-2019.08.0\bin/generic.cmake
Call Stack (most recent call first):
  D:/zephyrproject/zephyr/cmake/app/boilerplate.cmake:540 (include)
  D:/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  D:/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Found BOARD.dts: D:/zephyrproject/zephyr/boards/riscv/hifive1_revb/hifive1_revb.dts
CMake Error at D:/zephyrproject/zephyr/cmake/dts.cmake:167 (message):
  command failed with return code: The system cannot find the file specified
Call Stack (most recent call first):
  D:/zephyrproject/zephyr/cmake/app/boilerplate.cmake:541 (include)
  D:/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  D:/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' '-DWEST_PYTHON=D:\zephyrproject\.venv\Scripts\python.exe' '-BD:\zephyrproject\zephyr\build' '-SD:\zephyrproject\zephyr\samples\basic\blinky' -GNinja
```

nashif, 2021-11-02:
board is hifive1_revb

TaLucGiaHoang, 2021-11-02:
Yes, I want to build for hifive1_revb. Could you please give me any instruction

tejlmand, 2021-11-09:
This:
```
CMake Error at D:/zephyrproject/zephyr/cmake/generic_toolchain.cmake:36 (include):
  include could not find requested file:

    D:/zephyrproject/zephyr/cmake/toolchain/D:\hifive1-revb\FreedomStudio-2019-08-2-win64\SiFive\riscv64-unknown-elf-gcc-8.3.0-2019.08.0\bin/generic.cmake
```
looks like you have used `ZEPHYR_TOOLCHAIN_VARIANT = D:\hifive1-revb\FreedomStudio-2019-08-2-win64\SiFive\riscv64-unknown-elf-gcc-8.3.0-2019.08.0\bin` and not `ZEPHYR_TOOLCHAIN_VARIANT = cross-compile`.

Because dividing that path into sub-components gives:
- `D:/zephyrproject/zephyr/` == `TOOLCHAIN_ROOT`, which defaults to `ZEPHYR_BASE`
- `cmake/toolchain` hard-coded lookup path, see below.
- `D:\hifive1-revb\FreedomStudio-2019-08-2-win64\SiFive\riscv64-unknown-elf-gcc-8.3.0-2019.08.0\bin` == ZEPHYR_TOOLCHAIN_VARIANT
- `generic.cmake` fixed filename to use for the Zephyr build system

Ref code:
https://github.com/zephyrproject-rtos/zephyr/blob/1238410914597aef1e0661e0b2e240a448f2dd60/cmake/generic_toolchain.cmake#L36

Could you please take a second look at your setup.

tejlmand, 2022-01-10:
Closing, question has been answered.

closed 2022-01-10 by tejlmand