zephyrproject-rtos/zephyr #39947: open-amp problem with dcache!

lenghonglin, 2021-10-31:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
Hello , everyone , these day i port open-amp with our self-design soc which is contain cortex-m33 and xtensa dsp hifi4。

i found these were so much cases with dcache enable , such as the vaule with the same share-memroy address is different.  We confirm that this is caused by cache. 

In my opinion, there are several ways to solve the problem:

1. use `SCB_DisableICache` to disable cache, but this way is a little rough.
2. we can use mpu to config part cache is non-cacheable. 
i see [mpu_init](https://github.com/zephyrproject-rtos/zephyr/blob/3f9f299499b6cf98ae72f7ee2cd9a28100a72935/arch/arm/core/aarch32/mpu/arm_mpu_v8_internal.h#L34) is config three memory regions.  My idea is to divide 320K SRAM into 300K and 20K through device tree, and the 20K SRAM config to Attribute-2 which is SRAM no cache-able regions by [int z_arm_mpu_init(void)](https://github.com/zephyrproject-rtos/zephyr/blob/3f9f299499b6cf98ae72f7ee2cd9a28100a72935/arch/arm/core/aarch32/mpu/arm_mpu.c#L317) .  but i can not find the api to do this operate.
3. in this [pull request](https://github.com/OpenAMP/open-amp/pull/293), open-amp support the interface to FLUSH or INVALIDATE. but zephyr is not update to this version.

carlocaione, 2021-11-02:
> in this pull request, open-amp support the interface to FLUSH or INVALIDATE. but zephyr is not update to this version.

Working on this:
- https://github.com/zephyrproject-rtos/open-amp/pull/11
- https://github.com/zephyrproject-rtos/libmetal/pull/12

I'm preparing now the `west.yaml` update for zephyr

carlocaione, 2021-11-02:
> but i can not find the api to do this opreate.

You probably also want to take a look at https://github.com/zephyrproject-rtos/zephyr/pull/39071

lenghonglin, 2021-11-02:
> > but i can not find the api to do this opreate.
> 
> You probably also want to take a look at #39071

In my way , enable `CONFIG_CPU_HAS_CUSTOM_FIXED_SOC_MPU_REGIONS` , and rewrite `mpu_regions.c`， and done it!

closed 2021-11-17 by galak