zephyrproject-rtos/zephyr #39948: tests failing on qemu_cortex_a9

cfriedt, 2021-10-31:
**Describe the bug**

The `main` branch is broken in CI currently
https://buildkite.com/zephyr/zephyr/builds/44959

**To Reproduce**
Any of the following:
```
twister -i -p qemu_cortex_a9 -s tests/kernel/fatal/exception/kernel.common.stack_sentinel
twister -i -p qemu_cortex_a53_smp -s tests/kernel/sched/schedule_api/kernel.scheduler
twister -i -p qemu_cortex_a53_smp -s tests/kernel/lifo/lifo_usage/kernel.lifo.usage
```

**Expected behavior**
All tests in CI should pass

**Impact**
This breaks CI on those platform and introduces failure modes that should be stable on all platforms.

**Logs and console output**
https://buildkite.com/zephyr/zephyr/builds/44959#c0ab1a7f-6222-4291-877a-9b4dbcf4515e

**Environment (please complete the following information):**
 - OS: CI
 - Toolchain: Zephyr SDK 0.13.1
 - Commit: 3f9f299499b6cf98ae72f7ee2cd9a28100a72935

**Additional context**
Looks to be traced back to PR #36983

stephanosio, 2021-11-01:
The `qemu_cortex_a53` failures described are likely only intermittent failures: https://github.com/zephyrproject-rtos/zephyr/pull/39949#discussion_r739938866

nashif, 2021-11-02:
issue still need to be resolved.

ibirnbaum, 2021-11-04:
I've looked into this today, and at least on my system, this testcase also reliably fails on qemu_cortex_r5. I don't have my trusty Lauterbach debugger available at the moment, so gdb had to do, but as far as I can tell, this appears to be some sort of race condition.

I breakpointed on k_sys_fatal_error_handler where the expected error condition is checked. From there, it's back to z_fatal_error, where k_thread_abort is called. It appears that by this point, alt_thread is the current thread, and its current state within z_thread_abort is _THREAD_QUEUED. The current thread pointer never gets changed during the whole procedure, by the way. Once the end of z_arm_fatal_error is reached, it's back to z_do_kernel_oops from where the branch to z_arm_int_exit is taken. This is where things get interesting. z_check_stack_sentinel is called, and by that time, *stack equals STACK_SENTINEL. Once the end of z_check_stack_sentinel is reached, it's back to exc_exit.S:108 (cpsid i). The exit procedure continues to the rfeia sp! instruction - and execution jumps to the tail end of z_check_stack_sentinel. So it seems that the timer interrupt struck exactly at the moment when a running execution of z_check_stack_sentinel had just detected *stack != STACK_SENTINEL and had therefore overwritten *stack and called z_except_reason.

A reliable workaround is to modify the function stack_sentinel_timer in the test case code: adding and offset of K_MSEC(100) (randomly picked value, haven't tried other values yet) instead of K_NO_WAIT to the k_timer_start function call will cause the test case to pass on both A9 and R5. 

At the time being, gdb doesn't provide me with enough insight on who called z_check_stack_sentinel by the time the timer interrupt occurrs, and I'm not really familiar with the inner workings of the kernel, so I don't want to propose this as the obvious solution. Does anyone have any ideas as to what the suspected race condition might be?

ibirnbaum, 2021-11-08:
Regarding the offset for the k_timer_start function call mentioned above: the value can be as little as K_NSEC(1). What obviously matters is the code path taken in case of any value != K_NO_WAIT (additional scheduler call/context switch?).

ibirnbaum, 2021-11-10:
This must be some sort of timing/scheduling issue. The unmodified test case (with timer offset value K_NO_WAIT) passes if tickless mode is disabled. So somehow, the next system tick after the timer-based context switch gets something moving again which stalls in tickless mode for some reason.

ibirnbaum, 2021-11-10:
The unmodified test passes on actual Zynq hardware btw.

stephanosio, 2022-01-26:
@ibirnbaum Has there been any progress in fixing this issue?

ibirnbaum, 2022-02-07:
> @ibirnbaum Has there been any progress in fixing this issue?

Hi Stephanos,
no, there hasn't been any further progress on my side. I had hoped that someone with more insight into the inner workings of the kernel might be able to explain why the test passes if the scheduler is invoked manually more or less directly (comp. my comment from Nov 8th). As I mentioned before, I think this is a race condition, but I don't have the inside knowledge of the kernel / scheduler / intention of the test case in order to be able to explain it.

The thing is: the test fails on both qemu_cortex_a9 and qemu_cortex_r5. So this issue can't be down to Cortex-A specific additions in the arch code. The simulated A9 and R5 run at different clock speeds and the system timer differs between the two (ARM global timer in the A9, Xilinx TTC in the R5). So this issue isn't down to the CPU clock speed or the system timer (and the fact the the global timer in the A9  - at least an actual hardware A9 - can be affected by an erratum whose handling can be enabled or disabled in the build configuration). The test passes when executed on an actual Zynq. I could test what the situation is with an actual UltraScale R5.

My conclusion is that if this is a race condition (and someone with more kernel knowledge agrees to that assumption), it's either QEMU-specific, or there's some other factor that contributes to it, but it doesn't just exist on qemu_cortex_a9 and/or the Zynq-7000 exclusively.

If any further investigation in the form of stepping through the test case with a hardware debugger and providing the information on the order of execution on actual hardware is considered helpful then I'll certainly assist with that. But as mentioned before, the test case "unfortunately" passes on actual HW.

stephanosio, 2022-02-07:
> The test passes when executed on an actual Zynq.

So this is not actually broken on real hardware and is likely just a QEMU emulation issue -- in that case, this issue should not be classified as "high" priority.

I have downgraded it to "low" priority since it is working in real hardware and is not actually broken; cc @dkalowsk.

closed 2022-04-23 by github-actions[bot]