zephyrproject-rtos/zephyr #39969: USB not automatically enabled when USB_UART_CONSOLE is set

lairdjm, 2021-11-01:
**Describe the bug**
In Zephyr 2.4, the `usb_enable()` function was added with a message that it must be used to start USB support during the application, however, with a note that this would not be required if USB_UART_CONSOLE is set:
```
The usb_enable() function, which, for some samples, was invoked automatically on system boot up, now needs to be explicitly called by the application in order to enable the USB subsystem. If your application relies on any of the following Kconfig options, then it shall also enable the USB subsystem:
    CONFIG_USB_DEVICE_NETWORK_ECM
    CONFIG_USB_DEVICE_NETWORK_EEM
    CONFIG_USB_DEVICE_NETWORK_RNDIS
    CONFIG_TRACING_BACKEND_USB
    **CONFIG_USB_UART_CONSOLE**
```
However when testing 265cdf8dc6d08320665c1f898bce5403eb4df1a8 I find this statement to be incorrect and I have to manually add `usb_enable(NULL);` to a sample application to get the USB CDC port started despite this being set in the configuration:
```
grep -r "USB_UART_CONSOLE" zephyr/.config
CONFIG_USB_UART_CONSOLE=y
```

**To Reproduce**
Build a sample application for a board with a USB CDC port used for the console, I am testing with `bl654_usb` and `samples/philosophers`, flash to the device and no USB device will be enumerated. Then add this to `src/main.c`:
```
void main(void)
{
        usb_enable(NULL);
```
Build and flash the application, suddenly USB works.

**Expected behavior**
USB to work without the above function call

**Impact**
Showstopper, the `bl654_usb` board is entirely broken because of this bug

**Logs and console output**
Output before change:
```
dmesg | tail
[ 7760.180521] usb 3-13: new full-speed USB device number 39 using xhci_hcd
[ 7760.311121] usb 3-13: New USB device found, idVendor=1915, idProduct=521f, bcdDevice= 1.00
[ 7760.311125] usb 3-13: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7760.311127] usb 3-13: Product: Nordic Open DFU Bootloader
[ 7760.311129] usb 3-13: Manufacturer: Laird Connectivity
[ 7760.311130] usb 3-13: SerialNumber: CE7B109A717C
[ 7760.314012] cdc_acm 3-13:1.0: ttyACM0: USB ACM device
[ 7774.686606] usb 3-13: USB disconnect, device number 39 <-- Bootloader, no USB enumeration after bootloader exits

lsusb | grep Nordic
```
Output after change:
```
dmesg | tail
[ 7855.047350] usb 3-13: new full-speed USB device number 40 using xhci_hcd
[ 7855.176054] usb 3-13: New USB device found, idVendor=1915, idProduct=521f, bcdDevice= 1.00
[ 7855.176059] usb 3-13: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7855.176061] usb 3-13: Product: Nordic Open DFU Bootloader
[ 7855.176062] usb 3-13: Manufacturer: Laird Connectivity
[ 7855.176063] usb 3-13: SerialNumber: CE7B109A717C
[ 7855.178549] cdc_acm 3-13:1.0: ttyACM0: USB ACM device
[ 7867.228169] usb 3-13: USB disconnect, device number 40
[ 7868.295407] usb 3-13: new full-speed USB device number 41 using xhci_hcd
[ 7868.424362] usb 3-13: New USB device found, idVendor=2fe3, idProduct=0100, bcdDevice= 2.07
[ 7868.424367] usb 3-13: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7868.424369] usb 3-13: Product: USB-DEV
[ 7868.424370] usb 3-13: Manufacturer: ZEPHYR
[ 7868.424371] usb 3-13: SerialNumber: 377ED198E19E7477
[ 7868.427149] cdc_acm 3-13:1.0: ttyACM0: USB ACM device

lsusb | grep Nordic
Bus 003 Device 041: ID 2fe3:0100 NordicSemiconductor USB-DEV
```

**Environment (please complete the following information):**
 - OS: Debian linux, also tried on windows 10
 - Toolchain: Zephyr SDK 0.13.1
 - Commit SHA or Version used: 265cdf8dc6d08320665c1f898bce5403eb4df1a8

jfischer-no, 2021-11-01:
> **Describe the bug** In Zephyr 2.4, the `usb_enable()` function was added with a message that it must be used to start USB support during the application, however, with a note that this would not be [...]

No, please read it again:
"If your application relies on any of the following Kconfig options, then it shall also enable the USB subsystem:
 ....
 CONFIG_USB_UART_CONSOLE"
https://docs.zephyrproject.org/latest/releases/release-notes-2.4.html

closed 2021-11-01 by jfischer-no

lairdjm, 2021-11-01:
@jfischer-no So what's the idea for boards with a USB CDC port then sample-wise so that the USB CDC port starts up open by default for console output as it would with a normal UART on other boards?

jfischer-no, 2021-11-02:
> @jfischer-no So what's the idea for boards with a USB CDC port then sample-wise so that the USB CDC port starts up open by default for console output as it would with a normal UART on other boards?

The way it was solved for CDC ACM before version 2.4 had no technical advantages over "enable the USB subsystem in application code". There is currently no solution, and I know that it is not convenient for USB dongle like boards.