zephyrproject-rtos/zephyr #39973: Bluetooth:hci_usb example returning "Unknown HCI Command" after reset.

AdrianEggenberger, 2021-11-01:
**Describe the bug**
A hci_usb firmware built on the Zephyr 2.7.0 with DTM support (CONFIG_BT_TESTING) is used for DTM tests. After a reset of the hardware a transmitter test can be started with the "LE Transmitter Test" command. When this running test is stopped with a "Controller Reset" (instead of a regular test termination) the next start of a test returns "Unknown HCI Command".

**To Reproduce**
Steps to reproduce the behavior:
1. Extend configuration for hci_usb sample with DTM support
2. Build project and flash it to a nrf52840dk board.
3. Initiate a "LE transmitter Test" command
4. Reset by a "Reset Controller" command
5. Initiate a 2nd "LE transmitter Test" command
6. See error

**Expected behavior**
It is expected that a DTM test may be restarted normally after a controller reset.

**Impact**
annoyance

**Logs and console output**
I've recorded the HCI communication and stored it to a cfa file for wireshark:
[HCI-Log.zip](https://github.com/zephyrproject-rtos/zephyr/files/7456058/HCI-Log.zip)

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Zephyr SDK
 - Version 2.7.0

AdrianEggenberger, 2021-11-13:
Today I had a closer look at the issue.

The problem is that the reset doesn't "stop" the test the had been started before the reset. In the module lll_test.c the variable started is still set to started after a reset. This causes the function to return 1. The module hci.c translates this somehow to a failure unknown command.

Is my expectation correct that the reset should reset the radio and also reset the variables that indicate a running test?

cvinayak, 2021-11-15:
@AdrianEggenberger Thank you for the investigation, I have created a PR #40342, please test and leave your comment in the PR.

PR #40342 (2021-11-16): Bluetooth: Controller: Fix missing DTM Tx/Rx reset on HCI Reset Command
Fix missing implementation to reset DTM Tx/Rx reset on HCI
Reset Command.

Fixes #39973.
closed 2021-11-16 by carlescufi