zephyrproject-rtos/zephyr #39978: logging.log2_api_deferred tests fail on qemu_cortex_a9 with GNU Arm Embedded

stephanosio, 2021-11-02:
**Describe the bug**
The `logging.log2_api_deferred` tests fail at run-time on `qemu_cortex_a9` when compiling with the GNU Arm Embedded toolchain.

**To Reproduce**
```
twister -v -N -T tests/subsys/logging/log_api -p qemu_cortex_a9
```

**Expected behavior**
All tests pass.

**Impact**
"Deferred logging" API test is broken on `qemu_cortex_a9` with the GNU Arm Embedded toolchain.

**Logs and console output**
```
$ scripts/twister -v -N -T tests/subsys/logging/log_api -p qemu_cortex_a9
ZEPHYR_BASE unset, using "/home/stephanos/Dev/zephyrproject/zephyr"
Renaming output directory to /home/stephanos/Dev/zephyrproject/zephyr/twister-out.18
INFO    - Zephyr version: v2.7.99-878-gc03519fffc6a
INFO    - JOBS: 60
INFO    - Using 'gnuarmemb' toolchain.
INFO    - Building initial testcase list...
INFO    - 18 test scenarios (18 configurations) selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    -  1/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log2_api_immediate_rt_filter PASSED (qemu 2.604s)
INFO    -  2/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log_api_deferred_no_overflow PASSED (qemu 2.650s)
INFO    -  3/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log2_api_immediate PASSED (qemu 2.658s)
INFO    -  4/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log_api_immediate_static_filter PASSED (qemu 2.692s)
INFO    -  5/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log_api_immediate_rt_filter PASSED (qemu 2.591s)
INFO    -  6/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log_api_deferred_overflow PASSED (qemu 2.799s)
INFO    -  7/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log2_api_deferred_no_overflow PASSED (qemu 2.562s)
INFO    -  8/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log_api_deferred_overflow_rt_filter PASSED (qemu 2.663s)
INFO    -  9/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log2_api_immediate_static_filter PASSED (qemu 2.572s)
INFO    - 10/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log2_api_deferred_overflow_rt_filter FAILED Failed (qemu 2.649s)
ERROR   - see: /home/stephanos/Dev/zephyrproject/zephyr/twister-out/qemu_cortex_a9/tests/subsys/logging/log_api/logging.log2_api_deferred_overflow_rt_filter/handler.log
INFO    - 11/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log2_api_deferred_static_filter FAILED Failed (qemu 2.682s)
ERROR   - see: /home/stephanos/Dev/zephyrproject/zephyr/twister-out/qemu_cortex_a9/tests/subsys/logging/log_api/logging.log2_api_deferred_static_filter/handler.log
INFO    - 12/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log_api_deferred_func_prefix PASSED (qemu 2.828s)
INFO    - 13/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log2_api_immediate_64b_timestamp PASSED (qemu 2.434s)
INFO    - 14/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log_api_deferred_static_filter PASSED (qemu 2.708s)
INFO    - 15/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log2_api_deferred_overflow PASSED (qemu 2.731s)
INFO    - 16/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log2_api_deferred_func_prefix FAILED Failed (qemu 2.805s)
ERROR   - see: /home/stephanos/Dev/zephyrproject/zephyr/twister-out/qemu_cortex_a9/tests/subsys/logging/log_api/logging.log2_api_deferred_func_prefix/handler.log
INFO    - 17/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log2_api_deferred_64b_timestamp PASSED (qemu 2.832s)
INFO    - 18/18 qemu_cortex_a9            tests/subsys/logging/log_api/logging.log_api_immediate PASSED (qemu 2.703s)

INFO    - 15 of 18 test configurations passed (83.33%), 3 failed, 0 skipped with 0 warnings in 13.26 seconds
INFO    - In total -12 test cases were executed, 30 skipped on 1 out of total 402 platforms (0.25%)
INFO    - 18 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing xunit report /home/stephanos/Dev/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/stephanos/Dev/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

[logging.log2_api_deferred_overflow_rt_filter log](https://gist.github.com/stephanosio/597bf6812ff4846e7c2154fefb6d497b)
[logging.log2_api_deferred_static_filter](https://gist.github.com/stephanosio/8ef2a148f654e078cbbc6b5bba4fd0bb)
[logging.log2_api_deferred_func_prefix](https://gist.github.com/stephanosio/1d6aa212168ebc7a194972e242be594a)

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: GNU Arm Embedded 10-2020-q4-major
 - Commit SHA: c03519fffc6ac66b0e4feea3742701e05b7a38ec

**Additional context**
* These tests do not fail on `qemu_cortex_a9` when compiling with the Zephyr SDK.
* Across all the emulated platforms, GNU Arm Embedded-specific failures are observed only on `qemu_cortex_a9`.

stephanosio, 2021-11-02:
@ibirnaum FYI

ibirnbaum, 2021-11-03:
I'm looking into this. I got hold of the GNU ARM Embedded toolchain version 10-2020-q4-major as reported above, plus the latest major release from a few days ago. I'll look into the run-time behaviour of the test case using those two compilers plus the one from the Zephyr SDK, I'm hoping that I'll come across the difference in the output that causes the issue.

ibirnbaum, 2021-11-08:
This issue is reproducible on my system, although in my case, it only affects two test cases: deferred_func_prefix and deferred_overflow_filter. I've picked deferred_overflow_filter for further investigation. What I've found so far:

- Cause of failure: assertion failure in mock_backend.c:process(), called via 4 more levels on the call stack from test_log_overflow:365. Mismatch between the timestamp contained in the message vs. the timestamp of the corresponding expected message entry in the mock backend. Expected timestamp (hardcoded) is 1, actual timestamp is 2. If the test passes with the gcc from the Zephyr SDK, those two values are likely equal in that case.
- The assertion failure occurs with both the 2020-Q4 and the latest version 10.3-2021.10 of the GNU ARM Embedded toolchain.
- The assertion failure occurs with both the default optimization settings and "Optimize nothing".

How have those expected timestamp values been determined?

cfriedt, 2021-11-09:
So crazy that this could go all the way down to the toolchain!

nordic-krch, 2021-11-29:
We have seen failing test with zephyr toolchain: https://github.com/zephyrproject-rtos/zephyr/pull/40733#issuecomment-981716142

From my investigation it seems that the reason is that `__rodata_region_start` label is not pointing to the start of rodata section. `cbprintf` is using `__rodata_region_start` and `__rodata_region_end` to detect if variable is read only. When this detection fails test fails.

Example map file for `qemu_cortex_a9` where `__rodata_region_start` is aligned and does not point to the first variable in rodata section:
```
               0x000000000010d000                . = ALIGN (_region_min_align)                                                                        
                0x000000000010d000                __rodata_region_start = .                                                                            
                                                                                                                                                       
initlevel       0x000000000010c1b4       0x40                                                                                                          
                0x000000000010c1b4                __init_start = .                                                                                     
                0x000000000010c1b4                __init_PRE_KERNEL_1_start = .   
```

On other platforms (checked `qemu_cortex_m0` and `qemu_x86`) this section of map file looks like that (`__rodata_region_start` matches `__init_start`):
```
                0x0000000000113000                __rodata_region_start = .                                                                            
                                                                                                                                                       
initlevel       0x0000000000113000       0x68                                                                                                          
                0x0000000000113000                __init_start = .                                                                                     
                0x0000000000113000                __init_PRE_KERNEL_1_start = .

carlescufi, 2021-12-02:
@stephanosio this is reproduceable with the Zephyr SDK as well? If so please update the description of the issue to reflect this.

PR #41513 (2022-01-07): arch: arm: aarch32: fix section alignment behind rodata start marker
Fixes #39978.

This commit assigns the __rodata_region_start marker to the ROMABLE region
prior to the inclusion of linker/common-rom.ld, linker/thread-local-storage.ld and
linker/cplusplus-rom.ld. Prior to this fix, the __rodata_region_start marker was
properly aligned and indicated the expected memory location for the start of the
rodata section and similar sections, but it was disconnected from the ROMABLE
region into which the subsequent sections are being integrated, resulting in place-
ment of those sections right behin [... cut]
closed 2022-01-07 by MaureenHelm