zephyrproject-rtos/zephyr #39985: Telnet shell breaks upon sending Ctrl+C character

lairdjm, 2021-11-02:
**Describe the bug**
The `shell_module` sample application is modified to host a telnet shell and loaded to a board with an ethernet connection, a linux machine connects to the board using the standard `telnet` command, at this point, basic shell usage works. Ctrl+C is used on the telnet session, at which point the telnet session is essentially dead and the remote device fails to continue working with the current telnet session, the telnet session must be closed and re-opened for it to resume working

**To Reproduce**
Shell application with telnet shell interface, connect, send Ctrl+C character, try to continue using shell

**Expected behavior**
The shell to continue working

**Impact**
Annoyance

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.1
 - Commit SHA or Version used: 351d065f63c04b3431b10a2485a122e2de713749

lairdjm, 2021-11-02:
Ctrl+Z causes the same issue also

jakub-uC, 2021-11-03:
@rlubos : Can you please help me with it?

rlubos, 2021-11-03:
@jakub-uC I'll have a look.

rlubos, 2021-11-04:
I've investigated this a bit, but it seems that it's not the Zephyr implementation that breaks. I've observed that when CTRL+C is typed in telent client, it sends "Interrupt process" command to the server, but Zephyr implementation doesn't even process that. After this event however the telnet client stops displaying any incoming data from the server. I've observed on wireshark that the client **still** sends data to the server when typed, and the server responds normally, but the data is not printed on the client terminal.

Perhaps it's a matter of negotiating some parameters on connection start (Zephyr implmenetation has a very limited set of supported commands, and they're disabled by default) in order to get this to work as expected. I've tested with linux telnetd server and they negotiate hard on the connection startup, but when CTR-C is on input, it jsut sends it ascii representation, not the "Interrupt process" command.

rlubos, 2021-11-10:
@lairdjm It turned out that the problem was `DO TIMING MARK` command sent along with the `INTERRUPT PROCESS` command - it has not been processed correctly causing the client to lock up. I've submitted a PR that improves the telnet commands handling: https://github.com/zephyrproject-rtos/zephyr/pull/40252

PR #40252 (2021-11-12): shell: telnet: Fix CTRL-C handling
A few fixes for the issues spotted when investigating CTRL-C handling:
* Fix command processing, currently only the first command in the packet has been processed,
* Disable echo on shell - on ECHO command we reply that we won't echo characters, so the implementation should conform,
* Enable command handling in the sample by default - the client behaves considerably better when he receives responses for the commands it sends, i. e. it doesn't lock on the unresponded `DO TIMING MARK` command.

Fixes #39985
closed 2021-11-12 by cfriedt