zephyrproject-rtos/zephyr #40000: k_timer timeout handler is called with interrupts locked

nordic-krch, 2021-11-02:
**Describe the bug**
Since #38847 timeout handler is called with locked interrupts. It is extremely dangerous since user is not aware of that and system has no control for how long interrupts are locked blocking higher priority interrupts.

**Expected behavior**
As before user handler should be called with interrupts unlocked.

**Impact**
Severe.

nordic-krch, 2021-11-02:
fyi @chen-png @andyross @nashif

nashif, 2021-11-02:
@nordic-krch how are you reproducing this? Can you provide more details?

andyross, 2021-11-02:
Not sure I see the concern here.

The k_timer callback is in interrupt context by design (that's what k_timer is -- it's a mechanism for real time apps to hook the timer interrupt).  It's true that this is a plausible latency bug, but only vs. higher priority hardware interrupts that would have been expected to be able to preempt the timer interrupt.  If app code has long-running timer handlers, it's almost certainly got much (much!) worse problems already.

andyross, 2021-11-02:
That said, it certainly wouldn't be wrong to release the lock around the call to timer->expiry_fn().

andyross, 2021-11-02:
Let me know if I've misunderstood something.  But I'm going to mark this as a low priority bug and assign to Chen Peng.  It's worth fixing, it's not a crisis.

nordic-krch, 2021-11-02:
@andyross It is critical. You have multiple levels of interrupts and calling user handler with all levels locked is something totally different than just calling user handler in the interrupt context. Especially, when it behaves differently than before. One example is bluetooth stack, it uses higher priority level than RTC interrupt (used for system timer) and does not accept latency.

In general, it is unaccepted to defer to an unknown code with interrupts locked. I would change it back to high priority.

nordic-krch, 2021-11-02:
@nashif 
> how are you reproducing this? Can you provide more details?

There is a test that relies on assumption that counter interrupt will preempt system timer interrupt (it is explicitly configured to higher level) and I had issues with that test. https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/drivers/uart/uart_mix_fifo_poll

andyross, 2021-11-02:
I repeat, though: k_timer handlers are **already** latency-critical code that have to be carefully designed, because by design they will preempt any thread in the system.  All this did was introduce the same latency issue that was already present to a subsystem (counters in your case) that was previously immune due to interrupt priority.

Meh, I take no strong position on priority here.  More glory for whoever adds the two lines of code :)

chen-png, 2021-11-03:
@andyross @nordic-krch Could we narrow down the range of timer spinlock in z_timer_expiration_handler like below,

the main problem in #38847 is that the timer expiration function maybe executed before thread has been added into timer's wait queue, so we only use the spinlock to sync this, do not include  timer->expiry_fn(timer);.
```
+       if (IS_ENABLED(CONFIG_MULTITHREADING)) {
+               thread = z_waitq_head(&timer->wait_q);
+               if (thread != NULL) {
+                       z_unpend_thread_no_timeout(thread);
+                       arch_thread_return_value_set(thread, 0);
+               } else {
+                       ;
+               }
+       } else {
+               ;
+       }
+       k_spin_unlock(&lock, key);
+
        /* invoke timer expiry function */
        if (timer->expiry_fn != NULL) {
                timer->expiry_fn(timer);
        }

-       if (!IS_ENABLED(CONFIG_MULTITHREADING)) {
-               k_spin_unlock(&lock, key);
-               return;
-       }
-
-       thread = z_waitq_head(&timer->wait_q);
-
-       if (thread == NULL) {
-               k_spin_unlock(&lock, key);
-               return;
+       if (thread != NULL) {
+               z_ready_thread(thread);
+       } else {
+               ;
        }
-
-       z_unpend_thread_no_timeout(thread);
-
-       arch_thread_return_value_set(thread, 0);
-
-       k_spin_unlock(&lock, key);
-
-       z_ready_thread(thread);
 }

```

PR #40048 (2021-11-08): kernel: timer: Call user handler without spinlock
Add spinlock unlocking before calling timer expiration
handler. Locking was introduced by dde3d6c.

Fixes #40000.
closed 2021-11-08 by nashif