zephyrproject-rtos/zephyr #40003: zephyr writes to disconnected device --> Bus fault

CodingGhost, 2021-11-02:
**Describe the bug**
I am trying to implement caching of a static Gatt database, and at first it seems to work, but when disconnecting the device, I get a Bus fault.

on the first connect, the disconnect is working (service discovery is launched before), on the second connect, service discovery is not launched, and I get this upon disconnecting:

`<inf> bbox: disconnecting bbox...
[00:02:24.383,575] <inf> bbox: err:0
[00:02:24.383,636] <dbg> bt_att.bt_att_recv: Received ATT chan 0x2000eac0 code 0x13 len 0
[00:02:24.383,636] <dbg> bt_att.att_handle_write_rsp: 
[00:02:24.383,697] <dbg> bt_att.att_handle_rsp: chan 0x2000eac0 err 0x00 len 0: 
[00:02:24.383,697] <dbg> bt_att.bt_att_req_free: req 0x2001cd48
[00:02:24.383,697] <dbg> bt_att.att_req_send_process: req 0x2001cd38
[00:02:24.383,697] <dbg> bt_att.bt_att_chan_req_send: req 0x2001cd38
[00:02:24.383,728] <dbg> bt_att.chan_req_send: chan 0x2000eac0 req 0x2001cd38 len 3
[00:02:24.383,728] <dbg> bt_att.chan_send: code 0x02
[00:02:24.383,728] <err> bt_conn: not connected!
[00:02:24.506,958] <dbg> bt_att.bt_att_disconnected: chan 0x2000eac8 cid 0x0004
[00:02:24.506,958] <dbg> bt_att.att_chan_detach: chan 0x2000eac0
[00:02:24.507,019] <dbg> bt_att.att_handle_rsp: chan 0x2000eac0 err 0x0e len 0: 
[00:02:24.507,019] <dbg> bt_att.bt_att_req_free: req 0x2001cd38
[00:02:24.507,019] <dbg> bt_att.att_req_send_process: req 0x2001cd38
[00:02:24.507,049] <wrn> bt_att: Not connected
[00:02:24.507,049] <dbg> bt_att.bt_att_req_free: req 0x2001cd38
[00:02:24.507,049] <dbg> bt_att.bt_att_req_free: req 0x2001cd48
[00:02:24.507,049] <err> os: ***** BUS FAULT *****`

the same thing on the first connect:

`bbox: disconnecting bbox...
[00:02:10.993,530] <inf> bbox: err:0
[00:02:10.993,591] <dbg> bt_att.bt_att_recv: Received ATT chan 0x2000eac0 code 0x13 len 0
[00:02:10.993,591] <dbg> bt_att.att_handle_write_rsp: 
[00:02:10.993,621] <dbg> bt_att.att_handle_rsp: chan 0x2000eac0 err 0x00 len 0: 
[00:02:10.993,652] <dbg> bt_att.bt_att_req_free: req 0x2001cd48
[00:02:11.117,492] <dbg> bt_att.bt_att_disconnected: chan 0x2000eac8 cid 0x0004
[00:02:11.117,523] <dbg> bt_att.att_chan_detach: chan 0x2000eac0
[00:02:11.117,584] <dbg> bt_att.bt_att_released: chan 0x2000eac0
[00:02:11.117,614] <inf> bthandler: dc callback
[00:02:11.117,645] <inf> bthandler: Disconnected (reason 22)`

and the same thing, but I dont skip service discovery, but launch it and do nothing ( just return GATT_ITER_STOP)

`bbox: disconnecting bbox...
[00:00:22.817,535] <inf> bbox: err:0
[00:00:22.817,749] <dbg> bt_att.bt_att_recv: Received ATT chan 0x2000ead0 code 0x13 len 0
[00:00:22.817,779] <dbg> bt_att.att_handle_write_rsp: 
[00:00:22.817,810] <dbg> bt_att.att_handle_rsp: chan 0x2000ead0 err 0x00 len 0: 
[00:00:22.817,840] <dbg> bt_att.bt_att_req_free: req 0x2001cd48
[00:00:22.817,840] <dbg> bt_att.att_req_send_process: req 0x2001cd38
[00:00:22.817,840] <dbg> bt_att.bt_att_chan_req_send: req 0x2001cd38
[00:00:22.817,840] <dbg> bt_att.chan_req_send: chan 0x2000ead0 req 0x2001cd38 len 3
[00:00:22.817,871] <dbg> bt_att.chan_send: code 0x02
[00:00:22.817,871] <err> bt_conn: not connected!
[00:00:22.941,680] <dbg> bt_att.bt_att_disconnected: chan 0x2000ead8 cid 0x0004
[00:00:22.941,680] <dbg> bt_att.att_chan_detach: chan 0x2000ead0
[00:00:22.941,711] <dbg> bt_att.att_handle_rsp: chan 0x2000ead0 err 0x0e len 0: 
[00:00:22.941,741] <dbg> bt_att.bt_att_req_free: req 0x2001cd38
[00:00:22.941,741] <dbg> bt_att.att_req_send_process: req 0x2001cd38
[00:00:22.941,741] <wrn> bt_att: Not connected
[00:00:22.941,772] <dbg> bt_att.bt_att_req_free: req 0x2001cd38
[00:00:22.941,772] <dbg> bt_att.bt_att_req_free: req 0x2001cd48
[00:00:22.941,772] <dbg> bt_att.bt_att_req_free: req 0x2001cd58
[00:00:22.941,833] <dbg> bt_att.bt_att_released: chan 0x2000ead0
[00:00:22.941,864] <inf> bthandler: dc callback
[00:00:22.941,894] <inf> bthandler: Disconnected (reason 22)`

it is 100% reproducable.

**To Reproduce**
one should be able to reproduce this, by doing the same thing as I described.
If its not possible to diagnose this without an example, I will try to make one, but its difficult because I cannot share the code and have to write a new example for it.

**Impact**
latency issue, not a show stopper but its bad.

**Environment (please complete the following information):**
Zephyr version 2.6.0 is being used, I cannot update to 2.7.0 for now. but I also didnt see any release notes that could solve this.

alwa-nordic, 2021-11-03:
@CodingGhost Thank you for your report! A minimal example would be really helpful! Can you provide that?

CodingGhost, 2021-11-24:
sorry, I have been busy and forgot about it. I will try to do it tomorrow.

CodingGhost, 2021-11-25:
the Problem was caused by 
SYS_SLIST_FOR_EACH_CONTAINER_SAFE in att.c  iterating too often (more than there were requests).
this PR fixes it: #39507

thanks to @carlescufi for helping me out!

closed 2021-11-25 by CodingGhost