zephyrproject-rtos/zephyr #40009: Bluetooth: EATT bearer L2CAP collision mitigation

sjanc, 2021-11-02:
**Is your feature request related to a problem? Please describe.**

BT Core Spec:
"If both devices request L2CAP connections (on EATT PSM) simultaneously and both devices
have limited resources, a device may reject the incoming request and find its
own request is also rejected. In this situation, the Central may retry
immediately but the Peripheral shall wait a minimum of 100 ms before retrying;
on LE connections, the Peripheral shall wait at least 2 ×
(connPeripheralLatency + 1) × connInterval if that is longer."

While spec is not requiring any specific behavior when collision occurs (other than not retrying too soon by peripheral), test specification requires that peripheral IUT re-try after mentioned timeout.

**Describe the solution you'd like**
Collision should be detected and either host should retry connection on L2CAP channel.

**Describe alternatives you've considered**
None.

**Additional context**
This is affecting L2CAP/TIM/BV-03-C qualification test case.

sjanc, 2021-11-23:
Some comments on this issue:
 -  since this is a GATT requirement on EATT PSM (and not L2CAP protocol one) it should be implemented in GATT/ATT code (ie not directly visible for application, and not implemented in L2CAP code)
 - I think we could have generic timer for establishing EATT as peripheral and then it can also be reused by collision handling
 - in GATT/ATT we can have simple counter on how many retries
 - timer can be possibly randomized (eg min_time + some rand) to improve IOP

carlescufi, 2022-01-12:
@sjanc what's the status here? Still planning an implementation for the 2.7.x series?

sjanc, 2022-01-13:
Hi,  yes it is on my TODO,  I'll work on that this month

hermabe, 2022-01-25:
I am currently working on this.

carlescufi, 2022-01-26:
> I am currently working on this.

OK, but in `main` I assume? and then you'll backport to v2.7-branch?

sjanc, 2022-01-26:
I can handle 2.7 backport if needed, once L2CAP/TIM/BV-03-C is passing

hermabe, 2022-01-26:
> in main I assume? 

Yes, working on top of `main` where we do other EATT work.

Vudentz, 2022-02-02:
> Some comments on this issue:
>
> * since this is a GATT requirement on EATT PSM (and not L2CAP protocol one) it should be implemented in GATT/ATT code (ie not directly visible for application, and not [...]

Im curious about how we come to the conclusion it is a GATT requirement? L2CAP/TIM/BV-03-C seem to be L2CAP specific as well, that said I don't see any new error that can be used for collision detection or the stack is supposed to detect this based on the ongoing connection matching the PSM and checking if all channels are refused with 0x0004 (Connection refused – no resources available) on the response? It would have been a lot clearer if a new error was introduced that way both stacks can confirm a collision has been detected and the retry procedure can start.

sjanc, 2022-02-03:
Hi,

Yes, those tests are in L2CAP TS but are related to GATT:
"4.15 Generic Attribute Timing tests
Test cases in this group check that the IUT respects the GATT timings."

You may check TSE17840 as well

PR #42268 (2022-02-22): Bluetooth: Host: L2CAP collision mitigation
This implements L2CAP collision mitigation as required by Core Spec Version 5.3 Vol. 3, Part G, Section 5.4, in addition to a test that verifies it. This fixes conformance test L2CAP/TIM/BV-03-C, although changes to auto-pts and the tester application are required.

Fixes #40009 

---
FYI: @sjanc
closed 2022-02-22 by nashif