zephyrproject-rtos/zephyr #40021: mimxrt1060_evk_hyperflash board definition is broken

henrikbrixandersen, 2021-11-02:
**Describe the bug**
A problem was introduced with #38676 causing the `mimxrt1060_evk_hyperflash` board definition (and perhaps others) to break.

**To Reproduce**
Steps to reproduce the behavior:
1. `west -v build -b mimxrt1060_evk_hyperflash samples/drivers/flash_shell`
2. See error:

```
ZEPHYR_BASE=/home/brix/Projects/zephyrproject/zephyr (origin: env)
-- west build: generating a build system
cmake version 3.21.3 is OK; minimum version is 3.13.1
Running CMake: /home/brix/Projects/zephyrproject/venv/bin/cmake -DWEST_PYTHON=/home/brix/Projects/zephyrproject/venv/bin/python3 -B/home/brix/Projects/zephyrproject/zephyr/build -S/home/brix/Projects/zephyrproject/zephyr/samples/drivers/flash_shell -GNinja -DBOARD=mimxrt1060_evk_hyperflash
Including boilerplate (Zephyr base): /home/brix/Projects/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/brix/Projects/zephyrproject/zephyr/samples/drivers/flash_shell
-- Zephyr version: 2.7.99 (/home/brix/Projects/zephyrproject/zephyr), build: v2.7.99-917-g71c0cebac904
-- Found Python3: /home/brix/Projects/zephyrproject/venv/bin/python3 (found suitable exact version "3.8.10") found components: Interpreter 
-- Found west (found suitable version "0.11.1", minimum required is "0.7.1")
-- Board: mimxrt1060_evk_hyperflash
-- Cache files will be written to: /home/brix/.cache/zephyr
-- Using toolchain: zephyr 0.13.1 (/opt/zephyr-sdk)
-- Found dtc: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/brix/Projects/zephyrproject/zephyr/boards/arm/mimxrt1060_evk/mimxrt1060_evk_hyperflash.dts
-- Generated zephyr.dts: /home/brix/Projects/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/brix/Projects/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/brix/Projects/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/brix/Projects/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/brix/Projects/zephyrproject/zephyr/samples/drivers/flash_shell/Kconfig
Loaded configuration '/home/brix/Projects/zephyrproject/zephyr/boards/arm/mimxrt1060_evk/mimxrt1060_evk_hyperflash_defconfig'
Merged configuration '/home/brix/Projects/zephyrproject/zephyr/samples/drivers/flash_shell/prj.conf'
Configuration saved to '/home/brix/Projects/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/brix/Projects/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Error at /home/brix/Projects/zephyrproject/zephyr/cmake/extensions.cmake:2708 (message):
  dt_prop(compat_flash ...) missing required argument: PATH
Call Stack (most recent call first):
  /home/brix/Projects/zephyrproject/zephyr/drivers/flash/CMakeLists.txt:30 (dt_prop)

-- Configuring incomplete, errors occurred!
See also "/home/brix/Projects/zephyrproject/zephyr/build/CMakeFiles/CMakeOutput.log".
See also "/home/brix/Projects/zephyrproject/zephyr/build/CMakeFiles/CMakeError.log".
FATAL ERROR: command exited with status 1: /home/brix/Projects/zephyrproject/venv/bin/cmake -DWEST_PYTHON=/home/brix/Projects/zephyrproject/venv/bin/python3 -B/home/brix/Projects/zephyrproject/zephyr/build -S/home/brix/Projects/zephyrproject/zephyr/samples/drivers/flash_shell -GNinja -DBOARD=mimxrt1060_evk_hyperflash

```
**Expected behavior**
Board definition should not be broken.

**Impact**
This affects CI for other PRs (see e.g. https://buildkite.com/zephyr/zephyr/builds/45183)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK,
 - Commit SHA: 71c0cebac9049ef2ce5ea377681ac8140a341598

henrikbrixandersen, 2021-11-02:
@Kludentwo FYI.

Kludentwo, 2021-11-02:
I know exactly why:
To determine which flexspi device we use the chosen zephyr,flash but mimxrt1060_evk_hyperflash.dts deletes this node.
Do you want me to make a fix for it?

henrikbrixandersen, 2021-11-02:
> Do you want me to make a fix for it?

Yes, please do.

PR #40022 (2021-11-03): boards: imxrt10x0_evk_xxxx: fixes to alternative flexspi configurations
This fixes the build issue with chosen zephyr,flash missing.

Fixes: #40021
closed 2021-11-03 by carlescufi