zephyrproject-rtos/zephyr #40030: STM32 SD hardware flow control gets disabled if disk_access_init is used

benediktibk, 2021-11-03:
**Describe the bug**
Recently the config SDMMC_STM32_HWFC has been added via #39364 to solve #39136. Unfortunately, the corresponding register get overriden if disk_access_init is used.

**To Reproduce**
1. Activate SDMMC_STM32_HWFC
2. Call disk_access_init

**Expected behavior**
It should be possible to use disk_access_init in combination with SDMMC_STM32_HWFC.

**Impact**
It is necessary to set the register manually again as a workaround.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
Commit 86165326c5e339b0c04c68253ce917a9973fddb9

**Additional context**
I already describe the issue in more detail in https://github.com/zephyrproject-rtos/zephyr/pull/39364/commits/86165326c5e339b0c04c68253ce917a9973fddb9, but this note unfortunately got overlooked, I suppose.

erwango, 2021-11-03:
> I already describe the issue in more detail in 8616532, but this note unfortunately got overlooked, I suppose.

I can't find any comment related to this. Nvm, I'm looking into it.

benediktibk, 2021-11-03:
There is one open review from me here: https://github.com/zephyrproject-rtos/zephyr/pull/39364

erwango, 2021-11-03:
> There is one open review from me here: #39364

Sorry, but I can't see it. Is it possible that it is not committed on your side ?
Anyway, I've just pushed https://github.com/zephyrproject-rtos/zephyr/pull/40034.

Would you mind having a test, I don't have required hw atm.

benediktibk, 2021-11-03:
Weird. Just for completeness, this was the comment:

Let me explain it a little bit further:

    zephyr initialization start
    disk_stm32_sdmmc_init
    zephyr initialization end
    main start
    disk_access_init
    stm32_sdmmc_access_init
    HAL_SD_init
    disk_access_ioctl

In my own application I am calling disk_access_init and disk_access_ioctl, the other calls in between are implicit function calls. With this PR after disk_stm32_sdmmc_init the hardwareflow control is enabled, but disabled again after HAL_SD_Init.

benediktibk, 2021-11-03:
It tested it already, it solves the issue for me, Thanks on the quick fix.

Although I am still confused about the relation of disk_stm32_sdmmc_init and stm32_sdmmc_access_init, because it does not look like one calls the other one. Hence, I'm uncertain if this will work now if a application does not call disk_access_init explictly. Or is this mandatory anway?

erwango, 2021-11-03:
Tbh, I'm not quite clear on the partition between these 2 functions (aside from the obvious stuff such as pure driver init done in  `disk_stm32_sdmmc_init`): thread start, semaphores, pinctrl, i/f registration. This for sure makes sense to be done early.
Anyway, despite the poor API documentation in disk.h, it seems right to assume that 'init' should be called before any other read/write operation. So I hope we should be safe from that point of view.

PR #40034 (2021-11-04): drivers/disk: sdmmc: stm32: Enable hw fc only after card init
During SDMMC card init, HW_FC is disabled by default, overwriting driver configuration.
To avoid this, move HW FC configuration after card init.

Fixes #40030
closed 2021-11-04 by cfriedt