zephyrproject-rtos/zephyr #40065: LoRaWAN - Allow runtime changing of region

ballardr, 2021-11-04:
**Is your enhancement proposal related to a problem? Please describe.**
Currently the LoRaWAN region is set as a kconfig and is not able to be changed during runtime. However, I'd like to have this changed so it can be set at runtime, ideally prior to a join. This would allow a single firmware to target multiple "regions". Some example use cases:
* in Australia, both AS923 and AU915 are common and as such it would be useful to allow for a simple device configuration change to change between networks
* A single firmware can be developed for multiple markets (eg Europe and Australia) with the hardware being different but the firmware identical

**Describe the solution you'd like**
I think a function that gets called prior to the first join request after boot is the best idea. This could replace the kconfig or the kconfig simple acts as the default if the function has not been called.

**Describe alternatives you've considered**
The alternative is to have separate firmware for each "region" which is much harder to manage and a little overkill.

ballardr, 2021-12-13:
@Mani-Sadhasivam, I'm just wondering if you have any update?

Mani-Sadhasivam, 2021-12-15:
@ballardr Sorry for the long delay. Not getting enought spare time these days...

For changing the region dynamically, we could make use of `LoRaMacDeInitialization` API provided by loramac-node. But for compatibility we still need to keep the Kconfig way of specifying the regions. I think a new API that does de-init and init of LoRaMAC with new region should be introduced. And this API can be called anytime during runtime but this should be considered as a MAC layer reset as all information would be lost.

```
int lorawan_set_region(enum lorawan_region region)
{
	LoRaMacStatus_t status;
	LoRaMacRegion_t mac_region;
	
	mac_region = lorawan_region2macregion(region);

	status = LoRaMacDeInitialization();
	if (status != LORAMAC_STATUS_OK) {
		LOG_ERR("LoRaMacDeInitialization failed: %s", lorawan_eventinfo2str(status));
		return lorawan_status2errno(status);
	}
	
	status = LoRaMacInitialization(&macPrimitives, &macCallbacks, mac_region);
	if (status != LORAMAC_STATUS_OK) {
		LOG_ERR("LoRaMacInitialization failed: %s", lorawan_eventinfo2str(status));
		return lorawan_status2errno(status);
	}
	
	return 0;
}
```

What do you think?

kartben, 2023-08-20:
Fixed with #52030

closed 2023-08-20 by kartben