zephyrproject-rtos/zephyr #40068: Test suite subsys.pm.device_runtime_api fail on qemu_x86_64

mengxianglinx, 2021-11-04:
**Describe the bug**
"pm->state" is accessed by multiple functions, only in runtime_suspend_work() it is protected by "pm->lock", for architectures with multiple CPU, the value of "pm->state" is unexpectable.

**To Reproduce**
twister -T tests/subsys/pm/device_runtime_api/ -p qemu_x86_64

**Expected behavior**
test cases pass.

**Impact**
Test coverage report

**Logs and console output**
Running test suite device_runtime_api
===================================================================
START - test_api
W: Unbalanced suspend
W: Unbalanced suspend
W: Unbalanced suspend

Assertion failed at ZEPHYR_BASE/tests/subsys/pm/device_runtime_api/src/main.c:151: test_api: state not equal to PM_DEVICE_STATE_SUSPENDED

FAIL - test_api in 0.5 seconds
===================================================================
Test suite device_runtime_api failed.
===================================================================

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

cgturner1, 2021-11-08:
not sure if this is also related but the same failure is seen for qemu_cortex_a53_smp

gmarull, 2021-11-08:
Could you please provide more details, test is passing on my setup:
```
DEBUG   - QEMU (1359901): SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
DEBUG   - QEMU (1359901): Booting from ROM..*** Booting Zephyr OS build v2.7.99-1182-g0ab1bf42b30e  ***
DEBUG   - QEMU (1359901): Running test suite device_runtime_api
DEBUG   - QEMU (1359901): ===================================================================
DEBUG   - QEMU (1359901): START - test_api
DEBUG   - QEMU (1359901): W: Unbalanced suspend
DEBUG   - QEMU (1359901): W: Unbalanced suspend
DEBUG   - QEMU (1359901): W: Unbalanced suspend
DEBUG   - QEMU (1359901): PASS - test_api in 0.3 seconds
DEBUG   - QEMU (1359901): ===================================================================
DEBUG   - QEMU (1359901): Test suite device_runtime_api succeeded
DEBUG   - QEMU (1359901): ===================================================================
DEBUG   - QEMU (1359901): PROJECT EXECUTION SUCCESSFUL
DEBUG   - QEMU (1359901) complete (passed) after 2.1003847122192383 seconds
DEBUG   - No timeout, return code from QEMU (None): 0
DEBUG   - return code from QEMU (None): 0
DEBUG   - run status: qemu_x86_64/tests/subsys/pm/device_runtime_api/subsys.pm.device_runtime_api passed
INFO    - 1/1 qemu_x86_64               tests/subsys/pm/device_runtime_api/subsys.pm.device_runtime_api PASSED (qemu 2.100s)

```
```
DEBUG   - QEMU (1358717): *** Booting Zephyr OS build v2.7.99-1182-g0ab1bf42b30e  ***
DEBUG   - QEMU (1358717): Secondary CPU core 1 (MPID:0x1) is up
DEBUG   - QEMU (1358717): Running test suite device_runtime_api
DEBUG   - QEMU (1358717): ===================================================================
DEBUG   - QEMU (1358717): START - test_api
DEBUG   - QEMU (1358717): W: Unbalanced suspend
DEBUG   - QEMU (1358717): W: Unbalanced suspend
DEBUG   - QEMU (1358717): W: Unbalanced suspend
DEBUG   - QEMU (1358717): PASS - test_api in 0.6 seconds
DEBUG   - QEMU (1358717): ===================================================================
DEBUG   - QEMU (1358717): Test suite device_runtime_api succeeded
DEBUG   - QEMU (1358717): ===================================================================
DEBUG   - QEMU (1358717): PROJECT EXECUTION SUCCESSFUL
DEBUG   - QEMU (1358717) complete (passed) after 2.0300774574279785 seconds
DEBUG   - No timeout, return code from QEMU (None): 0
DEBUG   - return code from QEMU (None): 0
DEBUG   - run status: qemu_cortex_a53_smp/tests/subsys/pm/device_runtime_api/subsys.pm.device_runtime_api passed
INFO    - 1/1 qemu_cortex_a53_smp       tests/subsys/pm/device_runtime_api/subsys.pm.device_runtime_api PASSED (qemu 2.030s)

```

mengxianglinx, 2021-11-09:
DEBUG   - QEMU (66881): SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)                                                                                                            DEBUG   - QEMU (66881): Booting from ROM..*** Booting Zephyr OS build zephyr-v2.4.0-13068-g5a218decfad8  ***
DEBUG   - QEMU (66881): Running test suite device_runtime_api
DEBUG   - QEMU (66881): ===================================================================
DEBUG   - QEMU (66881): START - test_api
DEBUG   - QEMU (66881): W: Unbalanced suspend
DEBUG   - QEMU (66881): W: Unbalanced suspend
DEBUG   - QEMU (66881): W: Unbalanced suspend
DEBUG   - QEMU (66881):
DEBUG   - QEMU (66881): Assertion failed at ZEPHYR_BASE/tests/subsys/pm/device_runtime_api/src/main.c:151: test_api: state not equal to PM_DEVICE_STATE_SUSPENDED
DEBUG   - QEMU (66881):
DEBUG   - QEMU (66881): FAIL - test_api in 0.6 seconds
DEBUG   - QEMU (66881): ===================================================================
DEBUG   - QEMU (66881): Test suite device_runtime_api failed.
DEBUG   - QEMU (66881): ===================================================================
DEBUG   - QEMU (66881): PROJECT EXECUTION FAILED
DEBUG   - QEMU (66881) complete (failed) after 2.1199686527252197 seconds
DEBUG   - No timeout, return code from QEMU (None): 0
DEBUG   - return code from QEMU (None): 0
DEBUG   - run status: qemu_x86_64/tests/subsys/pm/device_runtime_api/subsys.pm.device_runtime_api failed
INFO    - 1/1 qemu_x86_64               tests/subsys/pm/device_runtime_api/subsys.pm.device_runtime_api FAILED Failed (qemu 2.120s)

mengxianglinx, 2021-11-09:
DEBUG   - QEMU (70346): *** Booting Zephyr OS build zephyr-v2.4.0-13068-g5a218decfad8  ***
DEBUG   - QEMU (70346): Secondary CPU core 1 (MPID:0x1) is up
DEBUG   - QEMU (70346): Running test suite device_runtime_api
DEBUG   - QEMU (70346): ===================================================================
DEBUG   - QEMU (70346): START - test_api
DEBUG   - QEMU (70346): W: Unbalanced suspend
DEBUG   - QEMU (70346): W: Unbalanced suspend
DEBUG   - QEMU (70346): W: Unbalanced suspend
DEBUG   - QEMU (70346):
DEBUG   - QEMU (70346): Assertion failed at ZEPHYR_BASE/tests/subsys/pm/device_runtime_api/src/main.c:151: test_api: state not equal to PM_DEVICE_STATE_SUSPENDED
DEBUG   - QEMU (70346):
DEBUG   - QEMU (70346): FAIL - test_api in 0.9 seconds
DEBUG   - QEMU (70346): ===================================================================
DEBUG   - QEMU (70346): Test suite device_runtime_api failed.
DEBUG   - QEMU (70346): ===================================================================
DEBUG   - QEMU (70346): PROJECT EXECUTION FAILED
DEBUG   - QEMU (70346) complete (failed) after 2.0601963996887207 seconds
DEBUG   - No timeout, return code from QEMU (None): 0
DEBUG   - return code from QEMU (None): 0
DEBUG   - run status: qemu_cortex_a53_smp/tests/subsys/pm/device_runtime_api/subsys.pm.device_runtime_api failed
INFO    - 1/1 qemu_cortex_a53_smp       tests/subsys/pm/device_runtime_api/subsys.pm.device_runtime_api FAILED Failed (qemu 2.060s)

ceolin, 2021-11-09:
There is a race condition in SMP environment. It is possible that,  with 2 concurrent threads, the workqueue runs before the `get_runner` thread. The test is doing assumptions that there is only one core (`get_runner` priority would make it run before the workqueue), as the test is not testing smp nor require it, just changed the test to use only one core.

PR #40188 (2021-11-09): tests: pm: device_runtime: Avoid race condition
The test was done with one cpu in mind. In a SMP environment there is
race condition where the semaphor can be given before the other thread
take block waiting for it.

Fixes #40068
commit d7a2f6e8 (2021-11-09): tests: pm: device_runtime: Avoid race condition

The test was done with one cpu in mind. In a SMP environment there is
race condition where the semaphor can be given before the other thread
take block waiting for it.

Fixes #40068
closed 2021-11-09 by nashif