zephyrproject-rtos/zephyr #40069: Bluetooth CCM encryption bug in MIC generation

nth-eye, 2021-11-04:
**Describe the bug**
bt_ccm_decrypt() returns -EBADMSG after trying to decrypt the same data that was encrypted with bt_ccm_ecnrypt().

**To Reproduce**
Steps to reproduce the behavior:
1. Encrypt any data with MIC length greater than 0 and less than 16
2. Decrypt the result with the same MIC length
3. See error

Source code for example (test vector is taken from https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz-group/sub-1-ghz/f/sub-1-ghz-forum/917432/cc1310-aes-ccm-tags-and-nist-test-vectors***)

\`\`\`
#define AES_MIC_SIZE 4
uint8_t key[16] = {0x12, 0x34, 0x56, 0x78, 0x9a, 0xbc, 0xde, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};
uint8_t nonce[13] = {0x00, 0x00, 0xf0, 0xe0, 0xd0, 0xc0, 0xb0, 0xa0, 0x00, 0x00, 0x00, 0x00, 0x05};
uint8_t payload[20] = {0x14, 0xaa, 0xbb, 0x00, 0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0a, 0x0b, 0x0c, 0x0d, 0x0e, 0x0f};

uint8_t encrypted[24];
uint8_t decrypted[20];

bt_ccm_encrypt(key, nonce, payload, 20, NULL, 0, encrypted, AES_MIC_SIZE);

int ret = bt_ccm_decrypt(key, nonce, encrypted, 20, NULL, 0, decrypted, AES_MIC_SIZE);

if (ret) {
    printk("failed to decrypt -> %d", ret);
    return 0;
}
\`\`\`

**Expected behavior**
Decryption should work returning 0.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: nRF Connect SDK 1.7
 - Commit SHA or Version used

**Solution**

FIx bt_ccm_encrypt() function in aes_ccm.c file:

Replace
ccm_auth(key, nonce, enc_data, len, aad, aad_len, mic, mic_size);
with
ccm_auth(key, nonce, plaintext, len, aad, aad_len, mic, mic_size);

carlescufi, 2021-11-04:
@nth-eye could you send a Pull Request please?

joerchan, 2021-11-04:
@trond-snekvik Looked into it and the bug-report seems to be correct. Mesh is however not affected by this as it is providing the plaintext twice in the call to bt_ccm_encrypt, so enc_data in the call will also be plaintext.

So using bt_ccm_encrypt with enc_data not pointing to plaintext is broken.

PR #40071 (2021-11-08): Bluetooth: host: Fix MIC generation in Bluetooth CCM encryption
bt_ccm_encrypt only works when encrypting in place. To fix this ccm_auth() inside bt_ccm_encrypt() must take plaintext instead of enc_data, to not rely on assumption that plain and cypher data are the same memory.

Fixes: #40069
commit 2cab8772 (2021-11-08): Bluetooth: host: Fix MIC generation in Bluetooth CCM encryption

bt_ccm_encrypt only works when encrypting in place. To fix this
ccm_auth() inside bt_ccm_encrypt() must take plaintext instead of
enc_data, to not rely on assumption that plain and cypher data are the
same memory.

Fixes: #40069
commit 07680b27 (2021-11-30): Bluetooth: host: Fix MIC generation in Bluetooth CCM encryption

bt_ccm_encrypt only works when encrypting in place. To fix this
ccm_auth() inside bt_ccm_encrypt() must take plaintext instead of
enc_data, to not rely on assumption that plain and cypher data are the
same memory.

Fixes: #40069
closed 2021-11-08 by cfriedt