zephyrproject-rtos/zephyr #40074: sara-r4: socket call fails due to regression

emillindq, 2021-11-04:
**Describe the bug**
When using Ublox SARA-R4 driver with echo_server, UDP `socket()` call fails with 
```
[00:00:38.806,000] <err> modem_ublox_sara_r4: CONFIG_NET_SOCKETS_OFFLOAD must be enabled for this driver
[00:00:38.806,000] <err> net_echo_server_sample: Failed to create UDP socket (IPv4): 134
```
commit which introduced this problem: 8b851af02a4fd6eb975924d78170f3695cc550de
running the commit before (f4a5c7d680f16d9e062d8608878b309fb71adabb) it works well

I'm running echo_server with IPV6 & TCP disabled, with the Sparkfun SARA-R4 shield on STM32H747I-DISCO eval board.
Note that CONFIG_NET_SOCKETS_OFFLOAD is indeed enabled when checking autoconf.h

I suspect this problem exists for other modems as well.

Could you take a look at this @rlubos since this is your commit?

Thank you!
**To Reproduce**
Steps to reproduce the behavior:
1. Disable IPV6 & TCP in echo_server sample
2. Disable ethernet interface in stm32h747i_disco_m7 by removing L2_ETHERNET in board's Kconfig.defconfig & removing the mac in dts.
4. Add `arduino_serial` in front of uart8 like this `arduino_serial: &uart8 {`
3. `west build -b stm32h747i_disco_m7 zephyr/samples/net/sockets/echo_server/ -- -DSHIELD=sparkfun_sara_r4`
4. `west flash`

**Expected behavior**
Socket is created successfully

**Impact**
Showstopper since LTE is no longer working

**Logs and console output**
[image]

**Environment (please complete the following information):**
 - Using various version, but problem exist on latest (ab83b6e850b10d03b28be4544040443960841fa1)

rlubos, 2021-11-04:
@emillindq I don't have the actual hardware to test it, but it seems that the problem is that both native and offloaded socket implementations have the same priority (and in this case the processing priority is determined by the "socket name" provided in the initialization.  

Two possible solutions for the issue:
- disable native stack with `CONFIG_NET_NATIVE=n` when using Ublox SARA-R4, or
- (probably preferred) increase Ublox SARA-R4 socket processing priority [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/modem/ublox-sara-r4.c#L1975), so that it is processed before the native one. Note that lower number is higher priority.

emillindq, 2021-11-04:
@rlubos thank you for your input! Indeed, increasing the socket processing prio for the modem and it works.

Should I make a separate Kconfig symbol for the SARA-R4 socket processing prio and then use that one, or how should we resolve this? I can make a PR

rlubos, 2021-11-04:
Could be a Kconfig option or just a define, for NRF91 we've just added a define and build assert to make sure the priority is ok (https://github.com/nrfconnect/sdk-nrf/blob/main/lib/nrf_modem_lib/nrf91_sockets.c#L1394)

PR #40080 (2021-11-09): sara-r4: increase socket processing prio
Higher socket processing prio is required for offloaded sockets

Resolves #40074 

Ping @rlubos
commit 38eed4a4 (2021-11-09): sara-r4: increase socket processing prio

Higher socket processing prio is required for offloaded sockets.

Resolves #40074
closed 2021-11-09 by cfriedt