zephyrproject-rtos/zephyr #40077: driver: wdt: twrke18f: test_wdt fails

hakehuang, 2021-11-04:
**Describe the bug**

 `Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/watchdog/wdt_basic_api/src/test_wdt.c:362: test_wdt: test_wdt_callback_1() == TC_PASS is false`

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=twr_ke18f ..
3. make
4. make flash

**Expected behavior**
wdt testing pass

**Impact**
watchdog support

**Logs and console output**

   ```
 Running test suite wdt_basic_test

    ===================================================================

    START - test_wdt

    Testcase: test_wdt_no_callback

    Waiting to restart MCU

    Running test suite wdt_basic_test

    ===================================================================

    START - test_wdt

    Testcase: test_wdt_no_callback

    Testcase passed

    Testcase: test_wdt_callback_1

    Waiting to restart MCU

    Running test suite wdt_basic_test

    ===================================================================

    START - test_wdt

    Testcase: test_wdt_callback_1

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/watchdog/wdt_basic_api/src/test_wdt.c:362: test_wdt: test_wdt_callback_1() == TC_PASS is false

    

     FAIL - test_wdt in 0.17 seconds

    ===================================================================

    Test suite wdt_basic_test failed.

    ===================================================================

    PROJECT EXECUTION FAILED

```
**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version: v2.7.99-878-gc03519fffc6a

henrikbrixandersen, 2021-11-04:
Hi @hakehuang,

I am unable to reproduce this failure on the twr_ke18f locally (SHA: 608b358d6bcd3262fcf6a8839a463be42bca1201):

```
Running test suite wdt_basic_test
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running test suite wdt_basic_test
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Testcase passed
Testcase: test_wdt_callback_1
Waiting to restart MCU
Running test suite wdt_basic_test
===================================================================
START - test_wdt
Testcase: test_wdt_callback_1
Testcase passed
Testcase: test_wdt_bad_window_max
E: Invalid timeout
 PASS - test_wdt in 0.9 seconds
===================================================================
Test suite wdt_basic_test succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

hakehuang, 2021-11-05:
@henrikbrixandersen thanks for quick feedbakc, I find there is a probobilities on this issue. some time it will hapeen. but let me close it for now

closed 2021-11-05 by hakehuang