zephyrproject-rtos/zephyr #40107: lwm2m: if network drops during firmware update, lock occurs

rerickson1, 2021-11-04:
**Describe the bug**
Once a firmware update has been starting and the image is downloading, if the network drops and reconnects a lock occurs.

**To Reproduce**
1. Start a firmware update.
2. Once the firmware is downloading, break the network connection.
3. At this point the firmware stops downloading and the lock has not occurred. For example, you can still interact with UART shell and logging messages are reported.
4. Bring back the network connection. As soon as the interface is up, a lock occurs. UART shell no longer responds and logging system is dead as well.

**Expected behavior**
There should be a couple retries and then the download should be cancelled.

**Impact**
If not using a watchdog, this is a showstopper.

**Environment (please complete the following information):**
 - OS: macOS
 - Toolchain Zephyr SDK 0.13.1

rlubos, 2021-11-22:
What platform do you use? Is this reproducible in some emulated environment like qemu?

I'm wondering if it could be an issue with some underlying net stack/driver implementation. The unresponsive log/shell subsystem  suggests that a busy loop occurs somewhere in the system after the connection is back.

As for the LwM2M subsystem, the download mechanism is implemented so that the download should be aborted after serveral failures. If a timeout occurs, the [timeout callback](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/lwm2m/lwm2m_obj_firmware_pull.c#L341) will close the socket and report a download failure after several failed attempts. And if we get an error from `poll()` or `recv()` functions, the [fault handler](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/lwm2m/lwm2m_obj_firmware_pull.c#L420) should also abort the download if it fails to reopen the socket or resend the request.

rerickson1, 2021-11-22:
I have reproduced this error on our MG100 or Pinnacle 100 DVK platforms using this software release based on nRF Connect SDK 1.7.0: https://github.com/LairdCP/Pinnacle-100-Firmware-Manifest/releases
I have not tried in emu.

rerickson1, 2021-11-30:
I hooked up a debugger and found out a little.
When the LTE network interface goes down I call the API to stop the LwM2M engine. I have a feeling, it may not be stopping everything its supposed to:
```
[00:01:27.821,441] <inf> lwm2m_sensor: TEMPERATURE idx: 0 base: 60800 RSSI: -68
[00:01:28.185,943] <dbg> modem_hl7800.send_at_cmd: OUT: [AT+KUDPSND=2,"52.32.209.142",5784,67]
[00:01:28.202,148] <dbg> modem_hl7800.hl7800_rx: HANDLE CONNECT (len:0)
[00:01:28.207,977] <dbg> modem_hl7800.send_data: Sent 67 bytes
[00:01:28.216,461] <dbg> modem_hl7800.hl7800_rx: HANDLE OK (len:0)
[00:01:29.512,481] <dbg> modem_hl7800.hl7800_rx: HANDLE +CEREG:  (len:1)
[00:01:29.512,512] <dbg> modem_hl7800.on_cmd_network_report: Network status: 4
[00:01:29.512,512] <inf> modem_hl7800: Network State: 4 OUT_OF_COVERAGE
[00:01:29.512,603] <inf> attr: [177] lteNetworkState               4 Out Of Coverage
[00:01:30.012,817] <dbg> modem_hl7800.iface_status_work_cb: Updating network state...
[00:01:30.012,817] <dbg> modem_hl7800.send_at_cmd: OUT: [AT+COPS?]
[00:01:30.015,991] <dbg> modem_hl7800.hl7800_rx: HANDLE +COPS:  (len:1)
[00:01:30.015,991] <inf> modem_hl7800: Operator: 0
[00:01:30.016,510] <dbg> modem_hl7800.hl7800_rx: HANDLE OK (len:0)
[00:01:30.016,540] <dbg> modem_hl7800.iface_status_work_cb: HL7800 iface DOWN
[00:01:30.016,601] <dbg> modem_hl7800.iface_status_work_cb: Network state updated
[00:01:30.156,158] <inf> attr: [155] gatewayState                  4 Network Disconnected
[00:01:31.158,996] <dbg> modem_hl7800.send_at_cmd: OUT: [AT+KUDPCLOSE=2]
[00:01:31.166,473] <dbg> modem_hl7800.hl7800_rx: HANDLE OK (len:0)
[00:01:31.166,656] <inf> net_lwm2m_rd_client: Stop LWM2M Client: dvk_354616090303848
[00:01:31.166,870] <inf> attr: [155] gatewayState                  10 Cloud Wait For Disconnect
[00:01:32.167,114] <inf> attr: [155] gatewayState                  11 Cloud Disconnected
[00:01:33.167,358] <inf> attr: [155] gatewayState                  2 Wait For Network
[00:01:38.850,097] <err> net_lwm2m_engine: Failed to send packet, err 115
[00:01:39.817,047] <inf> lwm2m_sensor: TEMPERATURE idx: 3 base: 36108 RSSI: -91
[00:01:49.980,957] <inf> attr: [175] powerSupplyVoltage            4.0470e+00
[00:01:56.647,216] <inf> lwm2m_sensor: TEMPERATURE_1 idx: 4 base: 41524 RSSI: -77
[00:02:04.980,957] <inf> attr: [175] powerSupplyVoltage            4.0530e+00
[00:02:15.596,984] <inf> lwm2m_sensor: ULTRASONIC_1 idx: 2 base: 44144 RSSI: -61
[00:02:15.597,106] <dbg> lwm2m_sensor.set_filling_sensor_data: height: 1000 level: 0 measured distance: 1006 percent: 0
[00:02:16.458,007] <inf> lwm2m_sensor: CURRENT_1 idx: 1 base: 32792 RSSI: -69
[00:02:18.806,152] <err> net_lwm2m_engine: Failed to send packet, err 115
```

This is the log output I see after the network comes back and logging freezes along with the UART terminal:
```
[00:02:41.778,747] <dbg> modem_hl7800.hl7800_rx: HANDLE +CEREG:  (len:21)
[00:02:41.778,778] <dbg> modem_hl7800.on_cmd_network_report: Network status: 1,"0001","01A2D101",7
[00:02:41.778,778] <inf> modem_hl7800: Network State: 1 HOME_NETWORK
[00:02:41.778,900] <inf> attr: [177] lteNetworkState               1 Home Network
[00:02:41.779,113] <dbg> modem_hl7800.send_at_cmd: OUT: [AT+CCLK?]
[00:02:41.783,966] <dbg> modem_hl7800.hl7800_rx: HANDLE +CCLK:  (len:22)
[00:02:41.783,996] <dbg> modem_hl7800.wait_for_modem_data: cur:23, exp:25
[00:02:41.834,106] <inf> modem_hl7800: RTC string: '"21/11/30,14:52:10-24"'
[00:02:41.834,167] <dbg> modem_hl7800.hl7800_rx: HANDLE OK (len:0)
[00:02:41.834,259] <inf> lte: Epoch set to 1638305530
[00:02:42.040,008] <dbg> modem_hl7800.hl7800_rx: HANDLE +CEREG:  (len:1)
[00:02:42.040,039] <dbg> modem_hl7800.on_cmd_network_report: Network status: 0
[00:02:42.040,039] <inf> modem_hl7800: Network State: 0 NOT_REGISTERED
[00:02:42.040,161] <inf> attr: [177] lteNetworkState               0 Not Registered
[00:02:42.222,564] <dbg> modem_hl7800.hl7800_rx: HANDLE +CEREG:  (len:1)
[00:02:42.222,564] <dbg> modem_hl7800.on_cmd_network_report: Network status: 2
[00:02:42.222,564] <inf> modem_hl7800: Network State: 2 SEARCHING
[00:02:42.222,686] <inf> attr: [177] lteNetworkState               2 Searching
[00:02:42.227,294] <dbg> modem_hl7800: UNHANDLED RX
                                       2b 4b 43 4e 58 5f 49 4e  44 3a 20 31 2c 30 2c 31 |+KCNX_IN D: 1,0,1
[00:02:42.229,034] <dbg> modem_hl7800: UNHANDLED RX
                                       2b 4b 43 4e 58 5f 49 4e  44 3a 20 31 2c 30 2c 30 |+KCNX_IN D: 1,0,0
[00:02:42.230,773] <dbg> modem_hl7800.hl7800_rx: HANDLE +KUDP_NOTIF:  (len:3)
[00:02:42.230,773] <wrn> modem_hl7800: +K**P_NOTIF: 1,0
[00:02:42.232,513] <dbg> modem_hl7800.hl7800_rx: HANDLE +KUDP_NOTIF:  (len:3)
[00:02:42.232,513] <wrn> modem_hl7800: +K**P_N
```

In Ozone, I can stop the debugger and this is my call stack:
[image]

The LwM2M engine is still trying to poll the socket that was closed...
It may be executing that loop very fast because of that and hogging all the CPU.

rlubos, 2021-12-01:
> I have a feeling, it may not be stopping everything its supposed to

Current behaviour is that `lwm2m_rd_client_stop()` does not stop the ongoing download. This might not be the right behaviour though, since we don't have a separate API for that, and the FW download module used by the engine is tightly integrated with it. Feel free to send a patch to fix this.

Anyway, even if not stopped, I don't think the FW download process shall cause any harm, other that some `send()` errors in the logs. The root cause might be somewhere else.

> The LwM2M engine is still trying to poll the socket that was closed...
It may be executing that loop very fast because of that and hogging all the CPU.

This sounds suspicious, why would the closed socket be polled, in lwm2m we remove the socket from the fd table before it's closed. What is `poll()` returning when the loop is spinning, is this `poll()` error or does it rather keep returning some event on one of the sockets (which one?)?

rlubos, 2021-12-01:
One possible theory for busy-looping inside LwM2M engine loop - it could happen if the socket is in EOF state (which is not right for the datagram socket), in that case `poll()` returns `POLLIN` immediately. This could only take place though if the `hl7800` driver called `recv_cb()` with a NULL packet pointer - could this happen with this driver? Perhaps we should protect against that somewhere in the `sockets.c` implementation if that's the case.

carlescufi, 2021-12-15:
@rerickson1 ping on this one

rerickson1, 2021-12-15:
I won't be able to spend any time digging into this further until after the new year.

carlescufi, 2022-01-26:
@rerickson1 ping again

rerickson1, 2022-02-03:
taking a look at this now

rerickson1, 2022-02-03:
So this issue seems to be triggered by the HL7800 modem driver. I tested with the [bl5340_dvk](https://github.com/zephyrproject-rtos/zephyr/tree/main/boards/arm/bl5340_dvk) by pulling the ethernet cable in the middle of the download and the issue did not happen.

PR #42469 (2022-02-04): modem: hl7800: only send null packet for TCP
Only TCP sockets should send a NULL packet
if the server closes the connection or there
is a socket error.
UDP sockets do not need to do this because
they are connectionless.

fixes #40107
closed 2022-02-04 by carlescufi