zephyrproject-rtos/zephyr #40114: modules/crypto/mbetls: missing inclusion of the  library directory

nvlsianpu, 2021-11-05:
** the bug:**
After migration to mbedTLS 3.0.0 its `library` directory contains some headers which might be needed by the application.

For insentience MCUboot might require rsa_alt_helpers.h in order
to be build with RSA image encryption feature.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -d build -b nrf52840dk_nrf52840 bootloaer/mcuboot/boot/zephyr -- -DCONFIG_BOOT_ENCRYPT_RSA=y

See build faliure on missing "rsa_alt_helpers.h" header

**Expected behavior**
application is building well

**Impact**
Can't use some of mbedTLS features in the application

**Environment (please complete the following information):**
SHA https://github.com/zephyrproject-rtos/zephyr/commit/963d05bd525766701cbce44d0aaa63f57e26ae9b

nvlsianpu, 2021-11-08:
Mentioned directory shouldn't be inclusion directory as it contains mbedtls private header. Is an application requires any of such header - it should include it on it's own.

closed 2021-11-08 by nvlsianpu

darexa, 2022-02-18:
> Mentioned directory shouldn't be inclusion directory as it contains mbedtls private header. Is an application requires any of such header - it should include it on it's own.

Ran across exactly this problem just now.
I'm a little confused regarding the response.
I'm trying to enable RSA encryption in mcuboot and doing the same thing as you leads to the same issue.
Are you saying that enabling this CONFIG_BOOT_ENCRYPT_RSA also means I have to manually include the "rsa_alt_helpers.h" somehow? Or something else?