zephyrproject-rtos/zephyr #40119: OBJECT_TRACING for kernel objects

rdienstbeck, 2021-11-05:
Unfortunately, the object tracing for kernel objects was dropped from version 2.5 to 2.6. 

Our (Lauterbach) debuggers provide a Zephyr awareness, that allows to list and inspect threads and kernel objects. We used the linked list created by CONFIG_OBJECT_TRACING to show tables of created objects, e.g. semaphores or mutexes. Since Zephyr version 2.6, this linked list is no longer configurable, which makes it impossible for debuggers to find and display created objects.

It would be great to get the linked list back in the kernel objects, e.g. like it was in k_mutex of 2.5:
https://github.com/zephyrproject-rtos/zephyr/blob/8be935fe5cda14db37549446e2d571b485cddf0e/include/kernel.h#L3087

dleach02, 2021-11-05:
@rdienstbeck, can you bisect to the exact changeset that broke this feature for you?

rdienstbeck, 2021-11-05:
@dleach02, thanks for looking into this. The bisect was easy, the change log actually says "kernel: remove object tracing":
https://github.com/zephyrproject-rtos/zephyr/commit/4d994af03284ce94616c00c741e21b9b22e8ff62#diff-d425c69d5282cb213b62e545f89a0aa44326d0187725db1d83c8209a9bb1585f
Change from May 8 2021.

dleach02, 2021-11-05:
Okay. So it was removed with this PR #34843

tleksell-pe, 2021-11-08:
Interesting, I don't think there should be any problems re-adding this kind of feature as a "new" trace format within the updated tracing system that was introduced in PR #34843.

nashif, 2021-11-10:
Hi @rdienstbeck, and nice to meet you again, we often meet at Embedded World ;-)
There were issues with the design of the original feature and realized the new tracing support added in 2.5 would mostly replace that, but I guess this will not replace the availability of this data in the debugger. Will take a look and see how we can solve this.

rdienstbeck, 2021-11-10:
Hi @nashif, I remember our talks at EW, nice to talk to you again!
Thanks for looking into this, customers are already complaining...

edersondisouza, 2021-12-07:
@rdienstbeck have you had the chance to look at #40606 ?

rdienstbeck, 2021-12-07:
Hi @edersondisouza ,
I had a quick look at it, implemented it and tested it in our debugger. It looks like this includes exactly what we need. Thanks a lot for this!
I will modify our Zephyr Awareness to use the linked list as defined in #40606 and will report my results here.
It's now part of CONFIG_TRACING, which also offers a lot of options (and additional code) for tracing, while it is not exactly for tracing but only for keeping a list of objects. Me, personally, I'd put it to kernel debugging options rather than to tracing, but then I'm not a Zephyr expert...

dleach02, 2021-12-07:
@rdienstbeck, recommend you provide your feedback/results directly to #40606. When that gets approved/merged it will automatically close this issue

PR #40606 (2021-12-14): kernel: Bring back object tracking
~~[Still missing some documentation comments]~~

When CONFIG_TRACING_OBJECT_TRACKING is enabled, the kernel will keep
lists of some objects (detailed below), so that debuggers or other tools
can keep track of them.

The lists of objects are:

struct k_timer *_track_list_k_timer;
struct k_mem_slab *_track_list_k_mem_slab;
struct k_sem *_track_list_k_sem;
struct k_mutex *_track_list_k_mutex;
struct k_stack *_track_list_k_stack;
struct k_msgq *_track_list_k_msgq;
struct k_mbox *_track_list_k_mbox;
struct k_pipe *_track_list_k_pipe;
struct k_queue *_track_list_k_ [... cut]
closed 2021-12-14 by nashif