zephyrproject-rtos/zephyr #40129: tests/net/socket/tls/net.socket.tls.preempt fails with qemu_cortex_a9

carlescufi, 2021-11-05:
This commit: 798588e7092741996a5e27e4d0f411e60a35a606 breaks the preempt version of the tls socket test.

Reproduce:

```
scripts/twister -v -N -p qemu_cortex_a9 -T tests/net/socket/tls
```

stephanosio, 2021-11-05:
@ibirnbaum FYI

After the full series in #38603, only `qemu_cortex_a9` fails; so, it could be Cortex-A arch port side issue.

stephanosio, 2021-11-05:
> As per @rlubos, the issue is this timeout is too short in the case of qemu_cortex_a9.

https://github.com/zephyrproject-rtos/zephyr/blob/6432289baf44d27669a04f027fa6482068670d4a/boards/arm/qemu_cortex_a9/board.cmake#L7

`qemu_cortex_a9` uses the Xilinx QEMU (qemu fork by Xilinx) that is much slower than the upstream QEMU, so it makes sense that `qemu_cortex_a9`, being the only network-capable emulated platform that uses the Xilinx QEMU, performs much slower than the rest.

In that sense, it is justifiable to increase the timeout in this test.

tbursztyka, 2021-11-08:
So this test was not run in the PR I made? because CI did not complain back then.

tbursztyka, 2021-11-08:
note that introducing 798588e generated timing issues: the inner TCP thread sending the packets was too low in priority. And it could generate these timeouts on some tests. Thus another commit raising the inner thread priority. But looks like this is too brittle then if some test fail from time to time. (Again note that CI did not find anything at the time the PR was merged so was all the test ran or some still missing?)

ibirnbaum, 2021-11-08:
Somehow the ticks per second value of the board plays into this issue as well. I initially set up the board with 100 ticks per second, with this value, the unmodified test code reliably fails. Setting the ticks per second value to 1000 leads to the test case passing reliably. Fiddling further with the value revealed that it's fail for <= 500 ticks per seconds and pass for >= 600. Maybe, considering the problem mentioned regarding the thread priorities, the low ticks per second value doesn't allow for the required number of context switches?

carlescufi, 2021-11-08:
Can someone propose a fix? @tbursztyka or @rlubos ?

tbursztyka, 2021-11-09:
@ibirnbaum Or it just exposes a TCP scheduling issue. I would mostly bet on this rather than a context switch issue.

@carlescufi I can take a look.

tbursztyka, 2021-11-09:
I rechecked again the TCP scheduler, and so far all seems legit: everything is queued sequentially as it runs under a unique work queue, so no task can supersede another one.

> As per @rlubos, the issue is this timeout is too short in the case of qemu_cortex_a9.

Indeed, seems that 500ms passes then. I noticed various timers here and in other tests. These were done when local TCP replies would occur within the sender's thread (thing I changed due to local connection), I mean this was a known behavior: no context switch, shorter timings for sending packets (locally only) thus shorter timeouts.

So raising the above timeout is the right and only solution.

tbursztyka, 2021-11-09:
@rlubos  I added your signed-off in the commit as you came anyway with origin of the solution.

rlubos, 2021-11-09:
@tbursztyka Thank you for covering this.

PR #40191 (2021-11-09): tests/net: Raise TLS timeout option on waitall checks
Previous timeouts values were done when local TCP replies wourd occur
within sender's thread. This was a known behavior: no context switch,
shorter timings for sending packets (locally only) thus shorter
timeouts.

Such behavior changed when local replies had to go through TCP's work
queue. See commit 798588e7092741996a5e27e4d0f411e60a35a606

Fixes #40129
commit 404f67ab (2021-11-09): tests/net: Raise TLS timeout option on waitall checks

Previous timeouts values were done when local TCP replies wourd occur
within sender's thread. This was a known behavior: no context switch,
shorter timings for sending packets (locally only) thus shorter
timeouts.

Such behavior changed when local replies had to go through TCP's work
queue. See commit 798588e7092741996a5e27e4d0f411e60a35a606

Fixes #40129
closed 2021-11-09 by carlescufi