zephyrproject-rtos/zephyr #40130: Add driver to support ST iis3dwb digital vibration sensor

escherstair, 2021-11-05:
I'm going to develop a new project based on [ST iis3dwb digital vibration sensor](https://github.com/STMicroelectronics/iis3dwb).
I've been evaluating which RTOS better fits my need, and ZephyrOS could be one option.
I see there is a rich set of sensor drivers, but I can't find the driver for iis3dwb.

Is there any plan to add this driver?

erwango, 2021-11-05:
@avisconti ^^

avisconti, 2021-11-05:
I self assigned this item just to remember. I think I can try to add it in the next weeks.

avisconti, 2021-11-23:
@escherstair 
I'm in the process to add iis3dwb support into zephyr.
Just for my knowledge, can you possibly give me more details about the project?

escherstair, 2021-11-23:
It's a POC for a "smart" vibration sensor, based on iis3dwb + some ST Cortex-M processor (TBD)

avisconti, 2021-11-26:
@escherstair Can you possibly try this driver? I'm still testing it with SPI with DMA. Frequency is high 26K7 Hz, and I guess this is the way

escherstair, 2021-12-02:
@avisconti not in the next few days, but I try to do my best.
I agree that DMA is the solution

lucagessi, 2026-01-02:
The driver has been integrated in zephyr. 
There are some issues which talks about it  #90286 for example