zephyrproject-rtos/zephyr #40133: mimxrt1060-evk flash shell command causes shell deadlock

bbilas, 2021-11-05:
**Describe the bug**
After building and flashing smp_sv shell sample on mimxrt1060-evk the `flash` shell command causes a deadlock in the shell.

**To Reproduce**
Build and run smp-svr shell sample application.

**Expected behavior**
The correct flash shell command output:

<pre>
uart:~$ flash
flash: wrong parameter count
flash - Flash shell commands
Subcommands:
  erase  :[<device>] <page address> [<size>]
  read   :[<device>] <address> [<Dword count>]
  test   :[<device>] <address> <size> <repeat count>
  write  :[<device>] <address> <dword> [<dword>...]
</pre>

**Logs and console output**

<pre>
*** Booting Zephyr OS build v2.7.99-931-gf90b24536184  ***
I: Starting bootloader
I: Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
I: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: primary slot
I: Swap type: none
I: Bootloader chainload address offset: 0x10000
I: Jumping to the first image slot
*** Booting Zephyr OS build v2.7.99-931-gf90b24536184  ***

uart:~$ flash

</pre>

**Environment (please complete the following information):**
 - OS: Arch Linux
 - Toolchain Zephyr SDK 0.13.1
 - v2.7.99-931-gf90b24536184

**Additional context**
 Sometimes for unknown reasons, I was also getting kernel panic such as:

<pre>
uart:~$ flash
[00:00:05.980,000] <err> os: ***** USAGE FAULT *****
[00:00:05.980,000] <err> os:   Illegal use of the EPSR
[00:00:05.980,000] <err> os: r0/a1:  0x60029c60  r1/a2:  0x00000001  r2/a3:  0x80007ca4
[00:00:05.980,000] <err> os: r3/a4:  0x00000040 r12/ip:  0x00000000 r14/lr:  0x6001530b
[00:00:05.980,000] <err> os:  xpsr:  0x60000000
[00:00:05.980,000] <err> os: Faulting instruction address (r15/pc): 0x00000040
[00:00:05.980,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:05.980,000] <err> os: Current thread: 0x80002c28 (shell_uart)
[00:00:05.988,000] <err> os: Halting system
</pre>

de-nordic, 2021-11-05:
@bbilas Can you check if the problem occurs with the zephyr/samples/drivers/flash_shell/  sample?

bbilas, 2021-11-05:
Here you go:

<pre>
*** Booting Zephyr OS build v2.7.99-931-gf90b24536184  ***
Found flash controller FLEXSPI.
Flash I/O commands can be run.

uart:~$ 
  clear    device   devmem   flash    flash    help     history  kernel
  log      resize   shell
uart:~$ flash
flash - Flash related commands.
Subcommands:
  erase             :<off> <len>
                     Erase <len> bytes from device offset <off>, subject to
                     hardware page limitations.
  page_count        :Print the number of pages on the flash device.
  page_erase        :<page> [num]
                     Erase [num] pages (default 1), starting at page <page>.
  page_layout       :[start_page] [end_page]
                     Print layout of flash pages in the range [start_page,
                     end_page], which is inclusive. By default, all pages are
                     printed.
  page_read         :<page> <len> OR <page> <off> <len>
                     Read <len> bytes from given page, starting at page offset
                     <off>,or offset 0 if not given. No checks are made that
                     bytes read are all within the page.
  page_write        :<page> <off> <byte1> [... byteN]
                     Write given bytes to given page, starting at page offset
                     <off>. No checks are made that the bytes all fall within
                     the page. Pages must be erased before they can be written.
  read              :<off> <len>
                     Read <len> bytes from device offset <off>.
  set_device        :<device_name>
                     Set flash device by name. If a flash device was not found,
                     this command must be run first to bind a device to this
                     module.
  write             :<off> <byte1> [... byteN]
                     Write given bytes, starting at device offset <off>.
                     Pages must be erased before they can be written.
  write_block_size  :Print the device's write block size. This is the smallest
                     amount of data which may be written to the device, in
                     bytes.
  write_unaligned   :<off> <byte1> [... byteN]
                     Write given bytes, starting at device offset <off>.
                     Being unaligned, affected memory areas are backed up,
                     erased, protected and then overwritten.
                     This command is designed to test writing to large flash
                     pages.
  write_pattern     :<off> <len>
                     Writes a pattern of (0x00 0x01 .. 0xFF 0x00 ..) of
                     length<len> at the offset <off>.
                     Unaligned writing is used, i.e. protection and erasing are
                     automated.This command is designed to test writing to large
                     flash pages.
uart:~$ flash page
  page_count   page_erase   page_layout  page_read    page_write
uart:~$ flash page_count 
[00:00:15.012,000] <err> os: ***** MPU FAULT *****
[00:00:15.012,000] <err> os:   Instruction Access Violation
[00:00:15.012,000] <err> os: r0/a1:  0x6000def4  r1/a2:  0x80002478  r2/a3:  0x8000247c
[00:00:15.012,000] <err> os: r3/a4:  0xf6410f02 r12/ip:  0x00000000 r14/lr:  0x6000d09d
[00:00:15.012,000] <err> os:  xpsr:  0x60000000
[00:00:15.012,000] <err> os: Faulting instruction address (r15/pc): 0xf6410f02
[00:00:15.012,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:15.012,000] <err> os: Current thread: 0x80000230 (shell_uart)
[00:00:15.048,000] <err> os: Halting system

</pre>

de-nordic, 2021-11-05:
OK, so it is not smp_svr related; this reduces the scope.

de-nordic, 2021-11-05:
Can you check if increasing SHELL_STACK_SIZE helps?

bbilas, 2021-11-05:
Increased from 2048 to 4096 but it's still getting kernel panic:

<pre>
*** Booting Zephyr OS build v2.7.99-931-gf90b24536184  ***
Found flash controller FLEXSPI.
Flash I/O commands can be run.

uart:~$ kernel stacks 
0x80000460 sysworkq   (real size 1024): unused 864      usage 160 / 1024 (15 %)
0x80000230 shell_uart (real size 4096): unused 3184     usage 912 / 4096 (22 %)
0x80000178 logging    (real size 768):  unused 664      usage 104 / 768 (13 %)
0x800002e8 idle 00    (real size 320):  unused 272      usage 48 / 320 (15 %)
0x800033c0 IRQ 00     (real size 2048): unused 1524     usage 524 / 2048 (25 %)
uart:~$ flash page_count 
[00:00:13.804,000] <err> os: ***** MPU FAULT *****
[00:00:13.804,000] <err> os:   Instruction Access Violation
[00:00:13.804,000] <err> os: r0/a1:  0x6000def4  r1/a2:  0x80002c78  r2/a3:  0x80002c7c
[00:00:13.804,000] <err> os: r3/a4:  0xf6410f02 r12/ip:  0x00000000 r14/lr:  0x6000d09d
[00:00:13.804,000] <err> os:  xpsr:  0x60000000
[00:00:13.804,000] <err> os: Faulting instruction address (r15/pc): 0xf6410f02
[00:00:13.804,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:13.804,000] <err> os: Current thread: 0x80000230 (shell_uart)
[00:00:13.840,000] <err> os: Halting system
</pre>

danieldegrasse, 2021-11-05:
@bbilas The flash shell on the mimxrt1060_evk will default to using the flexspi controller, but the actual flash part is  "IS25WP064". Can you try running `flash set_device IS25WP064`, and then attempting to use the flash commands?

bbilas, 2021-11-05:
That was a good hint :)

<pre>
uart:~$ flash set_device IS25WP064
Leaving behind device FLEXSPI
uart:~$ flash page_count 
Flash device contains 2048 pages.
</pre>

So there is still a problem with the shell deadlock on the smp_svr sample.

danieldegrasse, 2021-11-05:
> So there is still a problem with the shell deadlock on the smp_svr sample.

If you run that sample under a debugger, can you see where the code hangs?

bbilas, 2021-11-08:
I've replaced the devboard with the new one and did a clean build and it seems that this issue had gone away but unfortunately there is still something wrong that causes a kernel panic. The stack is:

<pre>
#0  arch_system_halt (reason=reason@entry=0)
    at /home/bartekk/zephyr-builds/zephyr/kernel/fatal.c:32
#1  0x6000e6e4 in k_sys_fatal_error_handler (reason=reason@entry=0, 
    esf=esf@entry=0x80005478 <z_interrupt_stacks+1976>)
    at /home/bartekk/zephyr-builds/zephyr/kernel/fatal.c:44
#2  0x6000e792 in z_fatal_error (reason=reason@entry=0, esf=esf@entry=0x80005478 <z_interrupt_stacks+1976>)
    at /home/bartekk/zephyr-builds/zephyr/kernel/fatal.c:127
#3  0x6000ab48 in z_arm_fatal_error (reason=reason@entry=0, esf=esf@entry=0x80005478 <z_interrupt_stacks+1976>)
    at /home/bartekk/zephyr-builds/zephyr/arch/arm/core/aarch32/fatal.c:63
#4  0x6000b2fe in z_arm_fault (msp=<optimized out>, psp=<optimized out>, exc_return=<optimized out>, 
    callee_regs=<optimized out>)
    at /home/bartekk/zephyr-builds/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:1051
#5  0x6000b494 in z_arm_usage_fault ()
    at /home/bartekk/zephyr-builds/zephyr/arch/arm/core/aarch32/cortex_m/fault_s.S:102
#6  <signal handler called>
#7  0x00000040 in FLEXSPI_CheckAndClearError (base=0x40 <FLEXSPI_CheckAndClearError+64>, status=<optimized out>)
    at /home/bartekk/zephyr-builds/modules/hal/nxp/mcux/drivers/imx/fsl_flexspi.c:237
#8  0x6000ad86 in _isr_wrapper ()
    at /home/bartekk/zephyr-builds/zephyr/arch/arm/core/aarch32/isr_wrapper.S:234
#9  <signal handler called>
#10 0xaaaaaaaa in ?? ()

</pre>

and the kernel panic is:

<pre>
*** Booting Zephyr OS build v2.7.99-931-gf90b24536184  ***
[00:00:00.005,000] <inf> main: hello world
[00:00:00.006,000] <err> os: ***** USAGE FAULT *****
[00:00:00.006,000] <err> os:   Illegal use of the EPSR
[00:00:00.006,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x60016c14  r2/a3:  0x00000000
[00:00:00.006,000] <err> os: r3/a4:  0x00000040 r12/ip:  0xaaaaaaaa r14/lr:  0x6000ad87
[00:00:00.006,000] <err> os:  xpsr:  0x60000024
[00:00:00.006,000] <err> os: Faulting instruction address (r15/pc): 0x00000040
[00:00:00.006,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.006,000] <err> os: Fault during interrupt handling

[00:00:00.006,000] <err> os: Current thread: 0x80001700 (shell_uart)
[00:00:00.053,000] <err> os: Halting system
</pre>

To reproduce this error build simple hello world sample:

```
#include <zephyr.h>
#include <sys/printk.h>
#include <logging/log.h>

LOG_MODULE_REGISTER(main, LOG_LEVEL_INF);

void main(void)
{
	LOG_INF("hello world");
}
```

with the following configuration:

<pre>
CONFIG_NEWLIB_LIBC=y

CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_CDC_ACM=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y

CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_UART_LINE_CTRL=y

CONFIG_LOG=y
CONFIG_LOG_STRDUP_MAX_STRING=64
CONFIG_SHELL=y
CONFIG_SHELL_LOG_BACKEND=y
CONFIG_I2C=y
CONFIG_SENSOR=y
CONFIG_LSM6DSL=y

CONFIG_FLASH=y
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2304
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_ASSERT=y
</pre>

and board overlay

<pre>
&lpi2c1 {
	clock-frequency = <I2C_BITRATE_FAST>;

	lsm6dsl@6a {
		compatible = "st,lsm6dsl";
		reg = <0x6a>;
		irq-gpios = <&gpio1 26 GPIO_ACTIVE_HIGH>;
		label = "LSM6DSL";
	};
};

&lpuart2 {
	status = "okay";
	current-speed = <9600>;
};

&lpuart3 {
	status = "okay";
	current-speed = <115200>;
};

&lpuart6 {
	status = "okay";
	current-speed = <115200>;
};

&zephyr_udc0 {
	cdc_acm_uart0 {
		compatible = "zephyr,cdc-acm-uart";
		label = "CDC_ACM_0";
	};

	cdc_acm_uart1 {
		compatible = "zephyr,cdc-acm-uart";
		label = "CDC_ACM_1";
	};
};
</pre>

The kernel panic doesn't occur when the firmware is compiled with -O2, -Og or -O0 compiler optimizations.

bbilas, 2021-11-08:
@de-nordic @danieldegrasse There is another way to reproduce this issue using `flash-shell` sample with `CONFIG_FLASH_LOG_LEVEL_DBG` option enabled. Even though `flash page_count` works correctly, using e.g. `flash page_erase 2047` causes shell deadlock.

<pre>
*** Booting Zephyr OS build v2.7.99-1160-g131a633cdb7b  ***
Found flash controller FLEXSPI.
Flash I/O commands can be run.

[00:00:00.000,000] <dbg> flash_flexspi_nor.flash_flexspi_nor_get_vendor_id: Reading id
[00:00:00.000,000] <dbg> flash_flexspi_nor.flash_flexspi_nor_init: Vendor id: 0x9d
[00:00:00.000,000] <dbg> flash_flexspi_nor.flash_flexspi_nor_write_status: Writing status register
[00:00:00.000,000] <dbg> flash_flexspi_nor.flash_flexspi_nor_read_status: Reading status register
[00:00:00.000,000] <dbg> flash_flexspi_nor.flash_flexspi_nor_wait_bus_busy: status: 0x40
uart:~$ flash set_device IS25WP064
Leaving behind device FLEXSPI
uart:~$ flash page_count 
Flash device contains 2048 pages.
uart:~$ flash page_erase 2047
Erasing page 2047 (start offset 0x7ff000, size 

</pre>

and the same sample without `CONFIG_FLASH_LOG_LEVEL_DBG` enabled:

<pre>
*** Booting Zephyr OS build v2.7.99-1160-g131a633cdb7b  ***
Found flash controller FLEXSPI.
Flash I/O commands can be run.

uart:~$ flash set_device IS25WP064
Leaving behind device FLEXSPI
uart:~$ flash page_count
Flash device contains 2048 pages.
uart:~$ flash page_erase 2047
Erasing page 2047 (start offset 0x7ff000, size 0x1000)
uart:~$
</pre>

That is really really weird.

danieldegrasse, 2021-11-09:
> That is really really weird.

@bbilas It is. The actual fault you see in the `flash page_erase` call *seems* to occur at a `b.w 0x0` instruction in fsl_flexspi.c (NXP's HAL driver for the flexspi). 

I say that it seems to because this instruction shows up in gdb's assembly view when the flash shell example is built with `CONFIG_FLASH_LOG_LEVEL_DBG=n` and when built with `CONFIG_FLASH_LOG_LEVEL_DBG=y`. If I set a breakpoint near this instruction when `CONFIG_FLASH_LOG_LEVEL_DBG=n`, and then single step through instructions using gdb, I can still cause the error to occur, but it won't occur unless I set a breakpoint near the `b.w 0x0` instruction.

At this point, I'm still unsure exactly what is causing the issue. If you want to see exactly which instruction is causing the fault, place a breakpoint call on line 810 of fsl_flexspi.c (in modules/hal/nxp/mcux/drivers/imx/fls_flexspi.c), like so:
```
if (xfer->cmdType == kFLEXSPI_Command)
{
	__asm__ volatile("bkpt");
        result = FLEXSPI_CheckAndClearError(base, base->INTR);
}
```

from there, you should be able to see what address the `b.w` instruction is at, or if it's there at all in your build. I'm building on Linux, using Zephyr SDK 0.13.1, against  `v2.7.99-1183-ga8c9cf0b0a`.

To summarize what I know so far:
- Provided you build with CONFIG_FLASH_LOG_LEVEL_DBG=y, the fault will reproduce as far back as v2.7.0 (haven't tried further)
- Setting a breakpoint near the faulting instruction, and then single stepping (with `gdb si`), seems to cause the program to behave differently
- The `b.w 0x0` instruction appears when building with `CONFIG_FLASH_LOG_LEVEL_DBG=n`, but does not cause an issue unless you single step near it.

danieldegrasse, 2021-11-10:
@bbilas The root cause of the issue is due to the Cortex M7 accessing the flash memory for instruction fetches at the same time as the FlexSPI driver accesses it for writes, reads, or erases. This causes a variety of "weird" bugs. The FlexSPI driver itself is relocated to ITCM at boot, with calls like these in the relevant CMake files:
```zephyr_code_relocate(flash_mcux_flexspi_nor.c ${CONFIG_FLASH_MCUX_FLEXSPI_XIP_MEM}_TEXT)```
However, the logging subsystem is never relocated to ITCM. This means that if logging is enabled in the FlexSPI driver, it will call code in the logging subsystem while actively accessing the FlexSPI device. Since the code for the logging subsystem is stored in flash (on the FlexSPI itself), the flash is now being accessed for code execution while being written to.

For now, I would just ensure that you select `CONFIG_FLASH_LOG_LEVEL_OFF=y` in your project, to ensure the flash driver doesn't try to use the logging subsystem. I'll work up a PR sometime tomorrow to relocate the logging code into ITCM, or just remove it from the FlexSPI driver entirely.

@dleach02, pretty sure the FlexSPI being accessed concurrently is the root cause of  #39340. Setting `CONFIG_FLASH_LOG_LEVEL_OFF=y` fixes that as well.

bbilas, 2021-11-10:
It seems that setting `CONFIG_FLASH_LOG_LEVEL_OFF=y` fixes the issue with `flash-shell` sample. I've also checked it with my application that uses MCUMGR over uart and sometimes it's stucking during firmware uploading but it needs more investigating because I'm not sure if it's related to this issue at all.

de-nordic, 2021-11-10:
> It seems that setting `CONFIG_FLASH_LOG_LEVEL_OFF=y` fixes the issue with `flash-shell` sample. I've also checked it with my application that uses MCUMGR over uart and sometimes it's stucking during [...]

The stacking may by caused by halting the CPU for erase. Have you tried enabling Kconfig option `CONFIG_IMG_ERASE_PROGRESSIVELY`?

When you use mcumgr-cli, and have the above option enabled you may do `image upload -e`, this will make mcumgr erase pages as data comes instead of pre-erasing target slot.

bbilas, 2021-11-10:
> The stacking may by caused by halting the CPU for erase. Have you tried enabling Kconfig option `CONFIG_IMG_ERASE_PROGRESSIVELY`?
>
> When you use mcumgr-cli, and have the above option enabled you may [...]

After setting `CONFIG_IMG_ERASE_PROGRESSIVELY` option the stucking has gone, thanks for the hint :) So it seems that everything is working fine and we are waiting for the proper fix from @danieldegrasse. Thanks guys for your investigation.

de-nordic, 2021-11-10:
Btw, what I mean by 

> `image upload -e`, this will make mcumgr erase pages as data comes instead of pre-erasing target slot.

is that mcumgr-cli sends two SMP commands for the `image upload`: first is "erase slot 1" followed by "image upload to best available slot"; when you give it `-e` option it will only send the the second one.

danieldegrasse, 2021-11-10:
Ended up just removing the logging statements from the FlexSPI driver. Trying to track down every file that needs to be relocated to ITCM in the logging subsystem is probably going to end up creating more issues, since every logging backend also would need to be relocated.

PR #40270 (2021-11-23): drivers: mcux_flexspi: Removing Logging Statements from FlexSPI driver
Program flow will behave in undefined ways if the ARM core in NXP  i.MXRT parts attempts to access the flash memory using XIP while the flexspi driver is writing to or reading from flash. The undefined behavior manifests in a variety of ways, such as in #39340, where a load from flexspi memory returns 0x0 (and the subsequent branch causes a usage fault), or in #40133, where the program loads a garbage memory address and branches to it.

The flexspi driver itself is relocated to ITCM at runtime using `zephyr_code_relocate` m [... cut]
commit e70907f5 (2021-11-23): drivers: mcux_flexspi: Default logging to disabled when XIP is used

Program flow will behave incorrectly (memory and instruction fetches
return invalid data) if Flexspi is accessed by the Flexspi driver while
being used as XIP memory by the Cortex M7.

Set logging to disabled by when XIP mode is used in the memc and
flexspi drivers, and warn the user if they attempt to enable it.

Fixes #40133
PR #40794 (2021-12-08): [Backport v2.7-branch] drivers: mcux_flexspi: Default logging to disabled when XIP is used
Program flow will behave incorrectly (memory and instruction fetches
return invalid data) if Flexspi is accessed by the Flexspi driver while
being used as XIP memory by the Cortex M7.

Set logging to disabled by when XIP mode is used in the memc and
flexspi drivers, and warn the user if they attempt to enable it.

Fixes #40133
commit 644ebe54 (2021-12-08): drivers: mcux_flexspi: Default logging to disabled when XIP is used

Program flow will behave incorrectly (memory and instruction fetches
return invalid data) if Flexspi is accessed by the Flexspi driver while
being used as XIP memory by the Cortex M7.

Set logging to disabled by when XIP mode is used in the memc and
flexspi drivers, and warn the user if they attempt to enable it.

Fixes #40133
closed 2021-11-23 by cfriedt