zephyrproject-rtos/zephyr #40146: On the status of DT-defined regions and MPU

carlocaione, 2021-11-06:
I want to keep discussing what came up in https://github.com/zephyrproject-rtos/zephyr/pull/30403 in the light of the problem expressed (among others) in https://github.com/zephyrproject-rtos/zephyr/issues/40101.

# The problem

Sometime there's a need to have a memory segment that is not cached, which can be used for DMA descriptors, or small DMA buffers. Sometimes it's not really important the location of this memory segment but in other cases we must be able to put this segment in a very precise location in memory.

The problem can be described in terms of three different sub-tasks:
1. How to place a memory segment to a specific memory location described into the DT
2. How to set the attributes for this segment (cacheability) using MPU / MMU
3. How to use this buffer (statically place buffers in this region or use it as a pool to get memory from)
4. How to extend the previous points to support multiple segments at the same time

Right now Zephyr is missing a coherent and unified way to deal with all these points.

## Current state of (Zephyr) affairs
## `NOCACHE_MEMORY`
When `CONFIG_NOCACHE_MEMORY` is enabled a special section ([nocache.ld](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/common/nocache.ld)) is placed in the linker script. This section is delimited by [symbols](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/linker/linker-defs.h#L332-L336) that can be used by the [MPU driver](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/arm/core/aarch32/mpu/arm_core_mpu.c#L74-L80) to set the correct attributes for the region. The user can then statically place buffers in this region using some [variable attributes](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/linker/section_tags.h#L43) and use these for example to  [statically place variables](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/ethernet/eth_sam_gmac.c#L132-L133) or to [define](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/kernel.h#L5230-L5231) a new [heap](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/usb/device/usb_dc_mcux.c#L71).

This approach has several problems:
1. It is not possible to define the address of the non-cacheable region
2. It is possible to have only one single non-cacheable region
3. The MMU / MPU driver must be actively changed to set the correct attributes for the region
4. It relies on a lot of custom changes (linker script, MPU driver, user code, ...)

## `reserved-memory`
This was introduced to better address the problem of how to [define](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/linker/devicetree_reserved.h) several memory sections in the DT at specific address and with specific size that the user can easily access. This is a multi-purpose tool: for each [reserved-memory region in the DT](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/boards/qemu_cortex_a53/reserved_memory/boards/qemu_cortex_a53.overlay#L8-L25) a new section is created into the linker script and the user can then [access to location and size](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/boards/qemu_cortex_a53/reserved_memory/src/main.c) of this section to do whatever the user needs. For example the user could also [place buffers or variables](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/boards/qemu_cortex_a53/reserved_memory/src/main.c#L11-L12) in these regions.

If the reserved region is marked as non-cacheable in the MPU / MMU then we have solved the original problem.

The biggest problem with this approach is that this is a multi-purpose tool and we are missing the link between the reserved regions and the MPU / MMU driver to configure the attributes for the reserved regions. We are also missing a general way to define the memory attributes for each region.

## Shared multi-heap library
This is a [library](https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/os/shared_multi_heap.c) introduced recently that is leveraging the reserved-memory support and the [multi-heap](https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/os/multi_heap.c) library to partially overcome the limitations of the previous methods when dealing with dynamic allocations.

The shared multi-heap memory pool manager uses the multi-heap allocator to manage a set of [reserved memory regions with different capabilities / attributes](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/kernel/mem_heap/shared_multi_heap/boards/qemu_cortex_a53.overlay) (cacheable, non-cacheable, etc...) defined in the DT. The user can [request allocation from the shared pool](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/kernel/mem_heap/shared_multi_heap/src/main.c#L64) specifying the capability / attribute of interest for the memory (cacheable / non-cacheable memory, etc...).

This library is also trying to (at least partially) address the problem identified in the previous section about the missing link between the reserved regions and the MPU / MMU driver using a special [hook](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/multi_heap/shared_multi_heap.h#L60-L79) that can be used to signal to the MPU / MMU driver that special attributes are needed for the memory region is going to be added to the pool.

The problem is that this hook is used at run-time so the MPU / MMU driver must have the capability of configuring the region at run time. For example this is possible for [ARMv8](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/kernel/mem_heap/shared_multi_heap/src/main.c#L26-L29) and X86 but not for all the other architectures.

# Recap
- The `NOCACHE_MEMORY` approach is complex, it doesn't scale and not generic enough
- The `reserved-memory` approach is a general purposes tool and it is missing a way to communicate with the MPU / MMU driver to configure the regions
- The shared multi-heap library is valid only for dynamic allocations and when the MPU / MMU driver is able to configure the regions at run-time

# How is this different from https://github.com/zephyrproject-rtos/zephyr/issues/40101
It's not different, it's just more limited in scope. The discussion in #40101 is dealing with the more general problem of how to define SRAM sections in DT and configure the MPU accordingly. This is more specifically about smaller regions in the SRAM defined for a very specific goal.

# What can be done
Leveraging what was proposed in https://github.com/zephyrproject-rtos/zephyr/pull/30403 what's needed here is (either / or):

1. a sort of MPU API that can be used by other libraries / frameworks (like `reserved-memory` or the shared multi-heap library) to configure (also at run-time) the MPU regions
2. a way to automatically go from the regions defined in the DT to MPU regions to be inited at boot-time by the MPU driver

carlocaione, 2021-11-06:
Tagging the people already involved in the #40101 discussion: @gmarull @Nukersson @galak @erwango @ioannisg @carlescufi

erwango, 2021-11-09:
Question on `NOCACHE_MEMORY` on Cortex-M, since it is defined as `K_MEM_PARTITION_P_RW_U_NA_NOCACHE`, is it actually accessible to user (ie USERSPACE) ?

carlocaione, 2021-11-10:
> Question on `NOCACHE_MEMORY` on Cortex-M, since it is defined as `K_MEM_PARTITION_P_RW_U_NA_NOCACHE`, is it actually accessible to user (ie USERSPACE) ?

Not expert on the Cortex-M but the `U_NA` usually implies that memory is not available for user threads.

erwango, 2021-11-10:
> > Question on `NOCACHE_MEMORY` on Cortex-M, since it is defined as `K_MEM_PARTITION_P_RW_U_NA_NOCACHE`, is it actually accessible to user (ie USERSPACE) ?
> 
> Not expert on the Cortex-M but the `U_NA` usually implies that memory is not available for user threads.

This is what confusing me. Let me know if I'm off topic, but if that area is not accessible to users, then it can't be used by API users when USERSPACE is enabled, cf [this discussion](https://github.com/zephyrproject-rtos/zephyr/pull/32833/files#r648917589).
I think this part should also be considered into the picture you're drawing, because this is one of the need we have today on our side for M7 users which is not available at all today.

aurel32, 2021-11-10:
`NOCACHE_MEMORY` was designed for kernel drivers only, mostly for DMA descriptors which can't be handled by cache operation. It was not designed to be accessible to userspace.

When USERSPACE is enabled, each thread has a different memory space (the MPU is reconfigured on task switch), so you do not want to give them access to a common memory area, especially if it is already used by kernel drivers. Note however that you can give threads access to additional memory areas, including with specific caching attributes, using memory domains (https://docs.zephyrproject.org/latest/reference/usermode/memory_domain.html). However I haven't tried if it could solve the issue discussed in this thread.

dleach02, 2021-11-16:
@danieldegrasse

closed 2022-06-09 by carlocaione